serialport/node-serialport #661: close problem with moxa products (file descriptor not close)

mathieuwlo, 2016-01-26:
Hello,

I'm using moxa product like Uport 1650 or nport 5650, there isn't any problem to open a serial port but when I call close() function of node-serialport it doesn't close the port (file descriptor) so I can't reopen it whitout relaunching my node application.
I'm working on ubuntu 14.04 with kernel 3.19, node version 0.12.7
node serial port version 2.0.6
when I launch my node application with this command
strace -e trace=open,close node myprogramm.js I saw the open("/dev/tty",O_RDWR,O_CLOEXEC)= 13 (number of file descriptor) but when the close function is called I never saw the trace closing the file descriptor open before.

thanks

gfcittolin, 2016-02-16:
I had a similar problem, and found the problem to be a double call to `open()`, generating two file descriptors, and losing track of the first one:

https://github.com/voodootikigod/node-serialport/blob/master/src/serialport_unix.cpp#L134 and
https://github.com/voodootikigod/node-serialport/blob/master/src/serialport_unix.cpp#L186

Easy to see with some debug printf's:
```
* JS call - port.open
>>>>>> EIO_Open - before open path:/dev/ttyACM0
>>>>>> EIO_Open - after open path:/dev/ttyACM0, fd:12       <-- from L134
>>>>>>>> setup - fd:12
>>>>>>>> setup - before open fd:12
>>>>>>>> setup - after open fd:13                           <-- from L186, fd 12 gets lost
>>>>>>>> setup - before fcntl fd:13
>>>>>>>> setup - after fcntl fd:13, cloexec:0
>>>>>>>> setup - before tcgetattr fd:13
>>>>>>>> setup - after tcgetattr fd:13
>>>>>>>> setup - before cfsetispeed fd:13, baudRate:13
>>>>>>>> setup - after cfsetispeed fd:13, baudRate:13
>>>>>>>> setup - before cfsetospeed fd:13, baudRate:13
>>>>>>>> setup - after cfsetospeed fd:13, baudRate:13
>>>>>>>> setup - before tcflush fd:13
>>>>>>>> setup - before tcsetattr fd:13
>>>>>>>> setup - after tcsetattr fd:13
* JS callback - onOpen
```

The `open("/dev/tty",O_RDWR,O_CLOEXEC)= 13 ` you are seeing is node opening the terminal, not your serial device

reconbot, 2016-04-06:
We fixed a ton of bugs in our latest beta specifically ones addressing this issue. https://github.com/voodootikigod/node-serialport/issues/733 Please try it out `serialport@2.0.7-beta3` and report back!

reconbot, 2016-04-08:
We've had confirmed fixes for this in beta4

closed 2016-04-08 by reconbot