serialport/node-serialport #663: Serialport and data corruption sending data to Grbl using SerialPort

openhardwarecoza, 2016-01-29:
Hi

I use Serialport on my project, Laserweb, to drive CNC/Lasers from node.js

We have had a few users reporting strange movements, and I have been able to replicate this locally as well.

Grbl reports back an 'error' when the gcode command it receives is invalid.  
Sadly it cannot echo what it received, but I do know that the gcode placed into SerialPort is valid (running the job via another host application written in Golang, it does not encounter this issue)

So for example, you would be sending:
G1 X100 F200\n
G1 Y100 F200\n

But grbl would see G1 X100 F\n G1 Y 100 F200\n (so some chars after F didnt make it down the wire and it reports a Not a Number error  (random spots each time you test)
It's easy to blame hardware, but three people with different hardware, one being myself, its not easy to rule that being the issue. Secondly testing against the same hardware using other software, the issue goes away

First of all, I am totally happy to admit the problem may be in my implementation of SP, if its not a bug in SP. But either way I need some help/advise to track this issue down:

Refer to the Issue on Laserweb at https://github.com/openhardwarecoza/LaserWeb/issues/94

Some quick breakdown on how I use SP in my app:

1.  We iterate through all the ports on the macjhine and instate handler for each port:
```
for (var i=0; i<ports.length; i++) {

sp[i].handle = new SerialPort(ports[i].comName, {
			parser: serialport.parsers.readline("\n"),
			baudrate: config.serialBaudRate
		});
```

2.  Then later on in the app, we put the gcode in a queue (array split by \n) and write it to the handler:
```
sp[port].handle.write(t+"\n");
```
(where t = the line of gcode to send) 

3.  while writing code we also parse returned messages from a handler:
```
sp[i].handle.on("data", function (data) {
				serialData(data, i);
			});

unction serialData(data, port) {
	// new line of data terminated with \n
	//console.log('Port '+port+' got newline from serial: '+data);
       // do  a bunch of indexOf checks and websocket emits. 
	if (data.indexOf('#') == 0) {.....truncated for issue..... }
       ..... truncated
}
````
4.  On an interval we add another write to the handler (to poll for position feedback)
```
setInterval(function() {
			sp[port].handle.write("M114\n"); //for Marlin
		}, 1000);
```
or 
```
setInterval(function() {
			sp[port].handle.write("?\n"); //for grbl
		}, 1000);
```

My theory at this point is that the interval handler may write at the same time as one of the queue writes causing a race condition?  But commenting it out does not fully improve the situation

So my other theory is that the handler write we do happens asyncronously and thus two commands get sent out the port, the second before the first one finished writing?   

Please advise, is the method we use acceptable for SerialPort? Or is there a better way of handling a queue of serial commands (gcode)

(Line no's to production code)

- https://github.com/openhardwarecoza/LaserWeb/blob/master/server.js#L150 (connecting to ports and instating handlers)
- https://github.com/openhardwarecoza/LaserWeb/blob/master/server.js#L217 (Parse Serial Responses)
- https://github.com/openhardwarecoza/LaserWeb/blob/master/server.js#L677 (Start a Gcode job )
-  https://github.com/openhardwarecoza/LaserWeb/blob/master/server.js#L501 (Send lines in queue)
- https://github.com/openhardwarecoza/LaserWeb/blob/master/server.js#L222 (jnterval commands for position/temp feedback)

bulburDE, 2016-02-17:
I can report the same problem using grbl 0.9j with a baudrate of 115200. You can uncomment #define REPORT_ECHO_LINE_RECEIVED in grbl/config.h make grbl echo each line it receives. Using this I could observer that various errors occured due to missing chars. Here are some example lines echoed by grbl:

MG1X28.350023Y102.000080F1400S255
G1X28.350023Y105.000082F140X2M5
G1X22.350018Y102.0052.SX10M3G.0SX3400055
G1X22.350018Y0520S00GY4M2

I saved the gcode generated by LaserWeb and uploaded it to grbl using GrblWeb which has been successful every time I tested it. GrblWeb uses version 1.4.x of serialport. Maybe you can trace the error inspecting their implementation.

rodovich, 2016-02-25:
@openhardwarecoza for what it's worth, `?` is a [real-time command](https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.9#real-time-commands----and-ctrl-x) and doesn't require a `\n` after it. Non-real-time commands go into [a buffer](https://github.com/grbl/grbl/blob/master/grbl/protocol.h#L25) that by default holds 79 bytes (a circular buffer's capacity is 1 less than its size) so you have to make sure there's room for each new gcode line before you send it.

reconbot, 2016-03-27:
@openhardwarecoza I'm having a hard time following what you're doing. 

I can confirm that with Node being single threaded and with the nature of our c++ layer, we have a write queue per port. Unless you're doings something very strange, each write should happen in order.

closed 2016-05-01 by reconbot