serialport/node-serialport #665: win7 nw.js 0.12.3 can't find SerialPort

CheneyPeng, 2016-02-03:
I rebuild SerialPort as `node-pre-gyp rebuild --runtime=node-webkit --target=0.12.3`
and i use like 
`var com = require("serialport"); com.list(function (err, ports) {  ports.forEach(function(port) {console.log(port.comName);	 });	}); var s = new com.SerialPort("COM3");`
i can get the SerialPort list
but there has a error :
`Uncaught TypeError: undefined is not a function` said `new com.SerialPort("COM3")` undefined.
i don't know how to fix it. can help me? THX!!

JuppySingh, 2016-02-15:
I got the same error and after numerous combinations got it to work finally. Attached is the x64 build which you can try.

[serialport.zip](https://github.com/voodootikigod/node-serialport/files/130248/serialport.zip)

reconbot, 2016-04-06:
Please try the latest beta, I'm let to believe the upgrade to `Nan` might help this. We've also fixed a few windows issues. #733 Please try it out serialport@2.0.7-beta3 and report back!

reconbot, 2016-04-06:
@JuppySingh would you be able to share what you did to make it work?

JuppySingh, 2016-04-07:
I am able to install serialport@2.0.7-beta4 with the latest Node x64 v5.10.1.

I then try and rebuild serialport for nw.js using the following commands:

`node-pre-gyp configure --runtime=node-webkit --target=0.12.3
node-pre-gyp build --runtime=node-webkit --target=0.12.3`

However, when I try and run a test. I get the following error:

`node_modules\serialport\node_modules\bindings\bindings.js:83 
Uncaught Error: Module version mismatch. Expected 43, got 47.`

reconbot, 2016-04-07:
can you do the same thing again but remove the `out` folder and see what happens?

reconbot, 2016-04-07:
I think I have this solved in #737 but you testing that will confirm

JuppySingh, 2016-04-08:
It is working fine now. Thanks!

reconbot, 2016-04-08:
OK cool! Il have this fixed in beta5 or the next release!

closed 2016-04-08 by reconbot