serialport/node-serialport #666: Raspberry pi issue

DamoInUk, 2016-02-05:
Hoping someone can help me out. I am new to node.js and the raspberry pi for the matter, I am trying to get serialport installed. I have tried following the instructions and it seems to build OK. I get serialport@2.0.6 reporting successful built.

My problem comes when trying to use it. I get the following error:
Error: Cannot find module '/home/pi/my_node_proj/node_modules/serialport/build/Release/**node-v11-linux-arm**/serialport,node

When I look in that path the serialport.node file is not there and is in fact in:
/home/pi/my_node_proj/node_modules/serialport/build/Release/**node-v46-linux-arm**/serialport.node

Anybody know how I can resolve this?

I am pretty sure this a chair to keyboard issue as I am very new to node and the raspberry pi.

DamoInUk, 2016-02-05:
Still having problems but doesn't seem correct to report this as an issue

closed 2016-02-05 by DamoInUk

fasiha, 2016-04-04:
Tried making a symbolic link from what was there to what node is looking for:
```
$ cd node_modules/serialport/build/Release && ln -s node-v46-linux-arm node-v11-linux-arm
```
and that gets you farther but then you hit this error:
```
module.js:356
  Module._extensions[extension](this, filename);
                               ^
Error: /home/pi/Documents/Dev/node-adsb-serial/node_modules/serialport/build/Release/node-v46-linux-arm/serialport.node: undefined symbol: node_module_register
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Module.require (module.js:364:17)
    at require (module.js:380:17)
    at Object.<anonymous> (/home/pi/Documents/Dev/node-adsb-serial/node_modules/serialport/serialport.js:14:25)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Module.require (module.js:364:17)
```

fasiha, 2016-04-04:
Got it: per https://github.com/Unitech/pm2/issues/1477#issuecomment-126737121 an old version of node is being picked up (`node -v` showed 0.10…). Solution: `sudo apt-get remove nodejs` and reinstall Node 4.x or 5.x however you installed it before (in my case, download the ARMv7 binary from https://nodejs.org/en/download/, uncompress, and copy the directories to /usr/local).

@DamoInUk does that help?

fasiha, 2016-04-04:
Still, I don't think this should be closed—the build name in Release is wrong.

DamoInUk, 2016-04-05:
Thanks for the comment. I managed to sort it out a while ago now. Pretty much followed the same steps you mentioned. It was my first time using a pi and node.js so I wasn't really sure what I was doing or if there was an issue.