serialport/node-serialport #667: Unable to build on Node 5.6.0 or 4.3.0

crertel, 2016-02-11:
I'm trying to install a simple module that uses node-serialport (node-waterrower), and unfortunately it won't install due to errors. Build environment is Raspbian Jesse on an RPi 2.

A rough snippet of the errors looks like:

```
make: Entering directory '/home/pi/crewcut/node_modules/node-waterrower/node_modules/serialport/build'                                                                                [61/385]
  CXX(target) Release/obj.target/serialport/src/serialport.o
In file included from ../src/serialport.h:5:0,
                 from ../src/serialport.cpp:3:
../node_modules/nan/nan.h:261:25: error: redefinition of ‘template<class T> v8::Local<T> _NanEnsureLocal(v8::Local<T>)’
 NAN_INLINE v8::Local<T> _NanEnsureLocal(v8::Local<T> val) {
                         ^
../node_modules/nan/nan.h:256:25: note: ‘template<class T> v8::Local<T> _NanEnsureLocal(v8::Handle<T>)’ previously declared here
 NAN_INLINE v8::Local<T> _NanEnsureLocal(v8::Handle<T> val) {
                         ^
../node_modules/nan/nan.h:661:13: error: ‘node::smalloc’ has not been declared
     , node::smalloc::FreeCallback callback
             ^
../node_modules/nan/nan.h:661:35: error: expected ‘,’ or ‘...’ before ‘callback’
     , node::smalloc::FreeCallback callback
                                   ^
../node_modules/nan/nan.h: In function ‘v8::Local<v8::Object> NanNewBufferHandle(char*, size_t, int)’:
../node_modules/nan/nan.h:665:50: error: ‘callback’ was not declared in this scope
         v8::Isolate::GetCurrent(), data, length, callback, hint);
                                                  ^
../node_modules/nan/nan.h:665:60: error: ‘hint’ was not declared in this scope
         v8::Isolate::GetCurrent(), data, length, callback, hint);
                                                            ^
../node_modules/nan/nan.h: In function ‘v8::Local<v8::Object> NanNewBufferHandle(const char*, uint32_t)’:
../node_modules/nan/nan.h:672:67: error: call of overloaded ‘New(v8::Isolate*, const char*&, uint32_t&)’ is ambiguous
     return node::Buffer::New(v8::Isolate::GetCurrent(), data, size);
                                                                   ^
../node_modules/nan/nan.h:672:67: note: candidates are:
In file included from ../node_modules/nan/nan.h:25:0,
                 from ../src/serialport.h:5,
                 from ../src/serialport.cpp:3:
/home/pi/.node-gyp/4.3.0/include/node/node_buffer.h:31:40: note: v8::MaybeLocal<v8::Object> node::Buffer::New(v8::Isolate*, v8::Local<v8::String>, node::encoding) <near match>
 NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                        ^
/home/pi/.node-gyp/4.3.0/include/node/node_buffer.h:31:40: note:   no known conversion for argument 3 from ‘uint32_t {aka unsigned int}’ to ‘node::encoding’
/home/pi/.node-gyp/4.3.0/include/node/node_buffer.h:43:40: note: v8::MaybeLocal<v8::Object> node::Buffer::New(v8::Isolate*, char*, size_t) <near match>
 NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                        ^
/home/pi/.node-gyp/4.3.0/include/node/node_buffer.h:43:40: note:   no known conversion for argument 2 from ‘const char*’ to ‘char*’
In file included from ../src/serialport.h:5:0,
                 from ../src/serialport.cpp:3:
../node_modules/nan/nan.h: In function ‘v8::Local<v8::Object> NanNewBufferHandle(uint32_t)’:
../node_modules/nan/nan.h:676:61: error: could not convert ‘node::Buffer::New(v8::Isolate::GetCurrent(), size)’ from ‘v8::MaybeLocal<v8::Object>’ to ‘v8::Local<v8::Object>’
     return node::Buffer::New(v8::Isolate::GetCurrent(), size);
                                                             ^
```

There is a bit more output, but it seems pretty obvious that something is broken in nan.

I did a little more investigating, and notice that the version of nan for node-serialport is at `~2.0.0`. The current version of nan (on their project page) is `2.2.0`.

Is there any chance that you might want to upgrade to the newer version of nan? Would that help?

crertel, 2016-02-11:
GCC version in use is `gcc (Raspbian 4.9.2-10) 4.9.2`.

fivdi, 2016-02-12:
The latest version of `node-waterrower` published on npm doesn't correspond to the to the develop branch on github. The latest version on npm depends on serialport v1.7.4 which will not compile with Node.js v4 or greater. The version on github depends on serialport v2.0.6 which will compile with Node.js v4 or higher. What should work is `npm install "git://github.com/jamesnesfield/node-waterrower.git#develop"`. Whether or not it functions correctly is a different question. It's probably best to create an issue at `node-waterrower` and ask if a new version can be published on npm.

crertel, 2016-02-13:
That appears to have solved the problem!

Many thanks!

Sorry to take up time with a downstream problem. :(

closed 2016-02-13 by crertel