serialport/node-serialport #668: 'node-pre-gyp' dependancy breaks nw.js packaging with nw-builder (on Windows)

palavrov, 2016-02-11:
It was very difficult to dig to that.
node-pre-gyp has some very deep paths under node_modules that exceeds 260 path limit in windows. 

I'm able to package my application in zip file embedded to the nw.exe but after that id just doesn't work. What I see is that in %TEMP% that archive is not fully extracted so the nw binary just silently exits (which is another nw.js issue itself). Just deleting the node_modules subdirectory of serialport solves the extraction but obviously we are not able to load the serialport module.
For the moment the only workaround is to keep app files unpacked next to nw.exe.

In general I *think* that node-pre-gyp is more a development dependency i.e. we should not depend on it during deployment. The whole deployment of the node-serialport is pain in the ass - I need to build my setup package on windows with installed VS - there is no prebuild binary for the nw.js.

gbaumgart, 2016-03-07:
+100

the fact it doesnt play with npm3, prevents us from shipping it on windows-machines. 

what could be a fix? Can i move that folders in serialport/node_modules up in the actual folder (app/node_modules).

palavrov, 2016-03-07:
@gbaumgart actually I found how to solve it - with migration to electron.
It took me a while but at the end I've nice automated build of windows setup - something that was not able to produce with nw.js. Here are my configuration files in case that you want to try:

package.json
```json
{
  "name": "myapp",
  "version": "0.0.0",
  "description": "",
  "keywords": [],
  "devDependencies": {
    "electron-builder": "^2.7.2",
    "electron-packager": "^5.2.1",
    "electron-prebuilt": "^0.36.7",
    "electron-rebuild": "^1.1.3",
    "rimraf": "^2.5.2"
  },
  "engines": {
    "node": ">=0.8.0"
  },
  "builder": {
    "win": {
      "title": "MY-APP",
      "icon": "assets/myappico",
      "version": "0.0.0",
      "verbosity": 1
    },
    "osx": {
      "title": "MY-APP",
      "background": "assets/background.jpg",
      "icon": "assets/myapp.icns",
      "version": "0.0.0",
      "icon-size": 80,
      "contents": [
        {
          "x": 438,
          "y": 344,
          "type": "link",
          "path": "/Applications"
        },
        {
          "x": 192,
          "y": 344,
          "type": "file"
        }
      ]
    },
    "linux": {
      "title": "MY-APP",
      "version": "0.0.0"
    }
  },
  "scripts": {
    "start": "electron app",
    "preinstall": "cd app && bower install && npm install && cd .. && electron-rebuild -v 0.36.7 -p -f -m app -w serialport -e node_modules/electron-prebuilt",
    "clean:osx": "rimraf dist/osx",
    "clean:win": "rimraf dist/win",
    "clean:linux": "rimraf dist/linux",
    "clean": "rimraf app/node_modules && rimraf app/bower_components && rimraf dist && rimraf node_modules",
    "pack:win": "npm run clean:win && electron-packager app \"MY-APP\" --version=0.36.7 --out=dist/win --platform=win32 --arch=ia32 --icon=assets/MY-APP.ico --asar",
    "pack:osx": "npm run clean:osx && electron-packager app/ \"MY-APP\" --version=0.36.7 --out=dist/osx --platform=darwin --arch=x64 --icon=assets/FG-SYS.icns --asar",
    "pack:linux": "npm run clean:linux && electron-packager app \"MY-APP\" --version=0.36.7 --out=dist/linux --platform=linux --asar",
    "pack": "npm run pack:win && npm run pack:osx && npm run pack:linux",
    "build:win": "npm run pack:win && electron-builder \"dist/win/MY-APP-win32-ia32\" --platform=win --out=\"dist/win\"",
    "build:osx": "npm run pack:osx && electron-builder \"dist/osx/MY-APP-darwin-x64/MY-APP.app\" --platform=osx --out=\"dist/osx\"",
    "build:linux": "npm run pack:linux && electron-builder \"dist/linux/MY_APP\" --platform=linux --out=dist/linux",
    "build": "npm run build:win && npm run build:osx && npm run build:linux"
  }
}
```
app/package.json
```json
{
  "name": "my-app",
  "version": "0.0.1",
  "private": true,
  "main": "main.js",
  "dependencies": {
    "async": "^1.5.2",
    "batch": "^0.5.3",
    "h5.modbus": "^0.1.0",
    "moment": "^2.11.2",
    "serialport": "^2.0.6"
  }
}
```

gbaumgart, 2016-03-07:
excellent! I was talking about electron too. Thanks a lot.

JonathonAshworth, 2016-03-09:
I'm using node-serialport with Electron and having the same problem. I'm not entirely sure how your scripts fix it, since they're just running electron-builder, which uses electron-winstaller under the hood, which is what is failing for me.

Is there any chance the dependencies can be reworked to accommodate this? I don't really understand why node-pre-gyp is a runtime dependency?

That's the path lodash took, since npm have decided they're not implementing a proper fix to this problem. Right now I cannot build our electron app on Windows, which is a major problem.

gbaumgart, 2016-03-09:
it seems to have no problem on win/lin to delete the gyp stuff in node_modules/**

JonathonAshworth, 2016-03-09:
I also have no problem deleting, moving or copying the files. But I'm unable to get any kind of packager or installer to work with them. I've tried packaging into an installer with electron-builder, electron-winstaller, and Inno Setup. None worked. I can't even zip the files (it just fails silently).

It seems some tools can work with long path names, and some can't. And all the installer ones can't, which is why I'm curious how palavrov got it to work with electron-builder.

palavrov, 2016-03-09:
To be honest the problem with the too long path just disappear for me. And I took me a lot of efforts and debugging to realize that it was the problem with nw.js. May be you have different problem.

What I'm doing is using windows user name without spaces i.e. 'palavrov' for example and building the application in "C:\tmp\app" to shorten the build paths, Win7, Developer Command Prompt for VS2015.

The scripts that I show are working pretty good for me, the only change was to rename the real app name to MY-APP to avoid any troubles with my customer. Pay attention to the fact that lack of some settings can abort the setup build - for example `builder.win.icon` - this is not well documented and I was debugging the build until found them all.

gbaumgart, 2016-03-09:
@JonathonAshworth you tried 'nexe' already? works fine too. don't forget you can't package binaries anyway in asar or similar anyway. i gave up on trying and just ship the more cleaned node_modules folder.

reconbot, 2016-04-07:
We've got a thread tracking electron support, our latest beta makes this a lot easier. I think our final beta will solve it for good. https://github.com/voodootikigod/node-serialport/issues/538

npm3 will help solve the path too long issue. `npm dedupe` also helps clean up previously installed node_modules folders. (Advise from MS)

closed 2016-04-07 by reconbot

cheton, 2016-05-06:
Hi @reconbot,

I'm using npm3, but found strange things while upgrading from 3.1.0 to 3.1.1.

npm3 can install serialport@3.1.0 in a relatively flat structure, however, install serialport@3.1.1 will not flatten `node-pre-gyp` but keep the whole dependencies under `serialport/node_modules/node-pre-gyp`, which results in a very deep paths that exceeds the 260 characters limit in Windows.

It can be reproduced as shown below:
```bash
$ npm -v
3.8.8
$ rm -rf node_modules; npm install serialport@3.1.0
$ ls -al node_modules/serialport/node_modules/node-pre-gyp/
total 60
drwxrwxr-x.   4 cheton cheton  4096 May  6 12:00 .
drwxrwxr-x. 112 cheton cheton  4096 May  6 12:00 ..
-rw-rw-r--.   1 cheton cheton   639 Apr 30 03:14 appveyor.yml
drwxrwxr-x.   2 cheton cheton    48 May  6 12:00 bin
-rw-rw-r--.   1 cheton cheton  9383 Apr 30 03:13 CHANGELOG.md
drwxrwxr-x.   3 cheton cheton  4096 May  6 12:00 lib
-rw-rw-r--.   1 cheton cheton  1501 Sep 23  2014 LICENSE
-rw-rw-r--.   1 cheton cheton  3222 May  6 12:00 package.json
-rw-rw-r--.   1 cheton cheton 24088 Apr 20 07:27 README.md

$ rm -rf node_modules; npm install serialport@3.1.1
$ ls -al node_modules/serialport/node_modules/node-pre-gyp/node_modules/
total 40
drwxrwxr-x. 12 cheton cheton 4096 May  6 12:07 .
drwxrwxr-x.  5 cheton cheton 4096 May  6 12:07 ..
drwxrwxr-x.  2 cheton cheton   65 May  6 12:07 .bin
drwxrwxr-x.  6 cheton cheton 4096 May  6 12:07 mkdirp
drwxrwxr-x.  7 cheton cheton 4096 May  6 12:07 nopt
drwxrwxr-x.  5 cheton cheton 4096 May  6 12:07 npmlog
drwxrwxr-x.  5 cheton cheton 4096 May  6 12:07 rc
drwxrwxr-x.  4 cheton cheton 4096 May  6 12:07 request
drwxrwxr-x.  3 cheton cheton  103 May  6 12:07 rimraf
drwxrwxr-x.  4 cheton cheton 4096 May  6 12:07 semver
drwxrwxr-x.  6 cheton cheton 4096 May  6 12:07 tar
drwxrwxr-x.  4 cheton cheton 4096 May  6 12:07 tar-pack
```

Do you know what caused the changes in serialport@3.1.1?

reconbot, 2016-05-06:
Nothing that should have changed this. This is related to #753. I'll have a beta release for you later today that if it works should close that issue and solve this problem permanently.

reconbot, 2016-05-06:
@cheton can you test [`serialport@3.1.2-beta1`](https://github.com/voodootikigod/node-serialport/releases/tag/3.1.2-beta1)? I've removed the package bundling that's causing you issues. It works fine for a lot of situations but I don't have many windows environments to test on.

cheton, 2016-05-07:
`serialport@3.1.2-beta1` works great! Running `npm dedupe` after installing `serialport@3.1.1` can fix this issue as well. Thanks!