serialport/node-serialport #669: Failed to install node-serialport on raspberry pi "jessie"

stefferber, 2016-02-11:
Hello,

I am using a fresh debian jessie distribution "2016-02-09-raspbian-jessie" on my raspberry pi and installed node.js and npm like this:
`$ wget http://node-arm.herokuapp.com/node_latest_armhf.deb
$ sudo dpkg -i node_latest_armhf.deb
$ node -v
v4.2.1
$ npm -v
2.14.7`
$ npm install serialport

I get this error log:

[npm-debug_log.txt](https://github.com/voodootikigod/node-serialport/files/127127/npm-debug_log.txt)

Any idea what I do wrong?

fivdi, 2016-02-12:
If I'm not mistaken the error log indicates that there were errors installing serialport v1.7.4. On the other hand, `npm install serialport` should install serialport v2.0.6. Was the command executed really `npm install serialport`?

stefferber, 2016-02-13:
Thx a lot for your comment!

yes, it was that chain of command, but the context is - as always - more complex ;-)  I got an old serial version before as I got my github repository "cloned locally" which contained "serialport" from my Mac and is about 2 years old (Version: 1.1.3 - Released July 29, 2013):
[https://github.com/stefferber/velux-remote](https://github.com/stefferber/velux-remote)
That's why I deleted all files in "node_modules". That worked fine for other dependencies "node_modules/express".
As the 2 years old "package.json" named  "serialport" : "1.x" npm  install the newest "1.x" version. Right ?-) So I changed the dependency to "serialport" : "2.x" and I all worked fine!

@fivdi - this was exactly the right question! Thx a lot and sorry for bothering you with such a beginners mistake. The issue is resolved.

fivdi, 2016-02-13:
>As the 2 years old "package.json" named "serialport" : "1.x" npm install the newest "1.x" version. Right ?

Correct.

@stefferber as this issue has been resolved you could close it by hitting the `Close issue` button at the bottom of the page :)

closed 2016-02-13 by stefferber