serialport/node-serialport #673: Problem installing on Rasperry Pi

howardkitto, 2016-02-22:
I'm trying to install serialport on my Raspberry Pi

I ran
`sudo apt-get update`
 and 
`sudo apt upgrade -y`

but still get this error when I run 
`sudo npm install serialport`

Is there a solution?

> serialport@2.0.6 install /home/pi/Documents/NodeProjects/webservice/node_modules/serialport
> node-pre-gyp install --fallback-to-build

make: Entering directory '/home/pi/Documents/NodeProjects/webservice/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
In file included from ../src/serialport.h:5:0,
                 from ../src/serialport.cpp:3:
../node_modules/nan/nan.h:316:47: error: ‘REPLACE_INVALID_UTF8’ is not a member of ‘v8::String’
   static const unsigned kReplaceInvalidUtf8 = v8::String::REPLACE_INVALID_UTF8;
                                               ^
serialport.target.mk:84: recipe for target 'Release/obj.target/serialport/src/serialport.o' failed
make: *** [Release/obj.target/serialport/src/serialport.o] Error 1
make: Leaving directory '/home/pi/Documents/NodeProjects/webservice/node_modules/serialport/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/share/node-gyp/lib/build.js:267:23)
gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:809:12)
gyp ERR! System Linux 4.1.17+
gyp ERR! command "nodejs" "/usr/bin/node-gyp" "build" "--fallback-to-build" "--module=/home/pi/Documents/NodeProjects/webservice/node_modules/serialport/build/Release/node-v11-linux-arm/serialport.node" "--module_name=serialport" "--module_path=/home/pi/Documents/NodeProjects/webservice/node_modules/serialport/build/Release/node-v11-linux-arm"
gyp ERR! cwd /home/pi/Documents/NodeProjects/webservice/node_modules/serialport
gyp ERR! node -v v0.10.29
gyp ERR! node-gyp -v v0.12.2
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute 'node-gyp build --fallback-to-build --module=/home/pi/Documents/NodeProjects/webservice/node_modules/serialport/build/Release/node-v11-linux-arm/serialport.node --module_name=serialport --module_path=/home/pi/Documents/NodeProjects/webservice/node_modules/serialport/build/Release/node-v11-linux-arm' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/home/pi/Documents/NodeProjects/webservice/node_modules/serialport/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
node-pre-gyp ERR! stack     at maybeClose (child_process.js:755:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:822:5)
node-pre-gyp ERR! System Linux 4.1.17+
node-pre-gyp ERR! command "node" "/home/pi/Documents/NodeProjects/webservice/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /home/pi/Documents/NodeProjects/webservice/node_modules/serialport
node-pre-gyp ERR! node -v v0.10.29
node-pre-gyp ERR! node-pre-gyp -v v0.6.18
node-pre-gyp ERR! not ok 
Failed to execute 'node-gyp build --fallback-to-build --module=/home/pi/Documents/NodeProjects/webservice/node_modules/serialport/build/Release/node-v11-linux-arm/serialport.node --module_name=serialport --module_path=/home/pi/Documents/NodeProjects/webservice/node_modules/serialport/build/Release/node-v11-linux-arm' (1)
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian
 
npm ERR! serialport@2.0.6 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@2.0.6 install script.
npm ERR! This is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! System Linux 4.1.17+
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install"
npm ERR! cwd /home/pi/Documents/NodeProjects/webservice
npm ERR! node -v v0.10.29
npm ERR! npm -v 1.4.21
npm ERR! code ELIFECYCLE
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /home/pi/Documents/NodeProjects/webservice/npm-debug.log
npm ERR! not ok code 0

jacobrosenthal, 2016-02-22:
`npm install serialport` installs the lastest serialport, 2.x.x
However 2.x.x in order to add node4 support now requires gcc 4.8.x
The vast majority of raspberry pi users are on debian 'wheezy' with gcc 4.7.x
Debian 'jessie' added gcc 4.8.x support
So you either need to upgrade to jessie to use the latest serialport
or you need to install the 1.x version of serialport with `npm i serialport@1.7.4`

closed 2016-02-22 by jacobrosenthal

fivdi, 2016-02-22:
@howardkitto see https://github.com/fivdi/onoff/wiki/Node.js-v0.10.29-and-native-addons-on-the-Raspberry-Pi

jacobrosenthal, 2016-02-22:
@fivdi Thanks. Ideas on how to present all this in the readme?

fivdi, 2016-02-25:
@jacobrosenthal I'll attempt to come up with something.

fivdi, 2016-02-27:
@jacobrosenthal after re-reading this comments in this issue I'm no longer sure what you'd like to present in the readme.

Above you say the following:

>npm install serialport installs the lastest serialport, 2.x.x
However 2.x.x in order to add node4 support now requires gcc 4.8.x
The vast majority of raspberry pi users are on debian 'wheezy' with gcc 4.7.x
Debian 'jessie' added gcc 4.8.x support
So you either need to upgrade to jessie to use the latest serialport
or you need to install the 1.x version of serialport with npm i serialport@1.7.4

However, this is only partially true.

serialport v2.x.x doesn't require gcc 4.8.x. What does require gcc 4.8.x is Node.js v4.0+. serialport v2.x.x should work fine with gcc 4.6.x on Wheezy, if Node.js v0.10.x or v0.12.x is being used. It's not essential to use serialport v1.7.4 on Wheezy if Node.js v0.10.x or v0.12.x is being used.

So what exactly should be presented in the readme?

fivdi, 2016-02-27:
What messes things up here is the fact that the v0.10.29 version of Node.js that comes pre-installed with Raspbian Jessie is broken and it's not possible to compile any addons that have been updated to function with Node.js v4+.

ntsuandih, 2016-02-28:
Hello world! I have the same issue... I'm on Jessie. Sharing the subfolder that contains serialport in node_modules would be the solution. Copy this folder into the project...

fivdi, 2016-02-28:
@ntsuandih I'm not sure I understand. Are you suggesting that serialport should contain pre-compiled binaries for the Raspberry Pi? So that no compilation is required when serialport is installed on the Raspberry Pi?

ntsuandih, 2016-02-29:
here the solution : change the su password  to access root then do : npm install serialport