serialport/node-serialport #675: installation failed under ubuntu 15.04 running on arm V71 hardware

ericdai, 2016-02-25:
i  try to install serialport package on ubuntu 15.04 running on quad-core ARM A7 cpu (allwinner H3)
but failed .this is the log ,how to process it.

this is log file http://pan.baidu.com/s/1jGVHo6E

reconbot, 2016-03-27:
Can you put the log in a [gist](http://gist.github.com)?

fasiha, 2016-04-04:
@reconbot I made you a copy at https://gist.github.com/fasiha/70be544807898e3797b2caa6e729c221

reconbot, 2016-04-04:
I was able to find 

```
8629 info lifecycle serialport@2.0.6~install: Failed to exec install script
```

Which makes me wonder if gcc is installed. You can take a look at https://github.com/voodootikigod/node-serialport#ubuntu-linux

If it was an arm issue we'd see compile failures, but we have no output that it event started the compile. For the record you don't need to turn on full verbosity to get the compiler errors.

closed 2016-05-16 by reconbot