serialport/node-serialport #676: Does not work with NW.js Runtime without tweaks

dm430, 2016-02-28:
This module does not work with NW.js without some tweaks since the runtime is different. Im assuming that it has to do with the python tool. I guess this isn't really a massive issue, however I thought I would point it out.

reconbot, 2016-04-07:
Can you try with `serialport@2.0.7-beta4`? I think we've solved the issues around this by upgrading nan, but I've only confirmed with electron, I haven't had many people using nw.js try it out yet. 

#733

reconbot, 2016-04-11:
`serialport@2.1.0` was just released which should play nice with [nw-gyp](http://docs.nwjs.io/en/latest/For%20Users/Advanced/Use%20Native%20Node%20Modules/#nw-gyp)

If you are still having issues feel free to reopen.

closed 2016-04-11 by reconbot