serialport/node-serialport #677: serialport reopen trouble on linux

TBagaviev, 2016-03-04:
Connected serial devices stop working after closing connection on port and reopen it again.

Simple demo code:
`var sp = require('serialport');

var serialDevice1 = new sp.SerialPort('/dev/ttyUSB0', {
    baudrate: 9600,
    parser: sp.parsers.readline('\n', 'utf8')
}, false);

serialDevice1.on('data', function (data) {
    console.log('serialDevice1', 'data!', data);
});

serialDevice1.on('error', function (error) {
    console.error('serialDevice1', 'error!', error);
});

serialDevice1.on('disconnect', function (error) {
    console.log('serialDevice1', 'disconnect!', error);
});

serialDevice1.on('close', function () {
    console.log('serialDevice1', 'close!');
});

var serialDevice2 = new sp.SerialPort('/dev/ttyUSB0', {
    baudrate: 9600,
    parser: sp.parsers.readline('\n', 'utf8')
}, false);

serialDevice2.on('data', function (data) {
    console.log('serialDevice2', 'data!', data);
});

serialDevice2.on('error', function (error) {
    console.error('serialDevice2', 'error!', error);
});

serialDevice2.on('disconnect', function (error) {
    console.log('serialDevice2', 'disconnect!', error);
});

serialDevice2.on('close', function () {
    console.log('serialDevice2', 'close!');
});

serialDevice1.open(function () {
    setTimeout(function () {
        serialDevice1.close(function () {
            setTimeout(function () {
                serialDevice2.open(function () {
                    setTimeout(function () {
                        serialDevice2.close(function () {
                            console.log('tests finished!');
                        })
                    }, 3000);
                })
            }, 3000);
        })
    }, 3000);
});`

Simple demo output:
`tb@tbpc:~/projects/simpledemos$ node test/devices_reopen_fail.js 
serialDevice1 data! 
serialDevice1 data! r------------------------------------------------------------------7
serialDevice1 data! | * * * * * *  ~ Test Device, N/A N/A N/A N/A N/A ~ * * * * * * *  |
serialDevice1 data! |------------------------------------------------------------------|
serialDevice1 data! | * * * * * * < Test Device for Basic Features Tests > * * * * * * |
serialDevice1 data! | * * * * * * <       Created for N/A Project        > * * * * * * |
serialDevice1 data! | * * * * * * <      Builded with ARDUINO NANO       > * * * * * * |
serialDevice1 data! | * * * * * * <            Version 0.0.0.1           > * * * * * * |
serialDevice1 data! L------------------------------------------------------------------/
serialDevice1 data! 
serialDevice1 data! 
serialDevice1 data! 
serialDevice1 close!
serialDevice2 close!
tests finished!
tb@tbpc:~/projects/simpledemos$ `

reconbot, 2016-04-06:
We fixed a ton of bugs in our latest beta specifically ones addressing this issue. 

We've also documented that calling `.close()` removes all event handlers. This is not a good design and will be changed in the next major release. However issues reopening ports should be solved.

https://github.com/voodootikigod/node-serialport/issues/733 Please try it out `serialport@2.0.7-beta3` and report back!

TBagaviev, 2016-04-08:
Confirmed! Its working now. Thanks.
```
tb@tbpc:~/projects/simpledemos$ node test/devices_reopen_fail.js 
serialDevice1 data! 
serialDevice1 data! r------------------------------------------------------------------7
serialDevice1 data! | * * * * * *  ~ Test Device, N/A N/A N/A N/A N/A ~ * * * * * * *  |
serialDevice1 data! |------------------------------------------------------------------|
serialDevice1 data! | * * * * * * < Test Device for Basic Features Tests > * * * * * * |
serialDevice1 data! | * * * * * * <       Created for N/A Project        > * * * * * * |
serialDevice1 data! | * * * * * * <      Builded with ARDUINO NANO       > * * * * * * |
serialDevice1 data! | * * * * * * <            Version 0.0.0.1           > * * * * * * |
serialDevice1 data! L------------------------------------------------------------------/
serialDevice1 data! 
serialDevice1 data! 
serialDevice1 data! 
serialDevice1 close!
serialDevice2 data! 
serialDevice2 data! r------------------------------------------------------------------7
serialDevice2 data! | * * * * * *  ~ Test Device, N/A N/A N/A N/A N/A ~ * * * * * * *  |
serialDevice2 data! |------------------------------------------------------------------|
serialDevice2 data! | * * * * * * < Test Device for Basic Features Tests > * * * * * * |
serialDevice2 data! | * * * * * * <       Created for N/A Project        > * * * * * * |
serialDevice2 data! | * * * * * * <      Builded with ARDUINO NANO       > * * * * * * |
serialDevice2 data! | * * * * * * <            Version 0.0.0.1           > * * * * * * |
serialDevice2 data! L------------------------------------------------------------------/
serialDevice2 data! 
serialDevice2 data! 
serialDevice2 data! 
serialDevice2 close!
tests finished!
tb@tbpc:~/projects/simpledemos$ 
```

closed 2016-04-08 by TBagaviev