serialport/node-serialport #678: Serialport.list failure on openwrt (/bin/sh: /sbin/udevadm: not found)

rwaldron, 2016-03-11:
Re: https://github.com/voodootikigod/node-serialport/blob/c670840c03ffdce56f52aa65ee0be8ee46313403/serialport.js#L586

Running the following from a Tessel 2: 

```js
var sp = require("serialport");

sp.list((error, ports) => {
  if (error) {
    console.log(error);
    return;
  }

  console.log(ports);
});
```

```
{ [Error: Command failed: /bin/sh -c /sbin/udevadm info --query=property -p $(/sbin/udevadm info -q path -n /dev/ttyS0)
/bin/sh: /sbin/udevadm: not found
/bin/sh: /sbin/udevadm: not found
]
  killed: false,
  code: 127,
  signal: null,
  cmd: '/bin/sh -c /sbin/udevadm info --query=property -p $(/sbin/udevadm info -q path -n /dev/ttyS0)' }
```

I'll keep digging, maybe I can find some useful info to help out with this.

rwaldron, 2016-03-30:
> The t2 doesn't seem to ship with it, but openwrt does.

If `udevadm` doesn't exist, then we need to make it exist.

rwaldron, 2016-03-30:
This is the output after installing `udev`

```
[ { comName: 'ttyS0',
    manufacturer: undefined,
    serialNumber: undefined,
    pnpId: undefined,
    vendorId: '0xundefined',
    productId: '0xundefined' },
  { comName: 'ttyS1',
    manufacturer: undefined,
    serialNumber: undefined,
    pnpId: undefined,
    vendorId: '0xundefined',
    productId: '0xundefined' } ]
```

reconbot, 2016-03-30:
That's about right, fixed the weird 0x thing in an open PR

rwaldron, 2016-03-30:
That was with nothing attached. This is with something attached: 

```
[ { comName: 'ttyACM0',
    manufacturer: undefined,
    serialNumber: undefined,
    pnpId: undefined,
    vendorId: '0xundefined',
    productId: '0xundefined' },
  { comName: 'ttyS0',
    manufacturer: undefined,
    serialNumber: undefined,
    pnpId: undefined,
    vendorId: '0xundefined',
    productId: '0xundefined' },
  { comName: 'ttyS1',
    manufacturer: undefined,
    serialNumber: undefined,
    pnpId: undefined,
    vendorId: '0xundefined',
    productId: '0xundefined' } ]
```

reconbot, 2016-04-06:
This should look a bit better with the latest beta.

closed 2016-04-06 by reconbot

bbeers, 2016-08-26:
I am running into the same issue.  Any instructions on getting udev on the t2?

reconbot, 2016-08-26:
Mind creating a new issue for this?