serialport/node-serialport #679: continuously writing to serial port returns results -1 with a USB to serial adapter

ruizeng, 2016-03-14:
I'm using a ch430 usb to serial adapter to connect to my MCU through serial port. Below is my environment:
* OS: mac and ubuntu
* ARCH: x86-64
* node-serial version: 2.0.6
* node version: 4.2.*

my test code:

``` node
var SerialPort = require("serialport").SerialPort
var serialPort = new SerialPort("/dev/ttyUSB0", {
  baudrate: 115200
});

function writeCB(err, results) {
  console.log('err ' + err);
  console.log('write ' + results);
}

serialPort.on("open", function() {
  serialPort.on('data', function(data) {
    console.log('data received: ' + data);
  });

  console.log('open');

  setInterval(function() {
    serialPort.write(new Buffer([0xf0, 0xff, 0x01, 0x00, 0x01, 0xf7]), writeCB);
  }, 1);
});
```

And my MCU will also continuously send data to serial port.

The test program works well in a few seconds, but then the write callback 'writeCB' returns results -1 with no error. The MCU receives nothing, either. I have to change my test environment, and the result is here:

* I wrote the same test program in python(with `pyserial`), and it works perfectly well.  
* I tested the node program in RPI, it works well too.
* I tested it in WINDOWS 7, works well.
* I changed my adapter to `CP2102`, it came out the same problem as `CH430`

So I guess maybe there are bugs in unix/linux environment with usb to serial adapter? I wonder if any one faces the same problem with me.

reconbot, 2016-04-08:
What returns -1?

Our latest beta #733 exposes a bunch more errors that might help debug this problem. It also cleans up a lot of issues for osx and linux. Can you give it a try? `serialport@2.0.7-beta4`

fivdi, 2016-05-02:
> What returns -1?

The second argument passed to the `write` callback can be -1.

This issue can be reproduced with the program below on a Raspberry Pi 2. To reproduce the error, a relatively large amount of data needs to be sent, 5000 bytes in this case. This results in the write callback being called many times, 2664 times in this case. The write callback should be a completion callback that's only called once.

```js
var SerialPort = require('serialport').SerialPort,
  uart = new SerialPort('/dev/ttyAMA0', {baudRate: 9600});

uart.once('open', function () {
  var count = 0;

  uart.write(new Buffer(new Array(5000+1).join('a')), function (err, bytesWritten) {
    count += 1;
    console.log('count: ' + count + ', bytesWritten: ' + bytesWritten);
  });
});
```
Here's is the typical output of a test run:
```
count: 1, bytesWritten: -1
count: 2, bytesWritten: -1
count: 3, bytesWritten: -1
...
count: 2662, bytesWritten: -1
count: 2663, bytesWritten: -1
count: 2664, bytesWritten: 7
```

farfromrefug, 2016-05-04:
I have the exact same issue.
I try to send an image through serialport
I use ```serialport``` on both sides (rpi and OSX)

@reconbot How would you send big data? Could you share an example.
Might the issue be with the read buffer?

reconbot, 2016-05-05:
This is a problem with our write callback. This would cause high overhead
needlessly.

reconbot, 2016-05-16:
@farfromrefug @fivdi I've got a fix for the callback issue in #808 should save a lot of cpu when sending large amounts of data if you wouldn't mind testing it I'd be very grateful. 

You can test this branch by changing your package.json to include {"serialport": "voodootikigod/node-serialport#write-bindings"}

@ruizeng I don't think we've addressed your issue. The `-1` means the write is buffered, this might indicate a few things, the data isn't being written, it's taking a long time to write, or something else. If you could test the branch too, it might tell us more, eg, the write callback wont get called, or it takes for ever to get called.

fivdi, 2016-05-16:
@reconbot the branch installs successfully, the write completion callback is now only called once (which is correct,) but the `bytesWritten` argument passed to the completion callback is `undefined.` The 5000 bytes of data were successfully sent.

```
pi@raspberrypi:~/serialport $ npm install "git://github.com/voodootikigod/node-serialport.git#write-bindings"

> serialport@3.2.0-beta1 install /home/pi/serialport/node_modules/serialport
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download: https://github.com/voodootikigod/node-serialport/releases/download/3.2.0-beta1/serialport-v3.2.0-beta1-node-v48-linux-arm.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for serialport@3.2.0-beta1 and node@6.1.0 (node-v48 ABI) (falling back to source compile with node-gyp) 
make: Entering directory '/home/pi/serialport/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
  CXX(target) Release/obj.target/serialport/src/serialport_unix.o
  CXX(target) Release/obj.target/serialport/src/serialport_poller.o
  SOLINK_MODULE(target) Release/obj.target/serialport.node
  COPY Release/serialport.node
make: Leaving directory '/home/pi/serialport/node_modules/serialport/build'
/home/pi/serialport
└─┬ serialport@3.2.0-beta1  (git://github.com/voodootikigod/node-serialport.git#af7e40822a392653b2ff33a05b52fb3963deeae1)
  ├── bindings@1.2.1 
  ├─┬ commander@2.9.0 
  │ └── graceful-readlink@1.0.1 
  ├─┬ debug@2.2.0 
  │ └── ms@0.7.1 
  ├── es6-promise@3.2.1 
  ├── nan@2.3.3 
  ├─┬ node-pre-gyp@0.6.28 
  │ ├─┬ mkdirp@0.5.1 
  │ │ └── minimist@0.0.8 
  │ ├─┬ nopt@3.0.6 
  │ │ └── abbrev@1.0.7 
  │ ├─┬ npmlog@2.0.3 
  │ │ ├── ansi@0.3.1 
  │ │ ├─┬ are-we-there-yet@1.1.2 
  │ │ │ ├── delegates@1.0.0 
  │ │ │ └── readable-stream@2.1.2 
  │ │ └─┬ gauge@1.2.7 
  │ │   ├── has-unicode@2.0.0 
  │ │   ├─┬ lodash.pad@4.4.0 
  │ │   │ ├── lodash._baseslice@4.0.0 
  │ │   │ ├── lodash._basetostring@4.12.0 
  │ │   │ └── lodash.tostring@4.1.3 
  │ │   ├── lodash.padend@4.5.0 
  │ │   └── lodash.padstart@4.5.0 
  │ ├─┬ rc@1.1.6 
  │ │ ├── deep-extend@0.4.1 
  │ │ ├── ini@1.3.4 
  │ │ ├── minimist@1.2.0 
  │ │ └── strip-json-comments@1.0.4 
  │ ├─┬ request@2.72.0 
  │ │ ├── aws-sign2@0.6.0 
  │ │ ├── aws4@1.4.1 
  │ │ ├─┬ bl@1.1.2 
  │ │ │ └── readable-stream@2.0.6 
  │ │ ├── caseless@0.11.0 
  │ │ ├─┬ combined-stream@1.0.5 
  │ │ │ └── delayed-stream@1.0.0 
  │ │ ├── extend@3.0.0 
  │ │ ├── forever-agent@0.6.1 
  │ │ ├─┬ form-data@1.0.0-rc4 
  │ │ │ └── async@1.5.2 
  │ │ ├─┬ har-validator@2.0.6 
  │ │ │ ├─┬ chalk@1.1.3 
  │ │ │ │ ├── ansi-styles@2.2.1 
  │ │ │ │ ├── escape-string-regexp@1.0.5 
  │ │ │ │ ├─┬ has-ansi@2.0.0 
  │ │ │ │ │ └── ansi-regex@2.0.0 
  │ │ │ │ ├── strip-ansi@3.0.1 
  │ │ │ │ └── supports-color@2.0.0 
  │ │ │ ├─┬ is-my-json-valid@2.13.1 
  │ │ │ │ ├── generate-function@2.0.0 
  │ │ │ │ ├─┬ generate-object-property@1.2.0 
  │ │ │ │ │ └── is-property@1.0.2 
  │ │ │ │ ├── jsonpointer@2.0.0 
  │ │ │ │ └── xtend@4.0.1 
  │ │ │ └─┬ pinkie-promise@2.0.1 
  │ │ │   └── pinkie@2.0.4 
  │ │ ├─┬ hawk@3.1.3 
  │ │ │ ├── boom@2.10.1 
  │ │ │ ├── cryptiles@2.0.5 
  │ │ │ ├── hoek@2.16.3 
  │ │ │ └── sntp@1.0.9 
  │ │ ├─┬ http-signature@1.1.1 
  │ │ │ ├── assert-plus@0.2.0 
  │ │ │ ├─┬ jsprim@1.2.2 
  │ │ │ │ ├── extsprintf@1.0.2 
  │ │ │ │ ├── json-schema@0.2.2 
  │ │ │ │ └── verror@1.3.6 
  │ │ │ └─┬ sshpk@1.8.3 
  │ │ │   ├── asn1@0.2.3 
  │ │ │   ├── assert-plus@1.0.0 
  │ │ │   ├─┬ dashdash@1.13.1 
  │ │ │   │ └── assert-plus@1.0.0 
  │ │ │   ├── ecc-jsbn@0.1.1 
  │ │ │   ├─┬ getpass@0.1.6 
  │ │ │   │ └── assert-plus@1.0.0 
  │ │ │   ├── jodid25519@1.0.2 
  │ │ │   ├── jsbn@0.1.0 
  │ │ │   └── tweetnacl@0.13.3 
  │ │ ├── is-typedarray@1.0.0 
  │ │ ├── isstream@0.1.2 
  │ │ ├── json-stringify-safe@5.0.1 
  │ │ ├─┬ mime-types@2.1.11 
  │ │ │ └── mime-db@1.23.0 
  │ │ ├── node-uuid@1.4.7 
  │ │ ├── oauth-sign@0.8.2 
  │ │ ├── qs@6.1.0 
  │ │ ├── stringstream@0.0.5 
  │ │ ├── tough-cookie@2.2.2 
  │ │ └── tunnel-agent@0.4.3 
  │ ├─┬ rimraf@2.5.2 
  │ │ └─┬ glob@7.0.3 
  │ │   ├── inflight@1.0.4 
  │ │   ├─┬ minimatch@3.0.0 
  │ │   │ └─┬ brace-expansion@1.1.4 
  │ │   │   ├── balanced-match@0.4.1 
  │ │   │   └── concat-map@0.0.1 
  │ │   └── path-is-absolute@1.0.0 
  │ ├── semver@5.1.0 
  │ ├─┬ tar@2.2.1 
  │ │ ├── block-stream@0.0.9 
  │ │ ├─┬ fstream@1.0.9 
  │ │ │ └── graceful-fs@4.1.4 
  │ │ └── inherits@2.0.1 
  │ └─┬ tar-pack@3.1.3 
  │   ├── fstream-ignore@1.0.4 
  │   ├─┬ once@1.3.3 
  │   │ └── wrappy@1.0.1 
  │   ├─┬ readable-stream@2.0.6 
  │   │ ├── core-util-is@1.0.2 
  │   │ ├── isarray@1.0.0 
  │   │ ├── process-nextick-args@1.0.7 
  │   │ ├── string_decoder@0.10.31 
  │   │ └── util-deprecate@1.0.2 
  │   └── uid-number@0.0.6 
  └─┬ object.assign@4.0.3 
    ├─┬ define-properties@1.1.2 
    │ └── foreach@2.0.5 
    ├── function-bind@1.1.0 
    └── object-keys@1.0.9 

npm WARN enoent ENOENT: no such file or directory, open '/home/pi/serialport/package.json'
npm WARN serialport No description
npm WARN serialport No repository field.
npm WARN serialport No README data
npm WARN serialport No license field.
pi@raspberrypi:~/serialport $ 
```
```
pi@raspberrypi:~/serialport $ node -v
v6.1.0
pi@raspberrypi:~/serialport $ cat test.js 
var SerialPort = require('serialport').SerialPort,
  uart = new SerialPort('/dev/ttyAMA0', {baudRate: 9600});

uart.once('open', function () {
  var count = 0;

  uart.write(new Buffer(new Array(5000+1).join('a')), function (err, bytesWritten) {
    count += 1;
    console.log('count: ' + count);
    console.log('err: ' + err);
    console.log('bytesWritten: ' + bytesWritten);
  });
});
pi@raspberrypi:~/serialport $ node test.js 
count: 1
err: null
bytesWritten: undefined
```

farfromrefug, 2016-05-16:
@fivdi yes there is no more 'sent' argument to the write callback

farfromrefug, 2016-05-16:
@reconbot still getting high cpu usage on OSX. I am sending a 2.7mo image.

ruizeng, 2016-05-18:
@reconbot I have upgraded the package to the newest version and the issue never happen again. Seem like your fix works well.

reconbot, 2016-05-18:
@fivdi I've removed the bytes written parameter, it will always be all the bytes

fivdi, 2016-05-18:
@reconbot Ah, yes, of course `bytesWritten` should be removed. Another three references to it are removed with https://github.com/voodootikigod/node-serialport/pull/814.

fivdi, 2016-05-18:
@reconbot Will the removal of `bytesWritten` be a semver-major change? There may be people using it.

reconbot, 2016-05-18:
Yeah it will

---
Francis Gulotta
wizard@roborooter.com

closed 2016-05-23 by reconbot