serialport/node-serialport #680: when electron > 0.37.0, build failed, in windows

sinfere, 2016-03-17:
after days digging, I found when electron > 0.37.0, build failed, in windows,

in electron 0.37.0's change log, it says: Upgrade to Chrome 49. I guess that, is nan not compatible with Chrome 49? 

logs: 
```
$ node-gyp rebuild --target=0.37.0 --dist-url=https://atom.io/download/atom-shell
gyp info it worked if it ends with ok
gyp info using node-gyp@3.3.1
gyp info using node@5.8.0 | win32 | x64
gyp http GET https://atom.io/download/atom-shell/v0.37.0/node-v0.37.0.tar.gz
gyp http 200 https://atom.io/download/atom-shell/v0.37.0/node-v0.37.0.tar.gz
gyp http GET https://atom.io/download/atom-shell/v0.37.0/SHASUMS256.txt
gyp http GET https://atom.io/download/atom-shell/v0.37.0/node.lib
gyp http GET https://atom.io/download/atom-shell/v0.37.0/x64/node.lib
gyp http 200 https://atom.io/download/atom-shell/v0.37.0/SHASUMS256.txt
gyp http 200 https://atom.io/download/atom-shell/v0.37.0/node.lib
gyp http 200 https://atom.io/download/atom-shell/v0.37.0/x64/node.lib
gyp info spawn C:\Python27\python.EXE
gyp info spawn args [ 'C:\\Users\\dd\\AppData\\Roaming\\npm\\node_modules\\node-gyp\\gyp\\gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'msvs',
gyp info spawn args   '-G',
gyp info spawn args   'msvs_version=auto',
gyp info spawn args   '-I',
gyp info spawn args   'C:\\Users\\dd\\Desktop\\pro\\serialport\\build\\config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   'C:\\Users\\dd\\AppData\\Roaming\\npm\\node_modules\\node-gyp\\addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   'C:\\Users\\dd\\.node-gyp\\0.37.0\\common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
gyp info spawn args   '-Dnode_root_dir=C:\\Users\\dd\\.node-gyp\\0.37.0',
gyp info spawn args   '-Dnode_gyp_dir=C:\\Users\\dd\\AppData\\Roaming\\npm\\node_modules\\node-gyp',
gyp info spawn args   '-Dnode_lib_file=node.lib',
gyp info spawn args   '-Dmodule_root_dir=C:\\Users\\dd\\Desktop\\pro\\serialport',
gyp info spawn args   '--depth=.',
gyp info spawn args   '--no-parallel',
gyp info spawn args   '--generator-output',
gyp info spawn args   'C:\\Users\\dd\\Desktop\\pro\\serialport\\build',
gyp info spawn args   '-Goutput_dir=.' ]
gyp info spawn C:\Program Files (x86)\MSBuild\12.0\bin\msbuild.exe
gyp info spawn args [ 'build/binding.sln',
gyp info spawn args   '/clp:Verbosity=minimal',
gyp info spawn args   '/nologo',
gyp info spawn args   '/p:Configuration=Release;Platform=x64' ]
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
  serialport.cpp
  serialport_win.cpp
  enumser.cpp
..\src\win\enumser.cpp : warning C4819: The file contains a character that cannot be represented in the current code page (936)                  . Save the file in Unicode format to prevent data loss [C:\Users\dd\Desktop\pro\serialport\build\serialport.vcxproj]
..\src\win\enumser.cpp(443): warning C4996: 'GetVersionExA': was declared deprecated [C:\Users\dd\Desktop\pro\serialport\build\                  serialport.vcxproj]
          C:\Program Files (x86)\Windows Kits\8.1\Include\um\sysinfoapi.h(433) : see declaration of 'GetVersionExA'
C:\Users\dd\Desktop\pro\serialport\node_modules\nan\nan.h(590): error C2039: 'GCEpilogueCallback' : is not a member of 'v8::Iso                  late' (..\src\serialport_win.cpp) [C:\Users\dd\Desktop\pro\serialport\build\serialport.vcxproj]
          C:\Users\dd\.node-gyp\0.37.0\deps\v8\include\v8.h(5281) : see declaration of 'v8::Isolate'
C:\Users\dd\Desktop\pro\serialport\node_modules\nan\nan.h(590): error C2065: 'GCEpilogueCallback' : undeclared identifier (..\s                  rc\serialport_win.cpp) [C:\Users\dd\Desktop\pro\serialport\build\serialport.vcxproj]
C:\Users\dd\Desktop\pro\serialport\node_modules\nan\nan.h(591): error C2146: syntax error : missing ')' before identifier 'call                  back' (..\src\serialport_win.cpp) [C:\Users\dd\Desktop\pro\serialport\build\serialport.vcxproj]
C:\Users\dd\Desktop\pro\serialport\node_modules\nan\nan.h(591): error C2182: 'AddGCEpilogueCallback' : illegal use of type 'voi                  d' (..\src\serialport_win.cpp) [C:\Users\dd\Desktop\pro\serialport\build\serialport.vcxproj]
C:\Users\dd\Desktop\pro\serialport\node_modules\nan\nan.h(591): error C2433: 'AddGCEpilogueCallback' : '__forceinline' not perm                  itted on data declarations (..\src\serialport_win.cpp) [C:\Users\dd\Desktop\pro\serialport\build\serialport.vcxproj]
C:\Users\dd\Desktop\pro\serialport\node_modules\nan\nan.h(591): error C2059: syntax error : ')' (..\src\serialport_win.cpp) [C:                  \Users\dd\Desktop\pro\serialport\build\serialport.vcxproj]
C:\Users\dd\Desktop\pro\serialport\node_modules\nan\nan.h(591): error C2143: syntax error : missing ';' before '{' (..\src\seri                  alport_win.cpp) [C:\Users\dd\Desktop\pro\serialport\build\serialport.vcxproj]
C:\Users\dd\Desktop\pro\serialport\node_modules\nan\nan.h(591): error C2447: '{' : missing function header (old-style formal li                  st?) (..\src\serialport_win.cpp) [C:\Users\dd\Desktop\pro\serialport\build\serialport.vcxproj]
C:\Users\dd\Desktop\pro\serialport\node_modules\nan\nan.h(596): error C2039: 'GCEpilogueCallback' : is not a member of 'v8::Iso                  late' (..\src\serialport_win.cpp) [C:\Users\dd\Desktop\pro\serialport\build\serialport.vcxproj]
          C:\Users\dd\.node-gyp\0.37.0\deps\v8\include\v8.h(5281) : see declaration of 'v8::Isolate'
C:\Users\dd\Desktop\pro\serialport\node_modules\nan\nan.h(596): error C2065: 'GCEpilogueCallback' : undeclared identifier (..\s                  rc\serialport_win.cpp) [C:\Users\dd\Desktop\pro\serialport\build\serialport.vcxproj]
C:\Users\dd\Desktop\pro\serialport\node_modules\nan\nan.h(596): error C2146: syntax error : missing ')' before identifier 'call                  back' (..\src\serialport_win.cpp) [C:\Users\dd\Desktop\pro\serialport\build\serialport.vcxproj]
C:\Users\dd\Desktop\pro\serialport\node_modules\nan\nan.h(596): error C2182: 'RemoveGCEpilogueCallback' : illegal use of type '                  void' (..\src\serialport_win.cpp) [C:\Users\dd\Desktop\pro\serialport\build\serialport.vcxproj]
C:\Users\dd\Desktop\pro\serialport\node_modules\nan\nan.h(596): error C2433: 'RemoveGCEpilogueCallback' : '__forceinline' not p                  ermitted on data declarations (..\src\serialport_win.cpp) [C:\Users\dd\Desktop\pro\serialport\build\serialport.vcxproj]
C:\Users\dd\Desktop\pro\serialport\node_modules\nan\nan.h(596): error C2059: syntax error : ')' (..\src\serialport_win.cpp) [C:                  \Users\dd\Desktop\pro\serialport\build\serialport.vcxproj]
[... 47 lines cut ...]
C:\Users\dd\Desktop\pro\serialport\node_modules\nan\nan.h(607): error C2039: 'GCPrologueCallback' : is not a member of 'v8::Iso                  late' (..\src\serialport.cpp) [C:\Users\dd\Desktop\pro\serialport\build\serialport.vcxproj]
          C:\Users\dd\.node-gyp\0.37.0\deps\v8\include\v8.h(5281) : see declaration of 'v8::Isolate'
C:\Users\dd\Desktop\pro\serialport\node_modules\nan\nan.h(607): error C2065: 'GCPrologueCallback' : undeclared identifier (..\s                  rc\serialport.cpp) [C:\Users\dd\Desktop\pro\serialport\build\serialport.vcxproj]
C:\Users\dd\Desktop\pro\serialport\node_modules\nan\nan.h(607): error C2146: syntax error : missing ')' before identifier 'call                  back' (..\src\serialport.cpp) [C:\Users\dd\Desktop\pro\serialport\build\serialport.vcxproj]
C:\Users\dd\Desktop\pro\serialport\node_modules\nan\nan.h(607): error C2182: 'RemoveGCPrologueCallback' : illegal use of type '                  void' (..\src\serialport.cpp) [C:\Users\dd\Desktop\pro\serialport\build\serialport.vcxproj]
C:\Users\dd\Desktop\pro\serialport\node_modules\nan\nan.h(607): error C2433: 'RemoveGCPrologueCallback' : '__forceinline' not p                  ermitted on data declarations (..\src\serialport.cpp) [C:\Users\dd\Desktop\pro\serialport\build\serialport.vcxproj]
C:\Users\dd\Desktop\pro\serialport\node_modules\nan\nan.h(607): error C2059: syntax error : ')' (..\src\serialport.cpp) [C:\Use                  rs\dd\Desktop\pro\serialport\build\serialport.vcxproj]
C:\Users\dd\Desktop\pro\serialport\node_modules\nan\nan.h(607): error C2143: syntax error : missing ';' before '{' (..\src\seri                  alport.cpp) [C:\Users\dd\Desktop\pro\serialport\build\serialport.vcxproj]
[... cut ...]
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:204:12)
gyp ERR! System Windows_NT 6.1.7601
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\dd\\AppData\\Roaming\\npm\\node_modules\\node-gyp\\bin\\node                  -gyp.js" "rebuild" "--target=0.37.0" "--dist-url=https://atom.io/download/atom-shell"
gyp ERR! cwd C:\Users\dd\Desktop\pro\serialport
gyp ERR! node -v v5.8.0
gyp ERR! node-gyp -v v3.3.1
gyp ERR! not ok

```

related: https://github.com/nodejs/nan/issues/547

reconbot, 2016-03-29:
Is it ok to close this in favor of #538?

jacobq, 2016-03-29:
I suppose this issue ought to share resolution with #538, so it probably makes sense to close it. However, I will say that it was helpful to me when I was searching all around trying to figure out how to make my electron project build with node-serialport because if it weren't for this issue I would probably not have guessed that going back a few versions would magically fix it.

sinfere, 2016-03-30:
@jacobq yes, feel so confused, "after days digging"

sinfere, 2016-03-30:
I think this issue should not be closed @reconbot , expecting fixes, thanks!

reconbot, 2016-03-30:
Might be helped by #707

reconbot, 2016-04-06:
Well it's been merged, we released a new version that might have this fixed as we've upgraded `nan`. #733 Please try it out `serialport@2.0.7-beta4` and report back!

jacobq, 2016-04-06:
My application now builds successfully with electron 0.37.4 and serialport 2.0.7-beta4; thanks @reconbot @jacobrosenthal  and others for all your work!

sinfere, 2016-04-07:
thanks! it works! @reconbot

closed 2016-04-07 by sinfere