serialport/node-serialport #681: serialport module fails at build for win 7 x64

faytom, 2016-03-17:
The following result shown after I run build commands. However it fails at 

C:\Users\zouweijie\.nw-gyp\0.12.3\Release\nw.lib : fatal error LNK1106: 
------------------------------------------------------------------------------------------------------------------------------------
Can someone help with my issue?? Thanks in advance.

C:\node\salad\node_modules\serialport>node-pre-gyp configure --runtime=node-webk
it --target=0.12.3 --target_arch=ia32 --msvs_version=2010
node-pre-gyp info it worked if it ends with ok
node-pre-gyp info using node-pre-gyp@0.6.23
node-pre-gyp info using node@4.3.1 | win32 | ia32
gyp info it worked if it ends with ok
gyp info using nw-gyp@0.13.0
gyp info using node@4.3.1 | win32 | ia32
(node) child_process: options.customFds option is deprecated. Use options.stdio
instead.
gyp info spawn python
gyp info spawn args [ 'C:\\Users\\zouweijie\\AppData\\Roaming\\npm\\node_modules
\\nw-gyp\\gyp\\gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'msvs',
gyp info spawn args   '-G',
gyp info spawn args   'msvs_version=2010',
gyp info spawn args   '-I',
gyp info spawn args   'C:\\node\\salad\\node_modules\\serialport\\build\\config.
gypi',
gyp info spawn args   '-I',
gyp info spawn args   'C:\\Users\\zouweijie\\AppData\\Roaming\\npm\\node_modules
\\nw-gyp\\addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   'C:\\Users\\zouweijie\\.nw-gyp\\0.12.3\\common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
gyp info spawn args   '-Dnode_root_dir=C:\\Users\\zouweijie\\.nw-gyp\\0.12.3',
gyp info spawn args   '-Dmodule_root_dir=C:\\node\\salad\\node_modules\\serialpo
rt',
gyp info spawn args   '--depth=.',
gyp info spawn args   '-Dv13=0',
gyp info spawn args   '--generator-output',
gyp info spawn args   'C:\\node\\salad\\node_modules\\serialport\\build',
gyp info spawn args   '-Goutput_dir=.' ]
gyp info ok
node-pre-gyp info ok

C:\node\salad\node_modules\serialport>node-pre-gyp build --runtime=node-webkit -
-target=0.12.3 --target_arch=ia32 --msvs_version=2010
node-pre-gyp info it worked if it ends with ok
node-pre-gyp info using node-pre-gyp@0.6.23
node-pre-gyp info using node@4.3.1 | win32 | ia32
gyp info it worked if it ends with ok
gyp info using nw-gyp@0.13.0
gyp info using node@4.3.1 | win32 | ia32
(node) child_process: options.customFds option is deprecated. Use options.stdio
instead.
gyp info spawn C:\Windows\Microsoft.NET\Framework\v4.0.30319\msbuild.exe
gyp info spawn args [ 'build/binding.sln',
gyp info spawn args   '/clp:Verbosity=minimal',
gyp info spawn args   '/nologo',
gyp info spawn args   '/p:Configuration=Release;Platform=Win32' ]
C:\Users\zouweijie\.nw-gyp\0.12.3\Release\nw.lib : fatal error LNK1106: 文件无效
或磁盘
已满: 无法查找到 0x1CD70 [C:\node\salad\node_modules\serialport\build\serialport
.vcxpr
oj]
gyp ERR! build error
gyp ERR! stack Error: `C:\Windows\Microsoft.NET\Framework\v4.0.30319\msbuild.exe
` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Users\zouweijie\AppData\Roaming\np
m\node_modules\nw-gyp\lib\build.js:286:23)
gyp ERR! stack     at emitTwo (events.js:87:13)
gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_proces
s.js:200:12)
gyp ERR! System Windows_NT 6.1.7600
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\zouweijie\\Ap
pData\\Roaming\\npm\\node_modules\\nw-gyp\\bin\\nw-gyp.js" "build" "--runtime=no
de-webkit" "--target=0.12.3" "--target_arch=ia32" "--msvs_version=2010" "--modul
e=C:\\node\\salad\\node_modules\\serialport\\build\\Release\\node-webkit-v0.12.3
-win32-ia32\\serialport.node" "--module_name=serialport" "--module_path=C:\\node
\\salad\\node_modules\\serialport\\build\\Release\\node-webkit-v0.12.3-win32-ia3
2"
gyp ERR! cwd C:\node\salad\node_modules\serialport
gyp ERR! node -v v4.3.1
gyp ERR! nw-gyp -v v0.13.0
gyp ERR! not ok
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute 'nw-gyp.cmd build --runtime=nod
e-webkit --target=0.12.3 --target_arch=ia32 --msvs_version=2010 --module=C:\node
\salad\node_modules\serialport\build\Release\node-webkit-v0.12.3-win32-ia32\seri
alport.node --module_name=serialport --module_path=C:\node\salad\node_modules\se
rialport\build\Release\node-webkit-v0.12.3-win32-ia32' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (C:\Users\zouweijie\AppD
ata\Roaming\npm\node_modules\node-pre-gyp\lib\util\compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:87:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:821:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/chi
ld_process.js:211:5)
node-pre-gyp ERR! System Windows_NT 6.1.7600
node-pre-gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\zouw
eijie\\AppData\\Roaming\\npm\\node_modules\\node-pre-gyp\\bin\\node-pre-gyp" "bu
ild" "--runtime=node-webkit" "--target=0.12.3" "--target_arch=ia32" "--msvs_vers
ion=2010"
node-pre-gyp ERR! cwd C:\node\salad\node_modules\serialport
node-pre-gyp ERR! node -v v4.3.1
node-pre-gyp ERR! node-pre-gyp -v v0.6.23
node-pre-gyp ERR! not ok
Failed to execute 'nw-gyp.cmd build --runtime=node-webkit --target=0.12.3 --targ
et_arch=ia32 --msvs_version=2010 --module=C:\node\salad\node_modules\serialport\
build\Release\node-webkit-v0.12.3-win32-ia32\serialport.node --module_name=seria
lport --module_path=C:\node\salad\node_modules\serialport\build\Release\node-web
kit-v0.12.3-win32-ia32' (1)

reconbot, 2016-04-07:
Are you using NW.js? If so could you try our latest beta #733 `serialport@2.0.7-beta4`? I know it clears up issues with `electron` but I haven't seen much about NW.js yet.

reconbot, 2016-04-11:
This should be solved with `serialport@2.1.0` if you use `nw-gyp`. Please reopen if you're still having issues.

closed 2016-04-11 by reconbot

dfsq1311, 2016-08-20:
I encouter the error `fatal error LNK1106: ` too.

the issue I submit:https://github.com/EmergingTechnologyAdvisors/node-serialport/issues/912