serialport/node-serialport #682: serialport.node: __strcpy_chk: symbol not found on alpine linux

ArtworkAD, 2016-03-17:
I am using alpine linux 3.3 and node 4.3.

```
> serialport@2.0.6 install /app/node_modules/serialport
> node-pre-gyp install --fallback-to-build

[serialport] Success: "/app/node_modules/serialport/build/Release/node-v46-linux-x64/serialport.node" is installed via remote
serialport@2.0.6 node_modules/serialport
├── bindings@1.2.1
├── async@0.9.0
├── sf@0.1.7
├── debug@2.2.0 (ms@0.7.1)
├── nan@2.0.9
└── optimist@0.6.1 (wordwrap@0.0.3, minimist@0.0.10)
```

When I try to open a port I get an error:

```
npm ERR! Linux 3.19.0-30-generic
npm ERR! argv "/usr/bin/node" "/usr/bin/npm" "start"
npm ERR! node v4.3.0
npm ERR! npm  v2.14.12
npm ERR! code ELIFECYCLE
npm ERR! app@1.2.0 start: `node ./bin/www`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the app@1.2.0 start script 'node ./bin/www'.
npm ERR! This is most likely a problem with the app package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node ./bin/www
npm ERR! You can get their info via:
npm ERR!     npm owner ls app
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /app/app/npm-debug.log
module.js:435
  return process.dlopen(module, path._makeLong(filename));
                 ^

Error: Error relocating /app/app/node_modules/serialport/build/Release/node-v46-linux-x64/serialport.node: __strcpy_chk: symbol not found
    at Error (native)
    at Object.Module._extensions..node (module.js:435:18)
    at Module.load (module.js:344:32)
    at Function.Module._load (module.js:301:12)
    at Module.require (module.js:354:17)
    at require (internal/module.js:12:17)
    at Object.<anonymous> (/app/app/node_modules/serialport/serialport.js:14:25)
    at Module._compile (module.js:410:26)
    at Object.Module._extensions..js (module.js:417:10)
    at Module.load (module.js:344:32)
```

Everything worked fine with debian jessi and node 4.3. After switching to alpine linux I get the above error.

reconbot, 2016-03-20:
We've been cross building for arm chips on the tessel project. We don't have the latest beta but here's the latest version for arm.

https://packages.tessel.io/npm/serialport-2.0.6-Release.tgz

ArtworkAD, 2016-03-20:
@reconbot great! But how can I install it from https://packages.tessel.io/npm/serialport-2.0.6-Release.tgz?

reconbot, 2016-03-20:
Untar it and put it in your node modules folder! I don't think there's an
`npm` way to do it yet.

fivdi, 2016-03-21:
Is there some confusion here? It looks the reported error occurred on an Linux computer with an x86 64-bit processor (node-v46-linux-x64). The Tessel 2 however has a MIPS processor so the binaries at https://packages.tessel.io are most likely MIPS binaries, I think.

ArtworkAD, 2016-03-21:
@fivdi exactly. I use docker and alpine linux https://hub.docker.com/_/alpine/ as base image (based on busybox) because of its small size. node-serialport used to work with ubuntu/debian as base image but this creates very large images with around 600MB while with alpine I achieve size like 100MB. So I switched to alpine. Unfortunately node-serialport stopped working with the error message I pointed out.

It is installed on a server with Intel i7 processor.

reconbot, 2016-03-21:
You're right I'm totally confused. I assumed you were running alpine on an arm platform single board computer.

ArtworkAD, 2016-03-21:
@reconbot I found an issue with a similar error message: https://github.com/spark/particle-cli/issues/145

> @dbrgn this looks like a known issue with node-serialport's incompatibility with node 4.x, which is nearly resolved by the looks of it. In the mean time please see this thread on our community forums

mhart, 2016-03-30:
I believe the error's saying musl doesn't have this symbol (`__strcpy_chk`) – so it must be a glibc-specific symbol.

You can see some discussion around this on the musl mailing list: http://www.openwall.com/lists/musl/2015/06/17/1

But it doesn't look like it got merged/finished.

It does, however, look like you can use the patch to add `__strcpy_chk` yourself, because the author here managed to get `ksh93` built using their own `__strcpy_chk` implementation: http://www.openwall.com/lists/musl/2015/07/01/4

I think probably the best solution going forward would be to ping the musl folks again – it looks like that thread died, so I think it would just be a matter of picking it up again.

mhart, 2016-03-30:
(actually – just reading this again now – @ArtworkAD are you sure it *actually* got compiled?)

mhart, 2016-03-30:
So, when I try and build on Alpine with `npm install serialport --build-from-source=serialport`, I get:

```
> serialport@2.0.6 install /node_modules/serialport
> node-pre-gyp install --fallback-to-build

make: Entering directory '/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
  CXX(target) Release/obj.target/serialport/src/serialport_unix.o
../src/serialport_unix.cpp:33:26: fatal error: linux/serial.h: No such file or directory
 #include <linux/serial.h>
                          ^
compilation terminated.
serialport.target.mk:92: recipe for target 'Release/obj.target/serialport/src/serialport_unix.o' failed
make: *** [Release/obj.target/serialport/src/serialport_unix.o] Error 1
make: Leaving directory '/node_modules/serialport/build'
```

mhart, 2016-03-30:
Ah, ok, adding the `linux-headers` package on Alpine fixed that (`apk add --no-cache linux-headers`)

I've now compiled it – will let you know if it works!

mhart, 2016-03-30:
Ok, here's how far I got:

```
$ docker run -it --device=/dev/ttyS0 mhart/alpine-node sh
/ # apk add --no-cache make gcc g++ python linux-headers
...
/ # npm install serialport --build-from-source=serialport
...
```

All good so far. Then:
```
/ # node
> require("serialport").list((err, ports) => { if (err) return console.error(err); ports.forEach(console.log) })
> { [Error: Command failed: /sbin/udevadm info --query=property -p $(/sbin/udevadm info -q path -n /dev/ttyS0)
/bin/sh: /sbin/udevadm: not found
/bin/sh: /sbin/udevadm: not found
]
  killed: false,
  code: 127,
  signal: null,
  cmd: '/sbin/udevadm info --query=property -p $(/sbin/udevadm info -q path -n /dev/ttyS0)' }
```

So, `serialport` is relying on `udevadm` to exist, so I installed `udev` and tried again:

```
/ # apk add --no-cache udev
...
/ # node
> require("serialport").list((err, ports) => { if (err) return console.error(err); ports.forEach(console.log) })
> { comName: '/dev/ttyS0',
  manufacturer: undefined,
  serialNumber: undefined,
  pnpId: undefined,
  vendorId: '0xundefined',
  productId: '0xundefined' } 0 [ { comName: '/dev/ttyS0',
    manufacturer: undefined,
    serialNumber: undefined,
    pnpId: undefined,
    vendorId: '0xundefined',
    productId: '0xundefined' } ]
```

Success! As far as I can tell anyway...

mhart, 2016-03-30:
So, just to summarise, the steps to get it working on Alpine are (assuming node and npm are already installed):
```
$ apk add --no-cache make gcc g++ python linux-headers udev
$ npm install serialport --build-from-source=serialport
```

(`udev` I imagine is optional – I guess it's only if certain `serialport` functions rely on that)

ArtworkAD, 2016-03-30:
@mhart awesomee :dancer: thank you I just tried it and it works.

closed 2016-03-30 by ArtworkAD

reconbot, 2016-03-30:
@mhart I'd totally take a docs PR where we specify special cases install instructions for special operating systems https://github.com/voodootikigod/node-serialport#installation-special-cases

mhart, 2016-03-30:
Done yo: #710

rwaldron, 2016-03-30:
There are other issues filed re:  `/sbin/udevadm` https://github.com/voodootikigod/node-serialport/search?q=udevadm&type=Issues&utf8=%E2%9C%93 :\

I think that needs to be refactored to be more flexible re: udevadm