serialport/node-serialport #684: "There's no write queue for that file descriptor (write)!"

JonathonAshworth, 2016-03-23:
One of our users is getting this error consistently when using our application, and it appears to be getting thrown by node-serialport. Unfortunately they didn't give me too much information to go by, but looking at the log files, it appears they have multiple serial devices connected, whereas most of our users just have the one.

[image]

reconbot, 2016-03-28:
Does your program open and close the ports often?

JonathonAshworth, 2016-03-28:
Generally it'll open and close a port 4 times in a standard use case.

There's two "processes" the user can go through (downloading data off the device, and erasing the device), and for each one, the program attempts to open all the devices to check which ones are valid versions of our proprietary device, generating an array of {portName, valid} objects. Then if exactly one port has a valid device, it'll proceed to the next step which then opens and closes that device again.

The actual download and erase processes take half a minute, so a typical run will look something like this:

open-close-open..........(30 secs)........close

and usually the user will erase the device immediately after downloading, so it repeats:

open-close-open............close-open-close-open..............close

reconbot, 2016-03-29:
I bet this is part of the double open bug, might have been hit with the recently fixed race condition during reads. If you try the latest beta you might have better luck. The double open bug is on the chopping block for sometime in the next few weeks.

Try out `serialport@2.0.7-beta2` and let us know

reconbot, 2016-04-03:
@JonathonAshworth There's also a good chance that you're writing after the port is closing or closed. If you test with the latest master you'll get a proper error if you try to do that. We'll have a beta release with those changes published in the next few days.

reconbot, 2016-04-11:
You should upgrade to `serialport@2.1.0` and see if that helps! Worst case it will give better errors.

JonathonAshworth, 2016-04-13:
Thanks reconbot! I'm going to close the issue until I have time to test the latest version, and if the bug persists I'll re-open :)

closed 2016-04-13 by JonathonAshworth

Zensey, 2016-06-24:
I catched this bug on 2.0.6, even upgrade to serialport@2.1.0 didn't help me.

But I just found out what I was doing wrong:
while closing don't try to open until you get the callback from close

```
- port.close();
+ yield cb=> port.close(cb);

...

port = new SerialPort(path)
yield cb => process.nextTick(cb)
yield cb=> port.open(cb);

```

reconbot, 2016-06-24:
So no bug? Is also strongly consider upgrading to 3x as it's largely the same but fixes a lot of bugs. See our [update guide](https://github.com/EmergingTechnologyAdvisors/node-serialport/blob/master/UPGRADE_GUIDE.md#upgrading-from-2x-to-3x)

Zensey, 2016-06-24:
I think its a concurrency bug.
I noticed, that it occures only when OS returns the same fd-number for recently opened port equal to fd of port beeing closed.

reconbot, 2016-06-25:
Is it a problem if you wait for the callback from close? I think there might be some cleanup of the queues that happen after the callback but I'm pretty sure they're cleaned up before the next open even if get the same fd.

```
port = new SerialPort(path);
yield cb => port.on('open', cb);
yield cb => port.close(cb);
yield cb => port.open(cb);
yield cb => port.write("testing", cb);
```

Zensey, 2016-06-25:
> Is it a problem if you wait for the callback from close?
Its not a problem.
I just wasn't paying much attention to it

> Are you making a new port object or reusing the first one? 
not reusing, just making new

reconbot, 2016-06-25:
That makes sense to me. All the port objects share the same bindings, a single port alone wouldn't let you get that error, but making new ones wouldn't have the knowledge of the others to protect you from it.

reconbot, 2016-06-25:
In theory, you could close one port, and open another on another object, get the same FD and get this error. A perfectly valid use case and a race condition.