serialport/node-serialport #685: Raspberry Pi shuts down when trying to open serial port

dislick, 2016-03-24:
Hello

My Raspberry Pi 2 Model B (Raspian 4.1.19) shuts down (it won't come back up until I pull the power chord and plug it back in) when trying to open a serial port. Here is my code which is pretty much copied from the README:

```
var SerialPort = require("serialport").SerialPort
var serialPort = new SerialPort("/dev/ttyAMA0", {
  baudrate: 300,
  dataBits: 7,
  stopBits: 2,
  parity: 'even',
}, false); // this is the openImmediately flag [default is true]

serialPort.open(function (error) {
  if ( error ) {
    console.log('failed to open: '+error);
  } else {
    console.log('open');
    serialPort.on('data', function(data) {
      console.log('data received: ' + data);
    });
  }
});
```

I have a DCF77 antenna connected to the serial port. Does anyone know why this is happening?

Video demonstrating the shut down via ping: https://gfycat.com/BruisedGleamingArgentineruddyduck

reconbot, 2016-03-25:
If I had to guess it's your power supply or serial port wiring. When the port opens, there's more current draw. And if the wiring is an issue there might be a short. Pi's push the limits of most cell phone chargers.

Can you give more info on the hardware?

dislick, 2016-03-29:
Oops, I had to edit the `/boot/cmdline.txt` file as described [here](http://www.irrational.net/2012/04/19/using-the-raspberry-pis-serial-port/). Everything works fine now, sorry for the trouble!

closed 2016-03-29 by dislick