serialport/node-serialport #686: race condition closing serial port causes segfault

pabigot, 2016-03-24:
I'm using node-serialport on Linux under Node 4.4.0 on Ubuntu 14.04, in a highly asynchronous application involving ES6 promises.  I've encountered a situation where node aborts with core dump in uv__io_poll when epoll_ctl(2) returns EBADF because it was passed a descriptor for a closed serial port.

The hand-waving summary is that if SerialPort.close() is invoked while a read is active, the descriptor can be closed before a pending EAGAIN error is processed by afterRead.  This causes the poller to be restarted with a closed descriptor, which breaks libuv.

I'll submit a trivial patch that fixes the issue for me.

However, I'm also skeptical of the original code. The `fd` property is variously set to `null` or `0` when the port is closed, but the check for re-enabling the poller in that case is `self.fd >= 0`, so it's probably always going to pass.

I'd recommend consistently using `-1` as the value of the `fd` property when the port is closed as that value won't convert to something that looks like a valid descriptor.

reconbot, 2016-03-27:
This is a great catch, thank you.

reconbot, 2016-03-29:
fixed in serialport@2.0.7-beta2

reconbot, 2016-04-06:
I'm going to close this issue as I've pretty throughly tested this in linux on `2.0.7-beta4`. Thanks for raising and helping fix this issue.

closed 2016-04-06 by reconbot