serialport/node-serialport #688: Beta test serialport@2.0.7-beta1

reconbot, 2016-03-25:
We need confirmation that `serialport@2.0.7-beta1` works on the following platforms

 - [ ] Windows 7
 - [ ] Windows 8
 - [ ] Windows 10
 - [ ] OSX
 - [ ] Linux

Ideally with node 0.10, 0.12, 4 and 5.

rwaldron, 2016-03-28:
OSX: 

- installs successfully
- connects, disconnects successfully
- reading and writing successful

reconbot, 2016-03-29:
@rwaldron we just published `serialport@2.0.7-beta2` mind using that tomorrow?

fivdi, 2016-03-30:
Raspbian Jessie 2016-02-26:

Node.js v0.10.29, v4.4.1, v5.9.1

* installs successfully
* tested with the UART at /dev/ttyAMA0 so I'm unsure what connect and disconnect mean
* reading and writing successful

Node.js v6.0.0-nightly20160307061ebb39c9

`npm install serialport@2.0.7-beta2` results in compile errors:

```
pi@raspberrypi:~/serialport $ npm install serialport@2.0.7-beta2
(node:7165) fs: re-evaluating native module sources is not supported. If you are using the graceful-fs module, please update it to a more recent version.

> serialport@2.0.7-beta2 install /home/pi/serialport/node_modules/serialport
> node-pre-gyp install --fallback-to-build

make: Entering directory '/home/pi/serialport/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
In file included from ../src/serialport.h:5:0,
                 from ../src/serialport.cpp:3:
../../nan/nan.h:590:20: error: variable or field ‘AddGCEpilogueCallback’ declared void
       v8::Isolate::GCEpilogueCallback callback
                    ^
../../nan/nan.h:590:7: error: ‘GCEpilogueCallback’ is not a member of ‘v8::Isolate’
       v8::Isolate::GCEpilogueCallback callback
       ^
../../nan/nan.h:591:18: error: expected primary-expression before ‘gc_type_filter’
     , v8::GCType gc_type_filter = v8::kGCTypeAll) {
                  ^
../../nan/nan.h:596:20: error: variable or field ‘RemoveGCEpilogueCallback’ declared void
       v8::Isolate::GCEpilogueCallback callback) {
                    ^
../../nan/nan.h:596:7: error: ‘GCEpilogueCallback’ is not a member of ‘v8::Isolate’
       v8::Isolate::GCEpilogueCallback callback) {
       ^
../../nan/nan.h:601:20: error: variable or field ‘AddGCPrologueCallback’ declared void
       v8::Isolate::GCPrologueCallback callback
                    ^
../../nan/nan.h:601:7: error: ‘GCPrologueCallback’ is not a member of ‘v8::Isolate’
       v8::Isolate::GCPrologueCallback callback
       ^
../../nan/nan.h:602:18: error: expected primary-expression before ‘gc_type_filter’
     , v8::GCType gc_type_filter = v8::kGCTypeAll) {
                  ^
../../nan/nan.h:607:20: error: variable or field ‘RemoveGCPrologueCallback’ declared void
       v8::Isolate::GCPrologueCallback callback) {
                    ^
../../nan/nan.h:607:7: error: ‘GCPrologueCallback’ is not a member of ‘v8::Isolate’
       v8::Isolate::GCPrologueCallback callback) {
       ^
serialport.target.mk:90: recipe for target 'Release/obj.target/serialport/src/serialport.o' failed
make: *** [Release/obj.target/serialport/src/serialport.o] Error 1
make: Leaving directory '/home/pi/serialport/node_modules/serialport/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/home/pi/opt/node-v6.0.0-nightly20160307061ebb39c9-linux-armv7l/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:101:13)
gyp ERR! stack     at ChildProcess.emit (events.js:186:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:204:12)
gyp ERR! System Linux 4.1.18-v7+
gyp ERR! command "/home/pi/opt/node-v6.0.0-nightly20160307061ebb39c9-linux-armv7l/bin/node" "/home/pi/opt/node-v6.0.0-nightly20160307061ebb39c9-linux-armv7l/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/home/pi/serialport/node_modules/serialport/build/Release/node-v47-linux-arm/serialport.node" "--module_name=serialport" "--module_path=/home/pi/serialport/node_modules/serialport/build/Release/node-v47-linux-arm"
gyp ERR! cwd /home/pi/serialport/node_modules/serialport
gyp ERR! node -v v6.0.0-nightly20160307061ebb39c9
gyp ERR! node-gyp -v v3.2.1
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute '/home/pi/opt/node-v6.0.0-nightly20160307061ebb39c9-linux-armv7l/bin/node /home/pi/opt/node-v6.0.0-nightly20160307061ebb39c9-linux-armv7l/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/pi/serialport/node_modules/serialport/build/Release/node-v47-linux-arm/serialport.node --module_name=serialport --module_path=/home/pi/serialport/node_modules/serialport/build/Release/node-v47-linux-arm' (1)
[... 62 lines cut ...]
  TOUCH Release/obj.target/action_after_build.stamp
make: Leaving directory '/home/pi/serialport/node_modules/serialport/build'
/home/pi/serialport
└─┬ serialport@2.0.7-beta2  (git://github.com/fivdi/node-serialport.git#1e14890c47675c850ff5c6a9ccee9557c815dac6)
  ├── async@0.9.0 
  ├── bindings@1.2.1 
  ├─┬ debug@2.2.0 
  │ └── ms@0.7.1 
  ├── nan@2.2.1 

... 

  ├─┬ optimist@0.6.1 
  │ ├── minimist@0.0.10 
  │ └── wordwrap@0.0.3 
  └── sf@0.1.7 

npm WARN enoent ENOENT: no such file or directory, open '/home/pi/serialport/package.json'
npm WARN serialport No description
npm WARN serialport No repository field.
npm WARN serialport No README data
npm WARN serialport No license field.
npm WARN You are using a pre-release version of node and things may not work as expected
pi@raspberrypi:~/serialport $ 
```

reconbot, 2016-03-30:
Well this is the first node6 I've seen. Thanks!

fivdi, 2016-03-30:
It probably won't be the last v6 issue :)

reconbot, 2016-03-30:
Supporting it needs to be on the road map but it's behind a couple bugs.

reconbot, 2016-03-30:
That is to say, patches welcome.

fivdi, 2016-03-30:
#707 will fix the v6.0.0-nightly20160307061ebb39c9 issue. Note that #707 may also help with #680 which has similar compile errors.

closed 2016-04-06 by reconbot