serialport/node-serialport #689: Colud not use open method on custom baudrate (250000)

Heash, 2016-03-25:
I colud not use open method on custom baudrate (250000). The callback is working, device data is sending to device, but there is no answers. (I use Arduino and I see the blinking led when write data to serialport).
But if I use connecting from internal software the data via node serialport is reciving good even using 250000 baudrate.

reconbot, 2016-03-27:
Can you give us some more information?

What is `internal software`? What is your Arduino supposed to be sending? What operating system, node version and serialport version are you using?

Heash, 2016-03-27:
Yes, internal software is for example Pronterface. Or something another that tryes to connect Arduino. I use Ubuntu Linux 15.10, 64 bit. Arduino uses Marlin Firmware. Node version v0.10.37.

reconbot, 2016-03-28:
Great! We'll be doing this with the next release hopefully, I'm going to close your issue as a duplicate of #545 watch that issue for updates!

closed 2016-03-28 by reconbot