serialport/node-serialport #699: Won't install on Rasbpian; many errors

bskendig, 2016-03-27:
Operating System and Hardware: Raspberry Pi 2 with BrickPi board
NodeJS Version: 4.4.1
Serialport version: 1.6.3
Python Version: 2.7.3
(Run `node -v && npm -v && python --version` to get the version numbers.)

I have a Raspberry Pi 2 (running Raspbian Wheezy; fully "apt-get update"d) with a BrickPi board. I installed the latest NodeJS (4.4.1) for ARMv7, from "https://nodejs.org/dist/v4.4.1/node-v4.4.1-linux-armv7l.tar.xz".

I then ran "npm install serialport". It attempts to install version 1.6.3, which is several versions old, and it fails with numerous errors. I pasted the installation log at "https://gist.github.com/bskendig/589b3093ac15c74f614e".

fivdi, 2016-03-28:
Does the directory from where `npm install serialport` was called contain a `package.json` file that has serialport v1.6.3 as a dependency? If so, this would explain the attempt to install serialport v1.6.3.

In order to compile native addons with Node.js v4 or higher, g++ v4.8 or higher is required. g++ v4.8 doesn't appear to be installed here which would explain the compile errors. See [here](https://github.com/fivdi/onoff/wiki/Node.js-v4-and-native-addons) for further details. For Node.js v4 serialport v2 or higher is required.

fivdi, 2016-05-03:
@bskendig did you manage to install serialport?

bskendig, 2016-05-04:
No, not yet - I need to try again and try installing g++ 4.8 or higher. Do you know what package that's in, in Raspbian wheezy?

fivdi, 2016-05-04:
For a description of how to install g++ 4.8 on Raspbian Wheezy see https://github.com/fivdi/onoff/wiki/Node.js-v4-and-native-addons#installing-gccg-48-on-raspbian-wheezy-for-the-raspberry-pi

closed 2016-07-20 by reconbot