serialport/node-serialport #715: Unable to read serial port with node-serialport on raspberry pi

Ktulu2, 2016-04-01:
Hello!

I'm using a raspberry pi (the first model) running on Jessy (8), node v0.12.6 and serialport 2.0.6. I have connected the pin Rx on the pin Tx of the physical serial port.

It's working fine with `cat /dev/ttyAMA0` and `echo "Hello" > /dev/ttyAMA0`

The writing on the serial port with node-serialport is fine. I am using the code bellow (and using `cat` to read this) (source: https://www.npmjs.com/package/serialport)

```
var serialport = require("serialport");
var SerialPort = serialport.SerialPort;
var sp = new SerialPort("/dev/ttyAMA0", {
baudrate:9600,
databits: 8,
parity: 'none',
stopBits: 1,
flowControl: false,
parser: serialport.parsers.readline("\n"),
});

sp.on('open', function() {
console.log("sending");
sp.write("Hello");
});
```

I am now trying to read my serial port with node-serialport, but it doesn't work. When I am trying to read the serialport with node-serialport (and using `echo` to write on it), the data from echo are not writen in the terminal. The terminal only says "open". I am using this code, same source:

```
var serialport = require("serialport");
var SerialPort = serialport.SerialPort;
var sp = new SerialPort("/dev/ttyAMA0", {
baudrate:9600,
databits: 8,
parity: 'none',
stopBits: 1,
flowControl: false,
parser: serialport.parsers.readline("\n"),
});

sp.on('open', function() {
console.log('open');
sp.on('data', function(data) {
console.log('data received: ' + data);
});
});
```

I don't understand what's happening here. Any help would be much appreciated!

Thanks a lot! :)

Nicolas

fivdi, 2016-04-01:
Greetings @Ktulu2 

The exact same code worked for me on a Raspberry Pi 2

Terminal 1:
```
pi@raspberrypi:~/serialport $ cat test.js 
var serialport = require("serialport");
var SerialPort = serialport.SerialPort;
var sp = new SerialPort("/dev/ttyAMA0", {
baudrate:9600,
databits: 8,
parity: 'none',
stopBits: 1,
flowControl: false,
parser: serialport.parsers.readline("\n"),
});

sp.on('open', function() {
console.log('open');
sp.on('data', function(data) {
console.log('data received: ' + data);
});
});
pi@raspberrypi:~/serialport $ node test.js 
open
data received: hello1
data received: hello2
data received: hello3
data received: hello4
data received: hello5
```

Terminal 2:
```
pi@raspberrypi:/dev $ echo hello1 > ttyAMA0 
pi@raspberrypi:/dev $ echo hello2 > ttyAMA0 
pi@raspberrypi:/dev $ echo hello3 > ttyAMA0 
pi@raspberrypi:/dev $ echo hello4 > ttyAMA0 
pi@raspberrypi:/dev $ echo hello5 > ttyAMA0 
pi@raspberrypi:/dev $ 
```

Have shell and kernel messages on the serial connection been disabled with the raspi-config tool to prevent the kernel from using the serial port? (sudo raspi-config, Advanced-Options, Serial, No)

Ktulu2, 2016-04-01:
Hello @fivdi !
It wasn't disabled, it's working like a charm now!
Thank you very much for your answer :)

fivdi, 2016-04-01:
You're welcome.
If the issue has been resolved you can hit the `close issue` button at the bottom of the page :)

closed 2016-04-01 by Ktulu2

erikdung, 2017-01-19:
Hi, is it possible that Ubuntu prevent shell and kernel message too? I cannot read any data from my POS scale connected with RS232. How could I get the status about this in Ubuntu?

reconbot, 2017-01-19:
It would be possible but you'd have to check with your raspberry pi Ubuntu maintainers.

erikdung, 2017-01-19:
I have Ubuntu on normal PC, not Raspberry Pi. I have no success with reading data from POS(point of sale) scale. Any idea with this?

reconbot, 2017-01-19:
Please open a new issue with details of your program. And sample code.