serialport/node-serialport #721: listeners should not be removed on close

pabigot, 2016-04-03:
I don't expect this to be fixed since it'd be a behavior change, but for the record:

`SerialPort.prototype.close` [improperly](https://nodejs.org/dist/latest-v4.x/docs/api/events.html#events_emitter_removealllisteners_eventname) removes all listeners that the application might have registered with the instance.  Although the serial port can subsequently be re-opened with no difficulty and will re-use the configuration passed on construction the poor developer has to reconstruct the listeners and re-install them.

It'd probably be worth documenting that behavior.

rwaldron, 2016-04-03:
Duplicate https://github.com/voodootikigod/node-serialport/issues/638

rwaldron, 2016-04-03:
Also, everyone hates that it does this and @reconbot and I currently have our sights set to eliminate it

reconbot, 2016-04-03:
And also in disconnect, it shouldn't be there either

reconbot, 2016-04-03:
And it is documented https://github.com/voodootikigod/node-serialport/blob/master/README.md#close-callback

pabigot, 2016-04-03:
OK, I see it now.  I would have seen it before if it wasn't "hidden" down inside the discussion of the callback parameter.

I am glad to see there's serious effort going towards cleaning this module up.  Thanks.

reconbot, 2016-04-03:
agreed, it's poorly documented

closed 2016-04-03 by reconbot

reconbot, 2016-04-03:
@pabigot I made the docs a bit better around this situation

https://github.com/voodootikigod/node-serialport/blob/master/README.md#close-callback
https://github.com/voodootikigod/node-serialport/blob/master/README.md#ondisconect-callback

We'll stop doing that in a major release sometime soon.

And thank you for noticing and for being a part of it. =)