serialport/node-serialport #725: Serial Duplex Test with Arduino Echo with mismatched characters on multiple tests

peterwang998, 2016-04-03:
When I run serialDuplexTest.js along with ArduinoEcho, the sent character and received character do not match up. 

The mismatches are reproducible and repeats through the same cycle. 

Operating System and Hardware: Mac OS X 10.11.4, Arduino (test on Mega, Uno, and Leonardo)
NodeJS Version: 4.3.1
Serialport version: 2.0.6

### Expected behavior

### Actual behavior
`received: �
Byte count: 8169
Sent: *
received: :
Byte count: 8170
Sent: +
received: ;
Byte count: 8171
Sent: ,
received: �
Byte count: 8172
Sent: -
received: �
Byte count: 8173
Sent: .
received: �
Byte count: 8174
Sent: /
received: �
Byte count: 8175
Sent: 0
received: >
Byte count: 8176
Sent: 1
received: ?
Byte count: 8177
Sent: 2
received: 2
Byte count: 8178
Sent: 3
received: 3
Byte count: 8179
Sent: 4
received: 6
Byte count: 8180
Sent: 5
received: 7
Byte count: 8181
Sent: 6
received: 6
Byte count: 8182
Sent: 7
received: 7
Byte count: 8183
Sent: 8
received: >
Byte count: 8184
Sent: 9
received: ?
Byte count: 8185
Sent: :
received: :
Byte count: 8186
Sent: ;
received: ;
Byte count: 8187
Sent: <
received: >
Byte count: 8188
Sent: =
received: ?
Byte count: 8189
Sent: >
received: >
Byte count: 8190
Sent: ?
received: ?
Byte count: 8191
Sent: @
received: ~
Byte count: 8192
Sent: A
received: 
Byte count: 8193
Sent: B
received: r
Byte count: 8194
Sent: C
received: s
Byte count: 8195
Sent: D
received: f
Byte count: 8196
Sent: E
received: g
Byte count: 8197
Sent: F
received: f
Byte count: 8198
Sent: G
received: g
Byte count: 8199
Sent: H
received: N
Byte count: 8200
Sent: I
received: O
Byte count: 8201
Sent: J
received: z
Byte count: 8202
Sent: K
received: {
Byte count: 8203
Sent: L
received: N
Byte count: 8204
Sent: M
received: O
Byte count: 8205
Sent: N
received: N
Byte count: 8206
Sent: O
received: O
Byte count: 8207
Sent: P
received: �
Byte count: 8208
Sent: Q
received: �
Byte count: 8209
Sent: R
received: �
Byte count: 8210
Sent: S
received: �
Byte count: 8211
Sent: T
received: v
Byte count: 8212
Sent: U
received: w
Byte count: 8213
Sent: V
received: v
Byte count: 8214
Sent: W
received: w
Byte count: 8215
Sent: X
received: �
Byte count: 8216
Sent: Y
received: �
Byte count: 8217
Sent: Z
received: �
Byte count: 8218
Sent: [
received: �
Byte count: 8219
Sent: \
received: �
Byte count: 8220
Sent: ]
received: �
Byte count: 8221
Sent: ^
received: �
Byte count: 8222
Sent: _
received: �
Byte count: 8223
Sent: `
received: ~
Byte count: 8224
Sent: a
received: 
Byte count: 8225
Sent: b
received: r
Byte count: 8226
Sent: c
received: s
Byte count: 8227
Sent: d
received: f
Byte count: 8228
Sent: e
received: g
Byte count: 8229
Sent: f
received: f
Byte count: 8230
Sent: g
received: g
Byte count: 8231
Sent: h
received: n
Byte count: 8232
Sent: i
received: o
Byte count: 8233
Sent: j
received: z
Byte count: 8234
Sent: k
received: {
Byte count: 8235
Sent: l
received: n
Byte count: 8236
Sent: m
received: o
Byte count: 8237
Sent: n
received: n
Byte count: 8238
Sent: o
received: o
Byte count: 8239
Sent: p
received: ~
Byte count: 8240
Sent: q
received: 
Byte count: 8241
Sent: r
received: r
Byte count: 8242
Sent: s
received: s
Byte count: 8243
Sent: t
received: v
Byte count: 8244
Sent: u
received: w
Byte count: 8245
Sent: v
received: v
Byte count: 8246
Sent: w
received: w
Byte count: 8247
Sent: x
received: ~
Byte count: 8248
Sent: y
received: 
Byte count: 8249
Sent: z
received: z
Byte count: 8250
Sent: {
received: {
Byte count: 8251
Sent: |
received: ~
Byte count: 8252
Sent: }
received: 
Byte count: 8253
Sent: ~
received: ~
Byte count: 8254
Sent: 
received: 
Byte count: 8255
Sent:  
received: >
Byte count: 8256
Sent: !
received: ?
Byte count: 8257
Sent: "
received: 2
Byte count: 8258
Sent: #
received: 3
Byte count: 8259
Sent: $
received: &
Byte count: 8260
Sent: %
received: '
Byte count: 8261
Sent: &
received: &
Byte count: 8262
Sent: '
received: '
Byte count: 8263
Sent: (
received: �
Byte count: 8264
Sent: )
received: �
Byte count: 8265
Sent: *
received: :
Byte count: 8266
Sent: +
received: ;
Byte count: 8267
Sent: ,
received: �
Byte count: 8268
Sent: -
received: �
Byte count: 8269
Sent: .
received: �
Byte count: 8270
Sent: /
received: �
Byte count: 8271
Sent: 0
received: >
Byte count: 8272
Sent: 1
received: ?
Byte count: 8273
Sent: 2
received: 2
Byte count: 8274
Sent: 3
received: 3
Byte count: 8275
Sent: 4
received: 6
Byte count: 8276
Sent: 5
received: 7
Byte count: 8277
Sent: 6
received: 6
Byte count: 8278
Sent: 7
received: 7
Byte count: 8279
Sent: 8
received: >
Byte count: 8280
Sent: 9
received: ?
Byte count: 8281
Sent: :
received: :
Byte count: 8282
Sent: ;
received: ;
Byte count: 8283
Sent: <
received: >
Byte count: 8284
Sent: =
received: ?
Byte count: 8285
Sent: >
received: >
Byte count: 8286
Sent: ?
received: ?
Byte count: 8287
Sent: @
received: ~
Byte count: 8288
Sent: A
received: 
Byte count: 8289
Sent: B
received: r
Byte count: 8290
Sent: C
received: s
Byte count: 8291
Sent: D
received: f
Byte count: 8292
Sent: E
received: g
Byte count: 8293
Sent: F
received: f
Byte count: 8294
Sent: G
received: g
Byte count: 8295
Sent: H
received: N
Byte count: 8296
Sent: I
received: O
Byte count: 8297
Sent: J
received: z
Byte count: 8298
Sent: K
received: {
Byte count: 8299
Sent: L
received: N
Byte count: 8300
Sent: M
received: O
Byte count: 8301
Sent: N
received: N
Byte count: 8302
Sent: O
received: O
Byte count: 8303
Sent: P
received: �
Byte count: 8304
Sent: Q
received: �
Byte count: 8305
Sent: R
received: �
Byte count: 8306
Sent: S
received: �
Byte count: 8307
Sent: T
received: v
Byte count: 8308
Sent: U
received: w
Byte count: 8309
Sent: V
received: v
Byte count: 8310
Sent: W
received: w
Byte count: 8311
Sent: X
received: �
Byte count: 8312
Sent: Y
received: �
Byte count: 8313
Sent: Z
received: �
Byte count: 8314
Sent: [
received: �
Byte count: 8315
Sent: \
received: �
Byte count: 8316
Sent: ]
received: �
Byte count: 8317
Sent: ^
received: �
Byte count: 8318
Sent: _
received: �
Byte count: 8319
Sent: `
received: ~
Byte count: 8320
Sent: a
received: 
Byte count: 8321
Sent: b
received: r
Byte count: 8322
Sent: c
received: s
Byte count: 8323
Sent: d
received: f
Byte count: 8324
Sent: e
received: g
Byte count: 8325
Sent: f
received: f
Byte count: 8326
Sent: g
received: g
Byte count: 8327
Sent: h
received: n
Byte count: 8328
Sent: i
received: o
Byte count: 8329
Sent: j
received: z
Byte count: 8330
Sent: k
received: {
Byte count: 8331
Sent: l
received: n
Byte count: 8332
Sent: m
received: o
Byte count: 8333
Sent: n
received: n
Byte count: 8334
Sent: o
received: o
Byte count: 8335
Sent: p
received: ~
Byte count: 8336
Sent: q
received: 
Byte count: 8337
Sent: r
received: r
Byte count: 8338
Sent: s
received: s
Byte count: 8339
Sent: t
received: v
Byte count: 8340
Sent: u
received: w
Byte count: 8341
Sent: v
received: v
Byte count: 8342
Sent: w
received: w
Byte count: 8343
Sent: x
received: ~
Byte count: 8344
Sent: y
received: 
Byte count: 8345
Sent: z
received: z
Byte count: 8346
Sent: {
received: {
Byte count: 8347
Sent: |
received: ~
Byte count: 8348
Sent: }
received: 
Byte count: 8349
Sent: ~
received: ~
Byte count: 8350
Sent: 
received: 
Byte count: 8351
Sent:  
received: >
Byte count: 8352
Sent: !
received: ?
Byte count: 8353
Sent: "
received: 2
Byte count: 8354
Sent: #
received: 3
Byte count: 8355
Sent: $
received: &
Byte count: 8356
Sent: %
received: '
Byte count: 8357
Sent: &
received: &
Byte count: 8358
Sent: '
received: '
Byte count: 8359
Sent: (
received: �
Byte count: 8360
Sent: )
received: �
Byte count: 8361
Sent: *
received: :
Byte count: 8362
Sent: +
received: ;
Byte count: 8363
Sent: ,
received: �
Byte count: 8364
Sent: -
received: �
Byte count: 8365
Sent: .
received: �
Byte count: 8366
Sent: /
received: �
Byte count: 8367
Sent: 0
received: >
Byte count: 8368
Sent: 1
received: ?
Byte count: 8369
Sent: 2
received: 2
Byte count: 8370
Sent: 3
received: 3
Byte count: 8371`

### Steps and/or code to reproduce the issue
Copy ArduinoEcho.ino code and compile to Arduino
run node serialDuplexTest.js as instructed. 

This behavior was reproduced on two laptops and all versions of Arduino tested. (Uno, Mega, Leonardo).

reconbot, 2016-04-03:
I can't reproduce this issue on my setup (UNO and OSX). But I want to caution that the test isn't a perfect test. The arduino sends the string `READY` when it first connects which might be throwing off your count. You're also getting unprintable characters which shouldn't be happening. Are you reusing the same cable each time?

```
Sent:		s
Received:	s
Read Events:	10451
Sent:		t
Received:	t
Read Events:	10452
Sent:		u
Received:	u
Read Events:	10453
Sent:		v
Received:	v
Read Events:	10454
Sent:		w
Received:	w
Read Events:	10455
Sent:		x
Received:	x
Read Events:	10456
Sent:		y
Received:	y
Read Events:	10457
Sent:		z
Received:	z
Read Events:	10458
Sent:		{
Received:	{
Read Events:	10459
```

I pushed an updated version of the script that cleans up the output a little.

peterwang998, 2016-04-03:
I'll test the updated script when I get back to the lab early next week. But I did test it with a USB B and a micro USB cable (since I used different arduinos). It's certainly bazaar and I can't for the life of me figure out why it would produce the results.

closed 2016-05-01 by reconbot