serialport/node-serialport #735: Serialport.list has inconsistant representations for no data

reconbot, 2016-04-06:
Operating System and Hardware: all
NodeJS Version: all
Serialport version: 2.0.7-beta4

### Expected behavior
the same values for no data in the port description object

### Actual behavior
osx does empty string, linux does undefined, windows.. I don't know yet

### Steps and/or code to reproduce the issue
```js
require('serialport').list(console.log);
```

rwaldron, 2016-04-06:
Can we prefer `undefined`?

reconbot, 2016-04-06:
I'm worried fixing this is a breaking change worthy of a major version
bump. The problem is we never defined half of these things. I'll run a test
on all platforms to see how badly it deviates. The changes in the beta
already broke one person's code who worked around it. Thoughts?

Undefined seems right.

rwaldron, 2016-04-06:
I don't think I even care enough, TBH. I've never even hit this issue, likely because the code I write to handle inconsistent falsies is adequate.

reconbot, 2016-04-07:
Blocking #733

reconbot, 2016-04-08:
With `2.0.6`

# Windows
```
{
    comName: 'COM1',
    manufacturer: '(Standard port types)',
    serialNumber: '',
    pnpId: 'ACPI\\PNP0501\\1',
    locationId: '',
    vendorId: '',
    productId: '' 
}
```

# linux
```
{
    comName: 'ttyS0',
    manufacturer: undefined,
    serialNumber: undefined,
    pnpId: undefined,
    vendorId: '0xundefined',
    productId: '0xundefined' 
}
```

# osx
```
 { 
    comName: '/dev/cu.serial1',
    manufacturer: '',
    serialNumber: '',
    pnpId: '',
    locationId: '',
    vendorId: '',
    productId: '' 
}
```

And of course our readme has the empty string but that was introduced in the beta. Lets default to undefined. This might be breaking slightly breaking but it was buggy and undocumented and falsey checks will still continue to work. I can live with it.

PR #740 (2016-04-08): Make the output of .list undefined if it’s an empty string
 - Fixes #735 
 - Add some tests to start defining the bindings.

There has to be a better way to do the empty string checking.
closed 2016-04-08 by reconbot