serialport/node-serialport #741: Add docs to show off our new electron and nwjs support

reconbot, 2016-04-08:


pushtheworldllc, 2016-04-27:
+1 need help for sure

reconbot, 2016-05-23:
Here is the old nwjs issue https://github.com/EmergingTechnologyAdvisors/node-serialport/issues/374

chalkers, 2016-09-27:
I can tackle the Electron side. We use `serialport` in flasher.js.

reconbot, 2016-09-27:
Even a list of steps that I could try to follow would go a long way.

---
Francis Gulotta
wizard@roborooter.com

jbarguil, 2016-10-14:
+1 to that. I've commented on #538 something that worked for me (suggested by @reconbot , thanks again!)

bhaveshgohel, 2016-10-18:
@reconbot Pl check https://github.com/EmergingTechnologyAdvisors/node-serialport/issues/538#issuecomment-253756585

reconbot, 2017-07-10:
Electron rebuild still seems to work for most people closeing this issue. https://github.com/EmergingTechnologyAdvisors/node-serialport#electron

closed 2017-07-10 by reconbot