serialport/node-serialport #743: Serialport not worked when IRQ pins are same

kukdh1, 2016-04-08:
Operating System: Centos 6.7, CentOS 7 and Ubuntu server 15.10 (All x86_64)
Hardware: Small PC (http://www.dfitech.com/wp-content/uploads/2014/07/EC200-Product-Specs-20130220.pdf)
NodeJS Version: 5.10.1
Serialport version: 2.0.6 and 2.0.7-beta4
Python Version: 2.6 and 2.7

The EC 200 PC has 6 on-board serial ports. `dmesg | grep tty` result looks like:
```
00:02: ttyS0 at I/O 0x3f8 (irq = 4, base_baud = 115200) is a 16550A
00:03: ttyS1 at I/O 0x2f8 (irq = 3, base_baud = 115200) is a 16550A
00:04: ttyS2 at I/O 0x3e8 (irq = 11, base_baud = 115200) is a 16550A
00:05: ttyS3 at I/O 0x2e8 (irq = 10, base_baud = 115200) is a 16550A
00:06: ttyS4 at I/O 0x2f0 (irq = 11, base_baud = 115200) is a 16550A
00:07: ttyS5 at I/O 0x2e0 (irq = 10, base_baud = 115200) is a 16550A
```

As you can see, ttyS2 and ttyS4 shares IRQ pin and ttyS3 and ttyS5 shares IRQ pin.

I wrote example code for making error.
```javascript
'use strict'

// Load serialport module
const serialport = require('serialport')

// Open all 6 serial ports
const serial1 = new serialport.SerialPort('/dev/ttyS0', { baudrate: 9600 })
const serial2 = new serialport.SerialPort('/dev/ttyS1', { baudrate: 9600 })
const serial3 = new serialport.SerialPort('/dev/ttyS2', { baudrate: 9600 })
const serial4 = new serialport.SerialPort('/dev/ttyS3', { baudrate: 9600 })
const serial5 = new serialport.SerialPort('/dev/ttyS4', { baudrate: 9600 })
const serial6 = new serialport.SerialPort('/dev/ttyS5', { baudrate: 9600 })

// Close serialport when exit
process.on('exit', () => {
  if (serial1.isOpen()) {
    serial1.close()
  }

  if (serial2.isOpen()) {
    serial2.close()
  }

  if (serial3.isOpen()) {
    serial3.close()
  }

  if (serial4.isOpen()) {
    serial4.close()
  }

  if (serial5.isOpen()) {
    serial5.close()
  }

  if (serial6.isOpen()) {
    serial6.close()
  }
})

// Write 'test' to serial ports
// Arduino is connected to each serial ports. Arduino echo the message

serial1.on('open', () => {
  console.log('serial1 opened')
  serial1.write('test', (err, result) => {
    console.log('serial1: write finished res: ' + result)
    serial1.drain(err => {
      console.log('serial1: drain finished')
    })
  })
})

serial2.on('open', () => {
  console.log('serial2 opened')
  serial2.write('test', (err, result) => {
    console.log('serial2: write finished res: ' + result)
    serial2.drain(err => {
      console.log('serial2: drain finished')
    })
  })
})

serial3.on('open', () => {
  console.log('serial3 opened')
  serial3.write('test', (err, result) => {
    console.log('serial3: write finished res: ' + result)
    serial3.drain(err => {
      console.log('serial3: drain finished')
    })
  })
})

serial4.on('open', () => {
  console.log('serial4 opened')
  serial4.write('test', (err, result) => {
    console.log('serial4: write finished res: ' + result)
    serial4.drain(err => {
      console.log('serial4: drain finished')
    })
  })
})

serial5.on('open', () => {
  console.log('serial5 opened')
  serial5.write('test', (err, result) => {
    console.log('serial5: write finished res: ' + result)
    serial5.drain(err => {
      console.log('serial5: drain finished')
    })
  })
})

// Receive callback. It will receive 'test'

serial6.on('open', () => {
  console.log('serial6 opened')
})

serial1.on('data', (data) => {
  console.log('serial1 got data: ' + data)
})

serial2.on('data', (data) => {
  console.log('serial2 got data: ' + data)
})

serial3.on('data', (data) => {
  console.log('serial3 got data: ' + data)
})

serial4.on('data', (data) => {
  console.log('serial4 got data: ' + data)
})

serial5.on('data', (data) => {
  console.log('serial5 got data: ' + data)
})

serial6.on('data', (data) => {
  console.log('serial6 got data: ' + data)
})

```

When I run this on my machine, I got this log
```
serial1 opened
serial2 opened
serial3 opened
serial4 opened
serial5 opened
serial6 opened
serial1: write finished res: 4
serial2: write finished res: 4
serial3: write finished res: 4
serial4: write finished res: 4
serial5: write finished res: 4
serial1: drain finished
serial2: drain finished
serial5: drain finished
serial3: drain finished
serial1 got data: test
serial2 got data: test
serial3 got data: test
serial5 got data: test
```

As you can see, all serial ports are successfully opened and wrote data. But serial4 (/dev/ttyS3) could not finished drain.
Arduino received all data correctly and echo the message. And arduino receives 5 'test' messages. (drain callback of serial 4 was not called, but arduino received the message. Somtimes, arduino cannot received the message.)

I took all 3 days to solve this problem. And finally, I found this problem occur because IRQ pins are shared.

I use setserial command to disable irq on /dev/ttyS4 and /dev/ttyS5
IRQ pin 0 means 'timer channel 0' and driver will use polling method. (http://rampex.ihep.su/Linux/linux_howto/html/tutorials/Serial-HOWTO-6.html)
```
setserial /dev/ttyS4 irq 0
setserial /dev/ttyS5 irq 0
```
The result of `setserial -g /dev/ttyS*` changed:
```
/dev/ttyS0, UART: 16550A, Port: 0x03f8, IRQ: 4
/dev/ttyS1, UART: 16550A, Port: 0x02f8, IRQ: 3
/dev/ttyS2, UART: 16550A, Port: 0x03e8, IRQ: 11
/dev/ttyS3, UART: 16550A, Port: 0x02e8, IRQ: 10
/dev/ttyS4, UART: 16550A, Port: 0x02f0, IRQ: 0
/dev/ttyS5, UART: 16550A, Port: 0x02e0, IRQ: 0
```
When I run test code, All serial ports work very well.
```
serial1 opened
serial2 opened
serial3 opened
serial4 opened
serial6 opened
serial5 opened
serial1: write finished res: 4
serial2: write finished res: 4
serial3: write finished res: 4
serial4: write finished res: 4
serial5: write finished res: 4
serial1: drain finished
serial2: drain finished
serial3: drain finished
serial4: drain finished
serial5: drain finished
serial1 got data: test
serial2 got data: test
serial3 got data: test
serial4 got data: test
serial5 got data: test
```

Is this problem caused because of bug of node-serialport? Or the bug of Linux Kernel's IRQ shared pin handling method? I tested my code on Linux Kernel version 2.6, 3.10 and 4.5 but same situation caused.

reconbot, 2016-04-08:
I'm going to guess that it's a problem with the kernel, but there's a chance you're hitting a file descriptor issue. Could you test it with `serialport@2.0.7-beta5` and see if the bugs we've recently fixed around opening and closing work around your issue? If not, we can keep debugging and/or really blame the kernel.

reconbot, 2016-04-08:
Thinking about this further, I blame the kernel. We would be getting different file descriptors for each serial port but you're not getting separate events for each file descriptor.

kukdh1, 2016-04-09:
I tested my code with `serialport@2.0.7-beta5`, but it has same problem.

reconbot, 2016-04-09:
I blame the kernel. I'm glad you found a work around. I can tell you the
system calls we use if it would help, but I think you solved the issue
already.

kukdh1, 2016-04-09:
Thanks, then I should continue my work with irq = 0.

closed 2016-04-09 by reconbot