serialport/node-serialport #753: Remove node-pre-gyp as a bundled dependency and use preinstall script instead

bminer, 2016-04-13:
According to the new docs [here](https://github.com/mapbox/node-pre-gyp#1-add-new-entries-to-your-packagejson), it might be considered better practice to add preinstall script `npm install node-pre-gyp` instead of using the `bundledDependencies` key.  This would allow other NPM modules to re-use the same version of node-pre-gyp, for example.

See the [node-pre-gyp Github repo](https://github.com/mapbox/node-pre-gyp) page for more info.  This issue might also be relevant: https://github.com/mapbox/node-pre-gyp/issues/162

reconbot, 2016-04-19:
If it works it work, I'm for it

reconbot, 2016-04-19:
It works, I'll do a patch release in a little bit so you can use it.

reconbot, 2016-04-25:
This doesn't work, it's fine when you're installing serialport from a clone, but as a dependency it breaks everything and ends up in an endless loop. I have to revert this change. I've reopened the issue if you want to see if you can give it a go.

springmeyer, 2016-04-25:
Sorry you hit problems. As per https://github.com/mapbox/node-pre-gyp/issues/162#issuecomment-214469977 I've ask for help to know how to replicate to dig into this.

> This would allow other NPM modules to re-use the same version of node-pre-gyp, for example.

Incorrect. They still will install their own version (unless -g is passed). And this is needed since each module needs node-pre-gyp around before installing to install itself.

reconbot, 2016-04-25:
I had this happening using node v4.4.3 (npm v3.8.6) we support node `0.10`-`5.x` so it's bound to be on other platforms too

springmeyer, 2016-04-25:
I'm unable to replicate (https://github.com/mapbox/node-pre-gyp/issues/162#issuecomment-214519898) so I am presuming this is not a widespread problem and rather something specific to your setup.

reconbot, 2016-04-25:
It's happening to me on two different computers both with NVM. @bminer do you mind hopping on the mapbox thread and try to reproduce?

bminer, 2016-05-02:
Hmm... I can't replicate either... what version of node-serialport exhibits this problem?

And, I'm really confused on what the issue is exactly... you're saying that whenever you NPM install a package that depends on node-serialport, the `preinstall` script for node-serialport (which simply installs node-pre-gyp) runs in an infinite loop?  Also, why is "node-pre-gyp-github" required as a preinstall dep?

Thanks!

reconbot, 2016-05-04:
It was a mistake, I've got a branch with the unbundled package.json without pregyp github. https://github.com/voodootikigod/node-serialport/compare/unbundle  `npm install voodootikigod/node-serialport#unbundle` installs it and it's working for me now...

reconbot, 2016-05-04:
But not everywhere. I updated the node-pregyp issue with details. https://github.com/mapbox/node-pre-gyp/issues/162#issuecomment-216877736

reconbot, 2016-05-04:
Keeping this link here for our records, https://github.com/npm/npm/issues/12583 is preventing us from doing this

reconbot, 2016-05-06:
Or maybe https://github.com/mapbox/node-pre-gyp/issues/162#issuecomment-217167185

reconbot, 2016-05-06:
@bminer Please test `serialport@3.1.2-beta1` I've removed the bundling, I want to make sure it fixes your use case.

PR #759 (2016-04-19): Stop bundling node-pre-gyp to allow for upgrades
 - The latest and greatest recommendation from node-pre-gyp
 - Stops issues when people have different node-pre-gyp’s installed already

closes #753
closed 2016-05-06 by reconbot