serialport/node-serialport #754: Fails to build properly for MIPS/openwrt

johnnyman727, 2016-04-15:
Operating System and Hardware: ubuntu 14.04 VM
NodeJS Version:4.2.1
Serialport version:2.1.0
Python Version:2.7.6
(Run `node -v && npm -v && python --version` to get the version numbers.)

### Expected behavior
```
node-pre-gyp rebuild --target_platform=openwrt --target_arch=$ARCH --target=$NODE --debug
.... A lot of fancy compilation and then a successful termination
```
### Actual behavior
```
node-pre-gyp rebuild --target_platform=openwrt --target_arch=mipsel --target=4.2.1 --debug
node-pre-gyp info it worked if it ends with ok
node-pre-gyp info using node-pre-gyp@0.6.26
node-pre-gyp info using node@4.2.1 | linux | x64
gyp info it worked if it ends with ok
gyp info using node-gyp@3.3.1
gyp info using node@4.2.1 | linux | x64
gyp info ok
gyp info it worked if it ends with ok
gyp info using node-gyp@3.3.1
gyp info using node@4.2.1 | linux | x64
gyp WARN download NVM_NODEJS_ORG_MIRROR is deprecated and will be removed in node-gyp v4, please use NODEJS_ORG_MIRROR
gyp WARN download NVM_NODEJS_ORG_MIRROR is deprecated and will be removed in node-gyp v4, please use NODEJS_ORG_MIRROR
gyp info spawn /usr/bin/python2
gyp info spawn args [ '/home/vagrant/.nvm/versions/node/v4.2.1/lib/node_modules/node-gyp/gyp/gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'make',
gyp info spawn args   '-I',
gyp info spawn args   '/work/binary-module/package/build/config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/home/vagrant/.nvm/versions/node/v4.2.1/lib/node_modules/node-gyp/addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/home/vagrant/.node-gyp/4.2.1/include/node/common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
gyp info spawn args   '-Dnode_root_dir=/home/vagrant/.node-gyp/4.2.1',
gyp info spawn args   '-Dnode_gyp_dir=/home/vagrant/.nvm/versions/node/v4.2.1/lib/node_modules/node-gyp',
gyp info spawn args   '-Dnode_lib_file=node.lib',
gyp info spawn args   '-Dmodule_root_dir=/work/binary-module/package',
gyp info spawn args   '--depth=.',
gyp info spawn args   '--no-parallel',
gyp info spawn args   '--generator-output',
gyp info spawn args   'build',
gyp info spawn args   '-Goutput_dir=.' ]
gyp info ok
gyp info it worked if it ends with ok
gyp info using node-gyp@3.3.1
gyp info using node@4.2.1 | linux | x64
gyp WARN download NVM_NODEJS_ORG_MIRROR is deprecated and will be removed in node-gyp v4, please use NODEJS_ORG_MIRROR
gyp info spawn make
gyp info spawn args [ 'BUILDTYPE=Debug', '-C', 'build' ]
make: Entering directory `/work/binary-module/package/build'
  CXX(target) Debug/obj.target/serialport/src/serialport.o
  CXX(target) Debug/obj.target/serialport/src/serialport_unix.o
  CXX(target) Debug/obj.target/serialport/src/serialport_poller.o
  SOLINK_MODULE(target) Debug/obj.target/serialport.node
  COPY Debug/serialport.node
  COPY /work/binary-module/package/build/Debug/serialport.node
cp: cannot stat ‘Debug/serialport.node’: No such file or directory
make: *** [/work/binary-module/package/build/Debug/serialport.node] Error 1
make: Leaving directory `/work/binary-module/package/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/home/vagrant/.nvm/versions/node/v4.2.1/lib/node_modules/node-gyp/lib/build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:87:13)
gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
gyp ERR! System Linux 3.13.0-74-generic
gyp ERR! command "/home/vagrant/.nvm/versions/node/v4.2.1/bin/node" "/home/vagrant/.nvm/versions/node/v4.2.1/bin/node-gyp" "build" "--target_platform=openwrt" "--target_arch=mipsel" "--target=4.2.1" "--debug" "--module=/work/binary-module/package/build/Debug/serialport.node" "--module_name=serialport" "--module_path=/work/binary-module/package/build/Debug"
gyp ERR! cwd /work/binary-module/package
gyp ERR! node -v v4.2.1
gyp ERR! node-gyp -v v3.3.1
gyp ERR! not ok
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute '/home/vagrant/.nvm/versions/node/v4.2.1/bin/node /home/vagrant/.nvm/versions/node/v4.2.1/bin/node-gyp build --target_platform=openwrt --target_arch=mipsel --target=4.2.1 --debug --module=/work/binary-module/package/build/Debug/serialport.node --module_name=serialport --module_path=/work/binary-module/package/build/Debug' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/home/vagrant/.nvm/versions/node/v4.2.1/lib/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:87:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:818:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:211:5)
node-pre-gyp ERR! System Linux 3.13.0-74-generic
node-pre-gyp ERR! command "/home/vagrant/.nvm/versions/node/v4.2.1/bin/node" "/home/vagrant/.nvm/versions/node/v4.2.1/bin/node-pre-gyp" "rebuild" "--target_platform=openwrt" "--target_arch=mipsel" "--target=4.2.1" "--debug"
node-pre-gyp ERR! cwd /work/binary-module/package
node-pre-gyp ERR! node -v v4.2.1
node-pre-gyp ERR! node-pre-gyp -v v0.6.26
node-pre-gyp ERR! not ok
Failed to execute '/home/vagrant/.nvm/versions/node/v4.2.1/bin/node /home/vagrant/.nvm/versions/node/v4.2.1/bin/node-gyp build --target_platform=openwrt --target_arch=mipsel --target=4.2.1 --debug --module=/work/binary-module/package/build/Debug/serialport.node --module_name=serialport --module_path=/work/binary-module/package/build/Debug' (1)
```
### Steps and/or code to reproduce the issue
See https://github.com/tessel/t2-compiler/issues/21.

reconbot, 2016-04-15:
I'll argue this is a t2 compiler bug ;-)

rwaldron, 2016-04-16:
I'll support that argument. Something foolish is going on in t2-compiler

johnnyman727, 2016-04-18:
It could be but I'm curious about what clues give you that impression.

It looks like a pretty standard `node-pre-gyp` command being run and not anything related to `t2-compiler`: `node-pre-gyp rebuild --target_platform=openwrt --target_arch=mipsel --target=4.2.1 --debug
`

rwaldron, 2016-04-18:
Are you using the latest version of t2-compiler? target_platform isn't "openwrt", because that's not a valid platform: https://github.com/tessel/t2-compiler/commit/98b23ecff1da0df09bcdc14d8666e94b2b65f49a

rwaldron, 2016-04-18:
I've traced this issue to t2-compiler's use of pre-gypify: https://github.com/tessel/t2-compiler/blob/master/compile.sh#L32

johnnyman727, 2016-04-18:
Thanks @rwaldron closing this issue in favor of the one on `t2-compiler`.

closed 2016-04-18 by johnnyman727