serialport/node-serialport #757: node-pre-gyp out of date

Sorunome, 2016-04-19:
when installing this with "npm install" where package.json specifies "serialport": ">=2.1.0" it will use node-pre-gyp@0.6.25 instead of 0.6.26 resulting in not being able to install this via the atom package manager, support for that was added in 0.6.26

http://pastebin.com/WXmxGq8f

fivdi, 2016-04-19:
This is happening because [node-pre-gyp is a bundledDependency](https://github.com/voodootikigod/node-serialport/blob/2.1.0/package.json#L63). serialport v2.1.0 comes with node-pre-gyp@0.6.25 as a bundledDependency.

reconbot, 2016-04-19:
I'll bump the version we install, I don't know why we bundle. I'll see if we can track that down too.

reconbot, 2016-04-19:
I think we'll stop bundling it closing in favor of #753. I'm testing it out now.

closed 2016-04-19 by reconbot

fivdi, 2016-04-19:
I'd imagine that node-pre-gyp was bundled because that's what the [node-pre-gyp documentatiion](https://github.com/mapbox/node-pre-gyp/tree/v0.6.25#configuring) suggested doing up until very recently, i.e., up to and including node-pre-gyp@0.6.25.