serialport/node-serialport #768: Missing serialport-v3.0.0-node-v47-win32-x64.tar.gz

pushtheworldllc, 2016-04-27:
Operating System and Hardware: Windows
NodeJS Version: 5.11
Serialport version: 3.0.0
Python Version: 2.7.11
(Run `node -v && npm -v && python --version` to get the version numbers.)

### Expected behavior
Should have a download for serialport-v2.1.2-node-v47-win32-x64.tar.gz on releases page for 3.0.0
### Actual behavior
Missing a download for serialport-v2.1.2-node-v47-win32-x64.tar.gz
### Steps and/or code to reproduce the issue
Go to the [Releases](https://github.com/voodootikigod/node-serialport/releases/tag/3.0.0) page and see for your self...

cheton, 2016-04-28:
The file <b>serialport-v3.0.0-node-v47-win32-x64.tar.gz</b> was not uploaded to v3.0.0 release page.

reconbot, 2016-05-01:
Rebuilding this now, 3.0.1 is building so you can use that too.

reconbot, 2016-05-01:
It failed to publish again, I'm not sure why. Github is rejecting the file.

3.0.1 is out there (it's a smaller package size, not much else) so it's worth going for that.

closed 2016-05-01 by reconbot

cheton, 2016-05-04:
Hmmm. Sometimes GitHub rejected a file published from CI because the same file name has been marked broken, you may have to go to the <b>Releases</b> page and <b>[Edit]</b> the release to remove any broken files.

reconbot, 2016-05-04:
Spot on, I've deleted the broken file and kicked off the CI on this again, expect it to take an hour or so.

reconbot, 2016-05-04:
It's fixed!