serialport/node-serialport #769: Issues installing on Raspbian Jesse

EricB12, 2016-04-27:
Operating System and Hardware: Raspberry Pi 2 running Raspbian Jesse
NodeJS Version: 4.0.0
NPM Version: 2.14.2
Serialport version: unable to install
Python Version: 2.7.9
(Run `node -v && npm -v && python --version` to get the version numbers.)

### Actual behavior
I am having issues installing node.js on the Pi. I am using command "sudo npm install serialport".
<br>
Here is the npm-debug file: 
[npm-debug.txt](https://github.com/voodootikigod/node-serialport/files/239014/npm-debug.txt)
<br>
Below is the Command Prompt output:
```
sudo npm install serialport
> serialport@3.0.0 install /home/pi/node_modules/serialport
> node-pre-gyp install --fallback-to-build

gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/4.0.0"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/home/pi/node_modules/serialport/.node-gyp"
make: Entering directory '/home/pi/node_modules/serialport/build'
make: *** No rule to make target '../.node-gyp/4.0.0/include/node/common.gypi', needed by 'Makefile'.  Stop.
make: Leaving directory '/home/pi/node_modules/serialport/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:270:23)
gyp ERR! stack     at emitTwo (events.js:87:13)
gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
gyp ERR! System Linux 4.1.19-v7+
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/home/pi/node_modules/serialport/build/Release/serialport.node" "--module_name=serialport" "--module_path=/home/pi/node_modules/serialport/build/Release"
gyp ERR! cwd /home/pi/node_modules/serialport
gyp ERR! node -v v4.0.0
gyp ERR! node-gyp -v v3.0.1
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute '/usr/local/bin/node /usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/pi/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/home/pi/node_modules/serialport/build/Release' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/home/pi/node_modules/serialport/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:87:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:817:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:211:5)
node-pre-gyp ERR! System Linux 4.1.19-v7+
node-pre-gyp ERR! command "/usr/local/bin/node" "/home/pi/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /home/pi/node_modules/serialport
node-pre-gyp ERR! node -v v4.0.0
node-pre-gyp ERR! node-pre-gyp -v v0.6.26
node-pre-gyp ERR! not ok 
Failed to execute '/usr/local/bin/node /usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/pi/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/home/pi/node_modules/serialport/build/Release' (1)
npm ERR! Linux 4.1.19-v7+
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "serialport"
npm ERR! node v4.0.0
npm ERR! npm  v2.14.2
npm ERR! code ELIFECYCLE

npm ERR! serialport@3.0.0 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@3.0.0 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! This is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/npm-debug.log
```

fivdi, 2016-04-27:
Try installing with `npm install serialport` rather than `sudo npm install serialport` to see if it works.

EricB12, 2016-04-27:
I tried that and got a different exception.  After investigating, it turned out that my "node_modules" folder only had root access. I fixed the access and then "npm install serialport" worked fine.

So it turns out there wasn't anything wrong on your end.

Thanks!

fivdi, 2016-04-27:
No problem. If everything is ok now you could hit the `Close issue` button at the bottom of the page :)

closed 2016-04-27 by EricB12