serialport/node-serialport #770: Error installing on Win7 VS2015

Sylvainplourde, 2016-04-28:
Operating System and Hardware: Windows 7 on Visual Studio 2015
NodeJS Version:4.4.3
Serialport version:3.0.0
Python Version:2.7.11
Error installing
npm ERR! Windows_NT 6.1.7601
npm ERR! argv "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\bin\\npm-cli.js" "install" "serialport" "--save"
npm ERR! node v4.4.3
npm ERR! npm  v2.15.1
npm ERR! code ELIFECYCLE
npm ERR! serialport@3.0.0 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@3.0.0 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! This is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs serialport
npm ERR! Or if that isn't available, you can get their info via:
npm ERR! 
npm ERR!     npm owner ls serialport

No log file

cheton, 2016-04-29:
Try using serialport v2.1.2, there is a missing file in v3.0.0.

Related issue https://github.com/voodootikigod/node-serialport/issues/768

reconbot, 2016-05-01:
Not a lot to go on with this error. Can you try with `--verbose` mode?

Sylvainplourde, 2016-05-02:
Hi
I installed the version 3.0.1 this morning without any problem. Friday I asked to the admin to check the proxy server.  I am sure what was the problem but it’s working now.
Thanks for your good service

reconbot, 2016-05-04:
Glad to hear it!

closed 2016-05-04 by reconbot