serialport/node-serialport #771: creating a serialport on rPi

farfromrefug, 2016-04-29:
Hi,

is there  way to create a serialport on the rPi using your module?
I am trying to communicate between a rPi and a Mac over USB.

Thanks

fivdi, 2016-04-30:
It's possible for a Linux PC to communicate with a Raspberry Pi using serialport and a USB serial cable. Although I've never tried it, I'd imagine that it would be possible with a Mac too. The UART broken out onto the GPIO header on the Pi is used for communication in this case.

farfromrefug, 2016-04-30:
@fivdi wow thank you so much! I was about to give up.
I found [this](http://elinux.org/RPi_Serial_Connection) is that what you are talking about?

fivdi, 2016-04-30:
@farfromrefug yes, the technique is described on that page where it says "If you wish to connect your Raspberry Pi to a PC with a USB port, the simplest option is to use a USB-to-serial cable which uses 3.3 V logic levels ..."

farfromrefug, 2016-04-30:
@fivdi thanks so much! I just bought a cable ;)
I will try serialport with this and report success for other who might want to try

Best

closed 2016-04-30 by farfromrefug