serialport/node-serialport #772: Node.js v6.0.0 - Runtime error

ericogr, 2016-04-30:
Operating System and Hardware: Xubuntu 16.04LTS 64 bit, i7-4790K, 16GB RAM, kernel 4.4.0-21-generic
NodeJS Version: v6.0.0 npm 3.8.6
Serialport version: 3.0.0
Python Version: 2.7.11+

### Actual behavior
Runtime error

### Steps and/or code to reproduce the issue
`$ node`
```
> var SerialPort = require("serialport").SerialPort;
```
(node) v8::ObjectTemplate::Set() with non-primitive values is deprecated
(node) and will stop working in the next major release.

==== JS stack trace =========================================

Security context: 0x9bf71ac9fa9 <JS Object>#0#
    1: .node [module.js:568] [pc=0x3a0d76bb6b04] (this=0x274bed09fa39 <an Object with map 0x234ea4623371>#1#,module=0x274bed1720c9 <a Module with map 0x234ea4627361>#2#,filename=0x274bed171e61 <String[69]: /tmp/serialtest/node_modules/serialport/build/Release/serialport.node>)
    2: load [module.js:456] [pc=0x3a0d76b99f72] (this=0x274bed1720c9 <a Module with map 0x234ea4627361>#2#,filename=0x274bed171e61 <String[69]: /tmp/serialtest/node_modules/serialport/build/Release/serialport.node>)
    3: tryModuleLoad(aka tryModuleLoad) [module.js:415] [pc=0x3a0d76b99a9d] (this=0x9bf71a04189 <undefined>,module=0x274bed1720c9 <a Module with map 0x234ea4627361>#2#,filename=0x274bed171e61 <String[69]: /tmp/serialtest/node_modules/serialport/build/Release/serialport.node>)
    4: _load [module.js:407] [pc=0x3a0d76b96422] (this=0x274bed09f541 <JS Function Module (SharedFunctionInfo 0x3e0a9ad82639)>#3#,request=0x274bed171241 <String[69]: /tmp/serialtest/node_modules/serialport/build/Release/serialport.node>,parent=0x274bed163501 <a Module with map 0x234ea4627361>#4#,isMain=0x9bf71a04299 <false>)
    5: require [module.js:466] [pc=0x3a0d76b95e53] (this=0x274bed163501 <a Module with map 0x234ea4627361>#4#,path=0x274bed171241 <String[69]: /tmp/serialtest/node_modules/serialport/build/Release/serialport.node>)
    6: require(aka require) [internal/module.js:20] [pc=0x3a0d76b95b86] (this=0x9bf71a04189 <undefined>,path=0x274bed171241 <String[69]: /tmp/serialtest/node_modules/serialport/build/Release/serialport.node>)
    7: bindings [/tmp/serialtest/node_modules/bindings/bindings.js:76] [pc=0x3a0d76bb27b6] (this=0x9bf71ae8be9 <JS Global Object>#5#,opts=0x3e0a9add9201 <String[15]: serialport.node>)
    8: /* anonymous */ [/tmp/serialtest/node_modules/serialport/lib/bindings.js:3] [pc=0x3a0d76bb114e] (this=0x274bed15d421 <an Object with map 0x1dc73e207bc9>#6#,exports=0x274bed15d421 <an Object with map 0x1dc73e207bc9>#6#,require=0x274bed15e9f1 <JS Function require (SharedFunctionInfo 0x3e0a9ad93441)>#7#,module=0x274bed15d3d1 <a Module with map 0x234ea4627361>#8#,__filename=0x274bed15d311 <String[55]: /tmp/serialtest/node_modules/serialport/lib/bindings.js>,__dirname=0x274bed15e989 <String[43]: /tmp/serialtest/node_modules/serialport/lib>)
    9: _compile [module.js:541] [pc=0x3a0d76b9e504] (this=0x274bed15d3d1 <a Module with map 0x234ea4627361>#8#,content=0x274bed15e009 <String[784]\: 'use strict';\n\nvar bindings = require('bindings')('serialport.node');\nvar listUnix = require('./list-unix');\n\nvar linux = process.platform !== 'win32' && process.platform !== 'darwin';\n\nfunction listLinux(callback) {\n  callback = callback || function(err) {\n    if (err) { this.emit('error', err) }\n  }.bind(this);\n  return listUnix(callback);\n};\n\nvar platformOptions = {};\nif (process.platform !== 'win32') {\n  platformOptions = {\n    vmin: 1,\n    vtime: 0\n  };\n}\n\nmodule.exports = {\n  close: bindings.close,\n  drain: bindings.drain,\n  flush: bindings.flush,\n  list: linux ? listLinux : bindings.list,\n  open: bindings.open,\n  SerialportPoller: bindings.SerialportPoller,\n  set: bindings.set,\n  update: bindings.update,\n  write: bindings.write,\n  platformOptions: platformOptions\n};\n>,filename=0x274bed15d311 <String[55]: /tmp/serialtest/node_modules/serialport/lib/bindings.js>)
   10: .js [module.js:550] [pc=0x3a0d76b9ac8b] (this=0x274bed09fa39 <an Object with map 0x234ea4623371>#1#,module=0x274bed15d3d1 <a Module with map 0x234ea4627361>#8#,filename=0x274bed15d311 <String[55]: /tmp/serialtest/node_modules/serialport/lib/bindings.js>)
   11: load [module.js:456] [pc=0x3a0d76b99f72] (this=0x274bed15d3d1 <a Module with map 0x234ea4627361>#8#,filename=0x274bed15d311 <String[55]: /tmp/serialtest/node_modules/serialport/lib/bindings.js>)
   12: tryModuleLoad(aka tryModuleLoad) [module.js:415] [pc=0x3a0d76b99a9d] (this=0x9bf71a04189 <undefined>,module=0x274bed15d3d1 <a Module with map 0x234ea4627361>#8#,filename=0x274bed15d311 <String[55]: /tmp/serialtest/node_modules/serialport/lib/bindings.js>)
   13: _load [module.js:407] [pc=0x3a0d76b96422] (this=0x274bed09f541 <JS Function Module (SharedFunctionInfo 0x3e0a9ad82639)>#3#,request=0x3e0a9adc0799 <String[10]: ./bindings>,parent=0x274bed104101 <a Module with map 0x234ea4627361>#9#,isMain=0x9bf71a04299 <false>)
   14: require [module.js:466] [pc=0x3a0d76b95e53] (this=0x274bed104101 <a Module with map 0x234ea4627361>#9#,path=0x3e0a9adc0799 <String[10]: ./bindings>)
   15: require(aka require) [internal/module.js:20] [pc=0x3a0d76b95b86] (this=0x9bf71a04189 <undefined>,path=0x3e0a9adc0799 <String[10]: ./bindings>)
   16: /* anonymous */ [/tmp/serialtest/node_modules/serialport/lib/serialport.js:16] [pc=0x3a0d76b9ebee] (this=0x274bed104189 <an Object with map 0x1dc73e207bc9>#10#,exports=0x274bed104189 <an Object with map 0x1dc73e207bc9>#10#,require=0x274bed10c411 <JS Function require (SharedFunctionInfo 0x3e0a9ad93441)>#11#,module=0x274bed104101 <a Module with map 0x234ea4627361>#9#,__filename=0x274bed0fff71 <String[57]: /tmp/serialtest/node_modules/serialport/lib/serialport.js>,__dirname=0x274bed10c3a9 <String[43]: /tmp/serialtest/node_modules/serialport/lib>)
   17: _compile [module.js:541] [pc=0x3a0d76b9e504] (this=0x274bed104101 <a Module with map 0x234ea4627361>#9#,content=0x274bed104f79 <Very long string[13443]>#12#,filename=0x274bed0fff71 <String[57]: /tmp/serialtest/node_modules/serialport/lib/serialport.js>)
   18: .js [module.js:550] [pc=0x3a0d76b9ac8b] (this=0x274bed09fa39 <an Object with map 0x234ea4623371>#1#,module=0x274bed104101 <a Module with map 0x234ea4627361>#9#,filename=0x274bed0fff71 <String[57]: /tmp/serialtest/node_modules/serialport/lib/serialport.js>)
   19: load [module.js:456] [pc=0x3a0d76b99f72] (this=0x274bed104101 <a Module with map 0x234ea4627361>#9#,filename=0x274bed0fff71 <String[57]: /tmp/serialtest/node_modules/serialport/lib/serialport.js>)
   20: tryModuleLoad(aka tryModuleLoad) [module.js:415] [pc=0x3a0d76b99a9d] (this=0x9bf71a04189 <undefined>,module=0x274bed104101 <a Module with map 0x234ea4627361>#9#,filename=0x274bed0fff71 <String[57]: /tmp/serialtest/node_modules/serialport/lib/serialport.js>)
   21: _load [module.js:407] [pc=0x3a0d76b96422] (this=0x274bed09f541 <JS Function Module (SharedFunctionInfo 0x3e0a9ad82639)>#3#,request=0x3e0a9adb9389 <String[10]: serialport>,parent=0x274bed0afd79 <a Module with map 0x234ea4627361>#13#,isMain=0x9bf71a04299 <false>)
   22: require [module.js:466] [pc=0x3a0d76b95e53] (this=0x274bed0afd79 <a Module with map 0x234ea4627361>#13#,path=0x3e0a9adb9389 <String[10]: serialport>)
   23: require(aka require) [internal/module.js:20] [pc=0x3a0d76b95b86] (this=0x9bf71a04189 <undefined>,path=0x3e0a9adb9389 <String[10]: serialport>)
   24: /* anonymous */ [repl:1] [pc=0x3a0d76b94cf6] (this=0x9bf71ae8be9 <JS Global Object>#5#)
   28: defaultEval [repl.js:272] [pc=0x3a0d76b94200] (this=0x274bed0ab7d9 <an REPLServer with map 0x234ea4629511>#14#,code=0x274bed0f7fc9 <String[51]\: var SerialPort = require("serialport").SerialPort;\n>,context=0x9bf71ae8be9 <JS Global Object>#5#,file=0x9bf71ad7891 <String[4]: repl>,cb=0x274bed0f73a1 <JS Function finish (SharedFunctionInfo 0x3e0a9adb6fc9)>#15#)
   29: bound(aka bound) [domain.js:280] [pc=0x3a0d76b92f2e] (this=0x9bf71a04189 <undefined>,_this=0x274bed0ab7d9 <an REPLServer with map 0x234ea4629511>#14#,self=0x274bed0ab979 <a Domain with map 0x234ea4625fc9>#16#,cb=0x274bed0ab891 <JS Function defaultEval (SharedFunctionInfo 0x3e0a9ad8db71)>#17#,fnargs=0x274bed0f8021 <an Arguments with map 0x1dc73e20a509>#18#)
   30: eval(aka runBound) [domain.js:293] [pc=0x3a0d76b92d32] (this=0x274bed0ab7d9 <an REPLServer with map 0x234ea4629511>#14#)
   31: arguments adaptor frame: 4->0
[... cut ...]
   32: /* anonymous */ [repl.js:441] [pc=0x3a0d76b9061e] (this=0x274bed0ab7d9 <an REPLServer with map 0x234ea4629511>#14#,cmd=0x274bed0f60d9 <String[50]: var SerialPort = require("serialport").SerialPort;>)
   33: emitOne(aka emitOne) [events.js:101] [pc=0x3a0d76b84139] (this=0x9bf71a04189 <undefined>,handler=0x274bed0cee91 <JS Array[2]>#19#,isFn=0x9bf71a04299 <false>,self=0x274bed0ab7d9 <an REPLServer with map 0x234ea4629511>#14#,arg1=0x274bed0f60d9 <String[50]: var SerialPort = require("serialport").SerialPort;>)undefined

fivdi, 2016-04-30:
This isn't actually a crash, it's a dramatically formatted message indicating that there will be a change to V8 at some point in the future:
```
(node) v8::ObjectTemplate::Set() with non-primitive values is deprecated
(node) and will stop working in the next major release.
```
The message can be ignored and everything should function as expected. The message will no longer appear when serialport is upgraded to nan v2.3.2.

ericogr, 2016-04-30:
Yes, thats true. I'll change the title.

reconbot, 2016-05-01:
Some of these things are errors in the latest nan with node 6. So we can't upgrade it yet.

fivdi, 2016-05-01:
There's more to this than I initially assumed. https://github.com/nodejs/nan/pull/560 which was released with nan@2.3.2 modified `Nan::SetPrototypeMethod` to solve the deprecation message issue. However, the deprecation messages in serialport are being caused by `Nan::SetPrototypeTemplate` [here](https://github.com/voodootikigod/node-serialport/blob/3.0.1/src/serialport_poller.cpp#L63-L69). As can be seen in https://github.com/nodejs/nan/issues/565, `Nan::SetPrototypeTemplate` will not be modified.

If serialport uses `Nan::SetPrototypeMethod` rather than `Nan::SetPrototypeTemplate` the deprecation message should no longer appear.

reconbot, 2016-05-01:
merged the latest nan with @fivdi's cleanup in #778

We were getting inconsistent failures with node6 on osx so I added it to the "allowed failures" until it's clear this patch fixed that or we figure out that issue.

closed 2016-05-01 by reconbot