serialport/node-serialport #780: v8::ObjectTemplate::Set() with non-primitive values is deprecated  and will stop working in the next major release.

nfriedly, 2016-05-02:
Operating System and Hardware: OS X 10.11.4,  2015 Macbook Pro
NodeJS Version: v6.0.0
Serialport version: v3.0.1 or master
Python Version: 2.7.10

### Expected behavior
No warning message

### Actual behavior

Using this library on node v6, or even just running `npm test` prints this near the top of the output:

```
(node) v8::ObjectTemplate::Set() with non-primitive values is deprecated
(node) and will stop working in the next major release.

==== JS stack trace =========================================

Security context: 0x3f87617c9fa9 <JS Object>#0#
    1: .node [module.js:568] [pc=0x2a635285c24] (this=0x30a892bf1239 <an Object with map 0x1493c7842bb9>#1#,module=0x268b6bb69539 <a Module with map 0x1493c78181b9>#2#,filename=0x268b6bb69511 <String[61]: /Users/nfriedly/node-serialport/build/Release/serialport.node>)
    2: load [/Users/nfriedly/node-serialport/node_modules/coffee-script/lib/coffee-script/register.js:~40] [pc=0x2a6352134e2] (this=0x268b6bb69539 <a Module with map 0x1493c78181b9>#2#,filename=0x268b6bb69511 <String[61]: /Users/nfriedly/node-serialport/build/Release/serialport.node>)
    3: tryModuleLoad(aka tryModuleLoad) [module.js:415] [pc=0x2a634a3899d] (this=0x3f8761704189 <undefined>,module=0x268b6bb69539 <a Module with map 0x1493c78181b9>#2#,filename=0x268b6bb69511 <String[61]: /Users/nfriedly/node-serialport/build/Release/serialport.node>)
    4: _load [module.js:407] [pc=0x2a634a345e2] (this=0x30a892bf12b9 <JS Function 1: v8::Template::Set(v8::Local<v8::Name>, v8::Local<v8::Data>, v8::PropertyAttribute)
 2: SerialportPoller::Init(v8::Local<v8::Object>)
 3: init
 4: node::DLOpen(v8::FunctionCallbackInfo<v8::Value> const&)
 5: v8::internal::FunctionCallbackArguments::Call(void (*)(v8::FunctionCallbackInfo<v8::Value> const&))
 6: v8::internal::MaybeHandle<v8::internal::Object> v8::internal::(anonymous namespace)::HandleApiCallHelper<false>(v8::internal::Isolate*, v8::internal::(anonymous namespace)::BuiltinArguments<(v8::internal::BuiltinExtraArguments)1>)
 7: v8::internal::Builtin_HandleApiCall(int, v8::internal::Object**, v8::internal::Isolate*)
 8: 0x2a63490961b
 9: 0x2a635285c24
(node) v8::ObjectTemplate::Set() with non-primitive values is deprecated
(node) and will stop working in the next major release.

==== JS stack trace =========================================

Security context: 0x3f87617c9fa9 <JS Object>#0#
    1: .node [module.js:568] [pc=0x2a635285c24] (this=0x30a892bf1239 <an Object with map 0x1493c7842bb9>#1#,module=0x268b6bb69539 <a Module with map 0x1493c78181b9>#2#,filename=0x268b6bb69511 <String[61]: /Users/nfriedly/node-serialport/build/Release/serialport.node>)
    2: load [/Users/nfriedly/node-serialport/node_modules/coffee-script/lib/coffee-script/register.js:~40] [pc=0x2a6352134e2] (this=0x268b6bb69539 <a Module with map 0x1493c78181b9>#2#,filename=0x268b6bb69511 <String[61]: /Users/nfriedly/node-serialport/build/Release/serialport.node>)
    3: tryModuleLoad(aka tryModuleLoad) [module.js:415] [pc=0x2a634a3899d] (this=0x3f8761704189 <undefined>,module=0x268b6bb69539 <a Module with map 0x1493c78181b9>#2#,filename=0x268b6bb69511 <String[61]: /Users/nfriedly/node-serialport/build/Release/serialport.node>)
    4: _load [module.js:407] [pc=0x2a634a345e2] (this=0x30a892bf12b9 <JS Function 1: v8::Template::Set(v8::Local<v8::Name>, v8::Local<v8::Data>, v8::PropertyAttribute)
 2: SerialportPoller::Init(v8::Local<v8::Object>)
 3: init
 4: node::DLOpen(v8::FunctionCallbackInfo<v8::Value> const&)
 5: v8::internal::FunctionCallbackArguments::Call(void (*)(v8::FunctionCallbackInfo<v8::Value> const&))
 6: v8::internal::MaybeHandle<v8::internal::Object> v8::internal::(anonymous namespace)::HandleApiCallHelper<false>(v8::internal::Isolate*, v8::internal::(anonymous namespace)::BuiltinArguments<(v8::internal::BuiltinExtraArguments)1>)
 7: v8::internal::Builtin_HandleApiCall(int, v8::internal::Object**, v8::internal::Isolate*)
 8: 0x2a63490961b
 9: 0x2a635285c24
```

### Steps and/or code to reproduce the issue

clone the repo, run `npm install && npm test`

nfriedly, 2016-05-02:
I'm pretty sure it's this line causing the trouble: https://github.com/voodootikigod/node-serialport/blob/master/src/serialport_poller.cpp#L71 but I'm not yet sure what the appropriate change is.

nfriedly, 2016-05-02:
Op, sorry I see now that this is already covered in the readme. I was using it indirectly (via `particle-cli`) and didn't catch that at first.

fivdi, 2016-05-02:
@nfriedly particle-cli depends on serialport ^2.0.1, not 3.0.1. This issue should be fixed on master. Did you really clone master and see the issue?

nfriedly, 2016-05-02:
I cloned particle-cli master and bumped it to use node-serialport v3.0.1 hoping that would fix it, but I still saw the same issue. 

Then I cloned node-serialport master and just did `npm install; npm test` and saw that it to had the same issue cropping up in the test output.

nfriedly, 2016-05-02:
(I then spent about 20 minutes fiddling around with c++ and nan, hoping to send a PR rather than a bug report before admitting that I didn't know what I was doing...)

fivdi, 2016-05-02:
A few things need to be taken into account here.

* serialport@3.0.1 was published on npm yesterday.
* After serialport@3.0.1 was published on npm, the fix for the deprecation warnings was pushed to master on Github.
* This means serialport@3.0.1 on npm and master on Github are different.
* Bumping the version number for serialport in particle-cli will not change anything as this will result in serialport@3.0.1 being installed from npm which doesn't contain the fix.
* serialport uses pre-compiled binaries.
* This usage of pre-compiled binaries has an unfortunate side effect. Cloning the repo and doing `npm install` will not actually compile the source code in the cloned repo. It will download the outdated pre-compiled binaries for serialport@3.0.1.

After cloning the repo do `npm install --build-from-source` rather than `npm install` and everything should function as expected.

nfriedly, 2016-05-04:
Aah, yes, you are correct @fivdi: with `--build-from-source` I can run the tests locally and do not get any warnings logged. Should we close this now or wait until the next release comes out?

reconbot, 2016-05-04:
Both! I've got 3.0.2 in the works, just having [CI issues](https://twitter.com/reconbot/status/727907727075315712) they've stalled out on builds.

nfriedly, 2016-05-04:
Ok, cool! (Well, bummer about the CI issues, but cool otherwise.)

If you update this once 3.0.2 does get released, I'll send a PR to particle-cli with the update :)

reconbot, 2016-05-04:
3.1.0 has been released (keeping with semvar there are some new cli features so it's a minor release not a patch release) the binaries are building now. =)

closed 2016-05-04 by reconbot

nfriedly, 2016-05-04:
beautiful, thanks!

nfriedly, 2016-05-04:
Hey, so I sent the PR to particle-cli and it's passing on node v5 and v6 but failing on v4 and older: https://travis-ci.org/spark/particle-cli/builds/127913574 

Here's the relevant bit:

```
$ node --version
v4.4.3
$ npm --version
2.15.1
$ nvm --version
0.31.0
30.85s$ npm install 
npm WARN deprecated jade@0.26.3: Jade has been renamed to pug, please install the latest version of pug instead of jade
npm WARN deprecated graceful-fs@2.0.3: graceful-fs v3.0.0 and before will fail on node releases >= v6.0. Please update to graceful-fs@^4.0.0 as soon as possible. Use 'npm ls graceful-fs' to find it in the tree.
> serialport@3.1.0 install /home/travis/build/spark/particle-cli/node_modules/serialport
> node-pre-gyp install --fallback-to-build
module.js:327
    throw err;
    ^
Error: Cannot find module 'nopt'
    at Function.Module._resolveFilename (module.js:325:15)
    at Function.Module._load (module.js:276:25)
    at Module.require (module.js:353:17)
    at require (internal/module.js:12:17)
    at Object.<anonymous> (/home/travis/build/spark/particle-cli/node_modules/serialport/node_modules/node-pre-gyp/lib/node-pre-gyp.js:14:12)
    at Module._compile (module.js:409:26)
    at Object.Module._extensions..js (module.js:416:10)
    at Module.load (module.js:343:32)
    at Function.Module._load (module.js:300:12)
    at Module.require (module.js:353:17)
```

This might not actually be a serialport issue, but I thought I'd ask here first since you were just working with this. Any ideas?

reconbot, 2016-05-04:
It's our issue, something weird happened with our bundled deps.

reconbot, 2016-05-04:
@nfriedly can you try again with `serialport@3.1.1`?

nfriedly, 2016-05-05:
Yep, [all green now](https://travis-ci.org/spark/particle-cli/builds/127944469). Thanks!

reconbot, 2016-05-05:
Btw this has led me down a rabbit hole where I've discovered at least one
maybe two npm bugs. Go figure =)

nfriedly, 2016-05-05:
Haha, bugs all around! :bug: