serialport/node-serialport #791: Not able to get Arduino SerialNumber within docker container

mibzzz, 2016-05-07:
Hello,
I have an application which uses Arduino connected to /ttyACM0 port and is being run in dockerized way. The problem is I cannot get serial number for the connected ttyACM0 port. It returns me "undefined" value.
I start the container as follows:
`docker run --name ehealth_platform --net=host -it --privileged -d -p 127.0.1.1:3000:3000 -v /dev/ttyACM0:/dev/ttyACM0 test-image`

And the nodejs code I try to fetch the serialnumber with:
`var serialport = require('serialport');
var SerialPort = serialport.SerialPort;
var portName = '/dev/ttyACM0';
var ARDUINO_SERIAL_NUMBER = "";

serialport.list(function (err, ports) {
    ports.forEach(function(port) {
        if(port.comName == portName){
                ARDUINO_SERIAL_NUMBER = port.serialNumber;
                console.log(ARDUINO_SERIAL_NUMBER);
        }
    });
});

var connection = new SerialPort(portName, {
    baudRate:115200,
    parser:serialport.parsers.readline("\n")
});`

I would really appreciate your help.

reconbot, 2016-05-07:
We require `udevadm` to be present and in the path to lookup extended information about the serial ports. If you expose that to your docker container you should be set.

reconbot, 2016-05-07:
https://github.com/voodootikigod/node-serialport/blob/master/lib/list-unix.js#L76-L79

reconbot, 2016-05-16:
I'm going to close this for now, but if you'd like to continue with it feel free to reopen.

closed 2016-05-16 by reconbot