serialport/node-serialport #795: Error: Writing to COM port (GetOverlappedResult)

WhiteHusky, 2016-05-09:
Operating System and Hardware: Windows 10 Professional, Intel(R) Core(TM) i5-4590S CPU @ 3.00GHz
NodeJS Version: v4.4.4
Serialport version: 3.1.2-beta1

### Expected behavior
Open a port to COM3, a USB to Serial adapter connected to a ZPL Printer, and send ZPL commands with the documented configuration.
### Actual behavior
Script errors with `Error: Writing to COM port (GetOverlappedResult): operation aborted`
### Steps and/or code to reproduce the issue
Attempt to open a valid COM port.

WhiteHusky, 2016-05-09:
I noticed I might be doing something wrong because it's working by simply doing a bit of testing.

Will post findings.

WhiteHusky, 2016-05-09:
Found what the problem was. Drain didn't do what I expected and I thought it would callback once the data finished sending. Added callback to write and properly closed the port.

reconbot, 2016-05-16:
As a warning, `write()` might call the callback multiple times if you're writing a lot of data. The callback's behavior will be a lot better in an upcoming release working on #808 to clean that up.

closed 2016-05-16 by reconbot

Iplooky, 2017-11-02:
Hello, having the same issue on my LoPy when I want to upload my project. I cand only run the file but not sync it. Any advice ? Thank you !

reconbot, 2017-11-02:
Hi @lplooky please open a new issue and fill out the details in the form.