serialport/node-serialport #798: Port is Opening

proyectos1Insepet, 2016-05-10:
Operating System and Hardware:Jessy - BeagleBone Black Image 2016-01-24
NodeJS Version: 0.10.41
Serialport version: 3.1.2-beta1
Python Version (if you're having trouble building serialport):
(Run `node -v && npm -v && python --version` to get the version numbers.)

Some install problems I can`t use npm install serialport -g "can`t enter to root folder..."
then I install in the same folder that the Cloud9 with some errors and when I execute the code have the error "Port is opening" but if i delete the line 198 in the javascript I can execute the code, the line "if (this.opening){return this._error(new Error ('Port is opening'), callback);}"

reconbot, 2016-05-10:
The port will open automatically when you create the serialport object unless you tell it not to. So calling `open` yourself will result in an error. Can you share some code?

As for the local installation issue, is your npm functioning properly? Can you install other packages globally?

proyectos1Insepet, 2016-05-12:
I reinstalled the OS and the global installation worked. I have the error "Error: Port is opening
    at SerialPort.open (/usr/local/lib/node_modules/serialport/lib/serialport.js:198:24)
    at SerialPort.factory.SerialPort (/usr/local/lib/node_modules/serialport/lib/serialport.js:176:12)"
Some code:
var sp                         = require("serialport");
var port_mux              = '/dev/ttyO4';
var config_port_mux  = {baudrate: 9600, parser: sp.parsers.readline("*")};
var muxport                = new sp.SerialPort(port_mux,config_port_mux);
muxport.open(abrir);
function abrir(error){
   if (error){
       console.log('failed to open: '+error);
   } else{
       console.log('open '+port_mux);
       muxport.on('data',rx_data_mux);
   }
}

In a previous version (Im not sure which, a version in February maybe) I didn`t have that problem. If i delete that code lines my code works.
Regards

reconbot, 2016-05-12:
Previously we'd let you open a port twice. This would cause issues with being notified about data events, drain and flush, where the first open might be given the information but the second open wouldn't. It would also cause a crash on windows, not even a nice error message.

Your code is opening the serialport twice. Once in the constructor and once when you call open. The call inside of the constructer is asynchronously so your call to `open` happens first, and the constrictor's call fails without a callback, so an error event is emitted. You can fix this two ways.

Don't openImmediatly in the constructor by passing false.
```js
var muxport = new sp.SerialPort(port_mux, config_port_mux, false);
muxport.open(abrir);
```

Or Have the constructor open the port for you and don't call `.open()` yourself.
```js
var muxport = new sp.SerialPort(port_mux, config_port_mux, abrir);
```
[The docs have more information.](https://github.com/voodootikigod/node-serialport/blob/3.1.0/README.md#opening-a-port)

proyectos1Insepet, 2016-05-12:
Thanks a lot! I didn't understand why the code works before, I did the change and worked.
Have a nice day :)

closed 2016-05-12 by proyectos1Insepet