serialport/node-serialport #799: Installing serialport in Docker container fails

kimhorn, 2016-05-10:
NodeJS Version: 4.4.4
Serialport version: 3.1.1

The installation fails.
I have tryed to install  nodejs-legacy, but it did not help.

npm info build /usr/src/app/node_modules/serialport
npm info linkStuff serialport@3.1.1
npm info build /usr/src/app/node_modules/serialport/node_modules/node-pre-gyp
npm info preinstall node-pre-gyp@0.6.27
npm info linkStuff node-pre-gyp@0.6.27
npm ERR! Linux 4.1.13-boot2docker
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install"
npm ERR! node v4.4.4
npm ERR! npm  v2.15.1
npm ERR! path ../serialport/bin/serialport-list.js
npm ERR! code EPROTO
npm ERR! errno -71
npm ERR! syscall symlink

npm ERR! EPROTO: protocol error, symlink '../serialport/bin/serialport-list.js' -> '/usr/src/app/node_modules/.bin/serialportlist'
npm ERR! 
npm ERR! If you need help, you may report this error at:
npm ERR!     <https://github.com/npm/npm/issues>
npm info preuninstall serialport@3.1.1
npm info uninstall serialport@3.1.1
npm info postuninstall serialport@3.1.1

npm ERR! Please include the following file with any support request:
npm ERR!     /usr/src/app/npm-debug.log

reconbot, 2016-05-10:
I don't think this is a serialport issue, it's trying to make the symlinks for our binaries (serialport list and serialport term) Perhaps your docker environment doesn't have the destination path or has a filesystem that doesn't support symlinks?

I found this https://github.com/npm/npm/issues/9901 which might help

kimhorn, 2016-05-11:
Great thanks, that solved the problem.
But now I get another error that look like a serialport issue:

npm info build /usr/src/app/node_modules/serialport
npm info install serialport@3.1.1
2016-05-11T19:35:56.727772323Z 
> serialport@3.1.1 install /usr/src/app/node_modules/serialport
> node-pre-gyp install --fallback-to-build
sh: 1: node-pre-gyp: not found
npm info serialport@3.1.1 Failed to exec install script
npm ERR! Linux 4.4.8-boot2docker
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "--no-bin-links"
npm ERR! node v4.4.4
npm ERR! npm  v2.15.1
npm ERR! file sh
npm ERR! code ELIFECYCLE
npm ERR! errno ENOENT
npm ERR! syscall spawn
2016-05-11T19:35:56.835293964Z 
npm ERR! serialport@3.1.1 install: `node-pre-gyp install --fallback-to-build`
npm ERR! spawn ENOENT
npm ERR! 
npm ERR! Failed at the serialport@3.1.1 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! This is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs serialport
npm ERR! Or if that isn't available, you can get their info via:
npm ERR! 
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.
npm info preuninstall serialport@3.1.1
npm info uninstall serialport@3.1.1
npm info postuninstall serialport@3.1.1
npm ERR! Please include the following file with any support request:
npm ERR!     /usr/src/app/npm-debug.log
[npm-debug.log.txt](https://github.com/voodootikigod/node-serialport/files/260085/npm-debug.log.txt)

reconbot, 2016-05-11:
Not that you should need it, but can you try upgrading your npm to v3? Eg
npm install -g npm

kimhorn, 2016-05-11:
Thanks for the tip.
I have upgraded and now I get another error:

npm ERR! Linux 4.4.8-boot2docker
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "--no-bin-links"
npm ERR! node v4.4.4
npm ERR! npm  v3.8.9
npm ERR! path /usr/src/app/node_modules/.staging/serialport-cc06aeb5/node_modules/node-pre-gyp/node_modules/request/node_modules/har-validator/node_modules/is-my-json-valid/node_modules/generate-object-property/node_modules/is-property/is-property.js
npm ERR! code EPERM
npm ERR! errno -1
npm ERR! syscall open
2016-05-11T20:28:43.913166007Z 
npm ERR! Error: EPERM: operation not permitted, open '/usr/src/app/node_modules/.staging/serialport-cc06aeb5/node_modules/node-pre-gyp/node_modules/request/node_modules/har-validator/node_modules/is-my-json-valid/node_modules/generate-object-property/node_modules/is-property/is-property.js'
npm ERR!     at Error (native)
npm ERR!  { [Error: EPERM: operation not permitted, open '/usr/src/app/node_modules/.staging/serialport-cc06aeb5/node_modules/node-pre-gyp/node_modules/request/node_modules/har-validator/node_modules/is-my-json-valid/node_modules/generate-object-property/node_modules/is-property/is-property.js']
npm ERR!   errno: -1,
npm ERR!   code: 'EPERM',
npm ERR!   syscall: 'open',
2016-05-11T20:28:43.943799799Z npm ERR!   path: '/usr/src/app/node_modules/.staging/serialport-cc06aeb5/node_modules/node-pre-gyp/node_modules/request/node_modules/har-validator/node_modules/is-my-json-valid/node_modules/generate-object-property/node_modules/is-property/is-property.js',
npm ERR!   parent: 'PLCMock' }
npm ERR! 
npm ERR! Please try running this command again as root/Administrator.
npm ERR! Please include the following file with any support request:
npm ERR!     /usr/src/app/npm-debug.log
[npm-debug2.log.txt](https://github.com/voodootikigod/node-serialport/files/260197/npm-debug2.log.txt)

reconbot, 2016-05-11:
You're hitting basic npm issues. I know @mhart has had success with Alpine linux, but I bet this has more to do with your setup then serialport. This is beyond my docker and npm knowledge.

mhart, 2016-05-11:
@kimhorn it worked for me in an Alpine Docker container with these instructions: https://github.com/voodootikigod/node-serialport/pull/710/files

mhart, 2016-05-11:
So, I think this should work:

```Dockerfile
FROM mhart/alpine-node:4

WORKDIR /src

RUN apk add --no-cache make gcc g++ python linux-headers udev
RUN npm install serialport --build-from-source=serialport
```

mhart, 2016-05-11:
Interesting, I'm getting this during the install step:
```console
> serialport@3.1.2-beta2 install /src/node_modules/serialport
> node-pre-gyp install --fallback-to-build

module.js:327
    throw err;
    ^

Error: Cannot find module 'nopt'
    at Function.Module._resolveFilename (module.js:325:15)
    at Function.Module._load (module.js:276:25)
    at Module.require (module.js:353:17)
    at require (internal/module.js:12:17)
    at Object.<anonymous> (/src/node_modules/serialport/node_modules/node-pre-gyp/lib/node-pre-gyp.js:14:12)
    at Module._compile (module.js:409:26)
    at Object.Module._extensions..js (module.js:416:10)
    at Module.load (module.js:343:32)
    at Function.Module._load (module.js:300:12)
    at Module.require (module.js:353:17)
```

@reconbot any ideas? This used to work back when I posted these instructions

mhart, 2016-05-11:
@reconbot Ah, I see it: #800

mhart, 2016-05-11:
In any case, when #800 is fixed, those instructions should work 😸

mhart, 2016-05-11:
Or you can do this:
```Dockerfile
FROM mhart/alpine-node:4

WORKDIR /src

RUN apk add --no-cache make gcc g++ python linux-headers udev
RUN npm install serialport@3.1.1 --build-from-source=serialport
```
Just confirmed that works fine

@reconbot should the beta builds perhaps not be the default npm version?

reconbot, 2016-05-11:
Don't use the beta, it's broken

mhart, 2016-05-11:
Yeah, maybe mark it is a different tag in npm (`beta` or `next` perhaps) – instead of `latest` – more here if you need it: https://docs.npmjs.com/cli/dist-tag

reconbot, 2016-05-11:
Ahh sorry, I wasn't aware of that feature I thought it did it automatically from the semvar. I've updated the publish docs and fixed the tag to point back to the latest non beta

kimhorn, 2016-05-14:
For the record, my issues was fixed by:
- Not using symlinks: "npm install --no-bin-links"
- Installing node-pre-gyp as global 
- upgrading to the latest npm version (not sure if that made a difference)
- not using serialport 3.1.2-beat2

Thanks

closed 2016-05-14 by kimhorn

reconbot, 2016-05-15:
3.1.2 is now out of beta and will work for you.