serialport/node-serialport #800: npm install serialport broke since 3.0.1

fluentart, 2016-05-11:
Operating System and Hardware: MacBook Pro 13" 2015 128GB
NodeJS Version: 4.4.4
Serialport version: 3.1.2-beta2
Python Version (if you're having trouble building serialport): 2.7.10
(Run `node -v && npm -v && python --version` to get the version numbers.)

### Expected behavior
npm install serialport 
to work.

### Actual behavior
Error: Cannot find module 'nopt'
And some other strange modules.. 

npm install serialport@3.0.1
works fine.

### Steps and/or code to reproduce the issue
As above.
Something broke since 3.0.1

reconbot, 2016-05-11:
Ahaha lovely

reconbot, 2016-05-12:
`3.1.1` should work for you. The `nopt` bundling issue is a pain. I've created the [serialport-test-pilot](https://github.com/j5js/serialport-test-pilot) project to help catch these things before they hit npm

reconbot, 2016-05-13:
3.1.2-beta3 has the same issue, it seems to be caused by npm 3 bundling and being installed with npm 2 package manages.

`3.1.1` will install by default now if you don't specify instead of the beta releases.

Can you confirm your version of npm? if you upgrade it it should work.

reconbot, 2016-05-15:
3.1.2 is now out of beta and will work for you. Check out https://twitter.com/reconbot/status/731693005745356800 for details on how we'll stop bad releases in the future

closed 2016-05-15 by reconbot