serialport/node-serialport #802: Detect plugging and unplugging of VirtualComPorts on the fly

buejoh, 2016-05-12:
Operating System and Hardware: MAC OS El Capitan 10.11.4
NodeJS Version: 6.1.0
Serialport version: 3.1.2-beta2
Used VCP:

- Produkt-ID:	0x6001
- Hersteller-ID:	0x0403  (Future Technology Devices International Limited)
- Version:	6.00
- Seriennummer:	A50285BI
- Hersteller:	FTDI

Hi everybody,

i've been using node-serialport for an bunch of projects recently without problems. Thanks to the whole team providing this module to the community! 👍 

In a current project i am trying to detect plugging and unplugging of vcp-devices on the fly (FTDI USB VCP (Virtual Com Port)) with node-serialport. For testing purposes i am polling the .list-method via setInterval to detect a device with manufacturer FTDI. If such a device is plugged in, i create a serial port object, attach event handlers and everything just works as expected (until the device gets unplugged...). 

### Expected behavior
When i unplug the device, i am expecting the .disconnect event to be fired, causing all low-level references (lib_uv filedescriptor?) for this particular VCP to be released. Plugging the device again, i am expecting a positive detection in the .list callback to enable reconnecting this device **without a restart of my node application.**

### Actual behavior
After unplugging the device, the disconnect event gets fired once as expected. After re-pluggin the device, the .list method doesn't show up the device and so a reconnect isn't possible without a OS reboot or using a different USB port.

### Additional OS observations
Checking the USB bus devices via the system report on MAC OS after unplugging the device, it still gets listed there, but cannot be found via `ls /dev`. When i replug the device, it gets listed twice in the system report (same productID, vendorID, location ...). It still cannot be found via `ls /dev`.

I tested the same code, provided in this issue on a virtual machine with Windows 7 Professional (64 Bit), **this works like a charm**!

Is there a chance, to get this running on a MAC too? Do I have to edit the sources manually or to change something in my code? Or might it be a driver/lib_uv related issue?

Sorry for the long post :)

### Code to reproduce the issue
	var Serialport  = require('serialport');

	var busy = false;
	function setReady() { busy = false; }
	function setBusy() { busy = true; }

	// Poll available vcp's
	setInterval(function() {
	    console.log("Poll for new devices..");

	    Serialport.list(function (err, ports) {    
	        ports.forEach(function(port) {
	            // Create serialport instance, if not existing
	            if((port.manufacturer == "FTDI") && !busy) {
	                setBusy();
	                console.log("Device found.");
	                create(port.comName);
	            }
	        });
	    });
	}, 2000);

	// Create a serialport instance
	function create(comName) {    
	    var com = new Serialport.SerialPort(comName);

	    com.on("open", function(err) {
			console.log("Port opened. Error: ", err);
	    });

	    com.on('data', function (data) { 
	        console.log("data", data); 
	    });

	    com.on('error', function (err) { 
	        console.log('Error: ', err);
	    });

	    com.on('disconnect', function (err) { 
	        console.log("Disconnected. Error: ", err);
			setTimeout(function() {
	            setReady();
	        }, 2000);        
	    });
	}

### Console logs

On windows:

	Poll for new devices..
	Device found.
	Poll for new devices..
	Port opened. Error:  undefined
	Poll for new devices..
	Poll for new devices..
	Poll for new devices..
	Disconnected. Error:  [Error: Disconnected]
	Poll for new devices..
	Poll for new devices..
	Poll for new devices..
	Device found.
	Poll for new devices..
	Port opened. Error:  undefined
	Poll for new devices..
	Poll for new devices..
	Disconnected. Error:  [Error: Disconnected]
	Poll for new devices..
	Poll for new devices..
	Device found.
	Poll for new devices..
	Poll for new devices..
	Port opened. Error:  undefined
	Poll for new devices..
	Poll for new devices..
	Disconnected. Error:  [Error: Disconnected]
	Poll for new devices..

On MAC

	Poll for new devices..
	Device found.
	Port opened. Error:  undefined
	Poll for new devices..
	Poll for new devices..
	Disconnected. Error:  { Error: ENXIO: no such device or address, read
	    at Error (native) errno: -6, code: 'ENXIO', syscall: 'read' }
	Poll for new devices..
	Poll for new devices..
	Poll for new devices..
	Poll for new devices..
	Poll for new devices.. (device plugged in again)
	Poll for new devices..
	Poll for new devices..
	Poll for new devices..
	Poll for new devices..
	Poll for new devices..
	Poll for new devices..

reconbot, 2016-05-12:
This is a known but unexplored issue #702. And on my short list. I'm glad you did some research. I welcome the help.

I can tell you that linux is the only OS where we look at `/dev/` for the serialports. OSX has other facalties for that. I'm guessing we're not closing the FD, nor are we detecting all the states where we might be disconnected. Finally windows has it's own disconnect path which is silly. That path doesn't cleanup or fire all events.

buejoh, 2016-05-13:
Hey,

thanks for your answer. 

> That path doesn't cleanup

On windows, path gets cleaned up just fine or am i misunderstanding you?

Do you have a Xcode project or anything else, where i can start debugging with?

reconbot, 2016-07-03:
Hi @buejoh we've refactored this a little bit for 4.x. Still haven't fully tested disconnects but we do a bit more to ensure file descriptors get closed on all platforms.

buejoh, 2016-07-04:
Hi @reconbot,
just tested with serialport@4.0.0 but still no change with my setup.

reconbot, 2016-07-20:
I am able to disconnect and reconnect to an FTDI com port (in this case an arduino with the chipset) multiple times on my mac. OSX EL Capitan. Both with your code and manually testing. I'd imagine that it's something to do with the FTDI drivers. I don't have another FTDI device around to test with.

reconbot, 2016-07-21:
I should also note that you're OS notes answers the question. We use some OSX api's to enumerate the ports but if it's not in `/dev/` then we can't open it.

closed 2016-07-28 by reconbot