serialport/node-serialport #804: Garbage output?  ������

tagyoureit, 2016-05-13:
Operating System and Hardware: Rasberry Pi 3
NodeJS Version: v0.10.29
Serialport version: 3.1.1
Python Version (if you're having trouble building serialport): n/a
(Run `node -v && npm -v && python --version` to get the version numbers.)

### Expected behavior
Hopefully I'm missing something very simple here.  I am using a USB->RS485 cable.
When I capture the serial port from the command line it works correctly:

> pi@raspberrypi:~/ $ od -x < /dev/ttyUSB0
> 0000000 b200 0000 0000 0000 ff00 ffff ffff ffff
> 0000020 00ff a5ff 0f0a 0210 161d 000f 0040 0000
> 0000040 0000 0300 4000 5004 2050 3c00 0039 0400
> 0000060 0000 597a 0d00 b203 00ff a5ff 6000 0410
> 0000100 ff01 1902 00ff a5ff 1000 0460 ff01 1902
> 0000120 00ff a5ff 6000 0610 0a01 2601 00ff a5ff
> 0000140 1000 0660 0a01 2601 00ff a5ff 6000 0110
> 0000160 0204 04c4 02e2 ffc6 ff00 00a5 6010 0201
> 0000200 e204 fe01 00ff a5ff 6100 0410 ff01 1a02
> 0000220 00ff a5ff 1000 0461 ff01 1a02 00ff a5ff
> 0000240 6100 0610 0401 2101 00ff a5ff 1000 0661

### Actual behavior
With a simple program:
```
var serialport = require("serialport");
var SerialPort = serialport.SerialPort;
var sp = new SerialPort("/dev/ttyUSB0", {
    baudrate:9600,
    databits: 8,
    parity: 'none',
    stopBits: 1,
    flowControl: false,
    parser: serialport.parsers.raw,
});

sp.on('open', function() {
    console.log('open');
    sp.on('data', function(data) {
        console.log('data received: ' + data);
    });
});
```

I get the output 

> pi@raspberrypi:~/ $ node test.js 
> open
> data received: �
> data received: �
> data received: �
> data received: �
> data received: ��
> data received: �
> data received: ��
> data received: 
> data received: �
> data received: �
> 
> data received: 
> data received: 
> data received: 
> data received: 
> data received: 
> data received: @
> data received: 
> data received: 
> data received: 
> data received: 
Etc etc.

If I change to the readline('\n') parser it's not any better (much weirder in fact):
pi@raspberrypi:~/ $ node test3.js 

> open
> data received: �����������
> �����������ed: "@@PP <9zY
> �����������ed: "@@PP <9zY
> �����������ed: "@@PP <9zY
> �����������ed: "@@PP <9zY
> ����`�����������@@PP <9zY
> ����`�����������@@PP <9zY
> �����������ed: "@@PP <9zY
> ����a����a����a!���a!����������

### Steps and/or code to reproduce the issue
Nothing earth shattering here.
> node test.js

I have disabled shell access on the port, and the port is configured as:

> pi@raspberrypi:~/ $ stty -F /dev/ttyUSB0
> speed 9600 baud; line = 0;
> min = 1; time = 0;
> -brkint -icrnl -imaxbel
> -opost -onlcr
> -isig -icanon -iexten -echo -echoe -echok -echoctl -echoke

Any help is appreciated!

reconbot, 2016-05-13:
You're appending the buffer to a string, converting it to ascii in you logging. Try replacing it with 

```js
sp.on('open', function() {
    console.log('open');
});
sp.on('data', function(data) {
    console.log('data received: ', data.toJSON());
});
```

To see exactly what data you're getting.

tagyoureit, 2016-05-14:
Thank you!  Just to add to your comment (for others beginners in my position), the data.toJSON() actually represents an object so it displayed
>data received: [object Object]

I added 
`JSON.stringify(data.toJSON()));`
and then saw data like
>data received: {"type":"Buffer","data":[255,255,255,0,255,165,10,15,16,2,29,8,58,0]}
(Note the output is in hex->decimal implicit conversion.)

To see the raw HEX output, I realized I could just put in
`data.toString('hex'))`
and see
>data received: 00ffa5

thanks again!

reconbot, 2016-05-14:
If you pass the object as the second parameter to log instead of appending
to the string you'll get a shallow version of that automatically. =)

tagyoureit, 2016-05-14:
Ah!  Yes, I clearly missed that the first time.  Very nice.  TYVM.  Is there a way to output hex in the data.toJSON()?  The .toString method concatenates the buffer and it would be easier to work with the JSON object than convert back and forth.

reconbot, 2016-05-14:
The [`Buffer.toJSON()`](https://nodejs.org/api/buffer.html#buffer_buf_tojson) doesn't take any options unfortunately but there are a whole slew of methods you can play with.

closed 2016-05-14 by reconbot