serialport/node-serialport #807: Unreadable / incomplete data when multiple node apps are listening to the same port

ruudverheijden, 2016-05-15:
Operating System and Hardware: Raspbian / Raspberry Pi
NodeJS Version: 4.4.0
Serialport version: 3.1.1
Python Version (if you're having trouble building serialport): 2.7.9

### Expected behavior
I expect that serialport returns the complete message that was send to the serial port.

### Actual behavior
When I have 2 node applications listening to the same serial port I get incomplete and unreadable data in both applications. It seems that only 1 serialport instance is able to exclusively listen to the port at a time. Resulting in both applictions getting only parts of the data, which in turn is unusable.

Example of incorrect data (when running 2 applications listening to the same port):
```
/KFM5KAIFA-M1-3:0.2.8(42)^M
160514185613S)^M
3030303330303030)^M
```

Example of correct data (when running only 1 application):
```
/KFM5KAIFA-METER^M
^M
1-3:0.2.8(42)^M
0-0:1.0.0(160515115644S)^M
0-0:96.1.1(4530303033303030303035393034333134)^M
```

Analysis: for example, on the first line the text "ETER" and 2 newlines are missed.

### Steps and/or code to reproduce the issue
Run 2 instances of a node application that are listening to the same serial port

fivdi, 2016-05-15:
This is the expected behavior. The operating system sends the data received by the serial port back to the first process that reads it. The operating system then throws the data away and it's not available for reading by another process.

To achieve what you want to achieve a third process could be used. This third process could be used to read data from the serial port and send the data read to the other two processes.

ruudverheijden, 2016-05-15:
Thanks for your quick reply!

Do you know if there is a way to determine if another application is already listening to the serial port?
Cause I would like to keep my application as agnostic as possible.

reconbot, 2016-05-15:
We might be "locking" the port or giving an option to do that in a future
release. (Prs welcome) It's already the default on Windows. If you ran your
program there you'd get an error with the second open.

reconbot, 2016-05-15:
And to answer your question the program `lsof` is the only way I know to do that on linux/unix.

reconbot, 2016-05-16:
I opened #810 to address this. Thanks!

closed 2016-05-16 by reconbot