serialport/node-serialport #811: [docs] Best practices for sequences of events

reconbot, 2016-05-16:
I don't think we have good documentation around handling sequences of events that `parsers` doesn't quite cover. 

For example.
 - Waiting for a `ready` byte or bytes before switching to a parser.
 - Using serialport to implement simple protocols (eg, data encapsulation and command passing)
 - Using serialport to implement more complex protocols like [`firmata`](https://github.com/firmata/firmata.js/)

edmont, 2016-12-09:
Yes, I can't figure out how to handle a sequence of commands. I'm able to send the first one and read it's response data, but how can I send a second command and read it's response?

arthurmoises, 2017-10-17:
Hi @edmont! Had any success sending a second command? I'm managing to create a simple protocol, to send a big message from one device to another. The idea is to divide the message. In this case the slave must sent some info to signal that the first message was received allowing the master to send the second part.

edmont, 2017-10-17:
@arthurmoises, I managed to make it work with `await`, but I finally moved to Python :)

erikkallen, 2018-02-14:
I use seq-queue for this