serialport/node-serialport #812: Add node-pre-gyp support for node_abi v43 to enable serialport installation for Electron 1.1.0 without the need for rebuild

paulolc, 2016-05-16:
Hi, pls add serialport-v3.1.2-node-v43-XX-XX.tar.gz to the releases downloads in order to enable serialport installation for Electron 1.1.0 (node_abi v43) without the need for rebuild.

Operating System and Hardware:
Electron: 1.1.0
NodeJS Version: 6.1.0
Serialport version: 3.1.2

### Expected behavior
node-pre-gyp downloads precompiled binary and doesn't trigger node-gyp. This is very convenient as it doesnt require the build prereqs (VS stuff and all) for someone who is trying to get any app based on Electron1.1.0 with serialport.

### Actual behavior
node-gyp triggered

### Steps and/or code to reproduce the issue
```bash
node_modules\.bin\node-pre-gyp.cmd rebuild --target=1.1.0 --arch=x64 --dist-url=https://atom.io/download/atom-shell
```
node-pre-gyp then falls back to node-gyp since serialport-v3.1.2-node-v43-XX-XX.tar.gz is not available.

reconbot, 2016-05-16:
We don't current provide precompiled binaries for electron but this looks like a good way to do that. Our prebuild system is tied to our testing system. So I'd want to try to see how we can test these binaries if we're going to provide them. Pinging #538 as that's the thread where we're working on better supporting electron.

reconbot, 2016-06-25:
I'm going to keep this open. If anyone knows how to test and build binary modules against electron or has seen it in the wild. Please let me know!

reconbot, 2017-01-21:
This is done in beta we have one more open issue #1067

closed 2017-01-21 by reconbot