serialport/node-serialport #813: Question on listening

doubletaketech, 2016-05-16:
Have a general newbie question.  Does this listen to the port continuously and fire when the data changes? 
port.on('data', function(data) {
  console.log(data);
});

In my use case, the device connected to the port sends different codes and different times throughout the day. I need it to listen all the time so my code can handle things differently depending on the code received and send that to my browser.  It is basically changing the video to be played when a hardware button is pushed. 

Just wanted to make sure this is how things worked.

Thanks,
Donnie

reconbot, 2016-05-16:
Yep, the data event fires for all data received. Guaranteed to be at least
one byte. Not guaranteed to be all the data you send at one time.

doubletaketech, 2016-05-16:
ok. great. Thank you!

closed 2016-05-16 by reconbot