serialport/node-serialport #815: cant use it in ubuntu

buzai, 2016-05-20:
Operating System and Hardware:ubuntu 15.10
NodeJS Version: @6.0.0
Serialport version: @3.1.2
Python Version (if you're having trouble building serialport):@2.7.10
(Run `node -v && npm -v && python --version` to get the version numbers.)

### Expected behavior
I just wrote a line of code.

### Actual behavior

/home/chenhao/Desktop/comDemo/node_modules/bindings/bindings.js:91
  throw err
  ^

Error: Could not locate the bindings file. Tried:
 → /home/chenhao/Desktop/comDemo/node_modules/serialport/build/serialport.node
 → /home/chenhao/Desktop/comDemo/node_modules/serialport/build/Debug/serialport.node
 → /home/chenhao/Desktop/comDemo/node_modules/serialport/build/Release/serialport.node
 → /home/chenhao/Desktop/comDemo/node_modules/serialport/out/Debug/serialport.node
 → /home/chenhao/Desktop/comDemo/node_modules/serialport/Debug/serialport.node
 → /home/chenhao/Desktop/comDemo/node_modules/serialport/out/Release/serialport.node
 → /home/chenhao/Desktop/comDemo/node_modules/serialport/Release/serialport.node
 → /home/chenhao/Desktop/comDemo/node_modules/serialport/build/default/serialport.node
 → /home/chenhao/Desktop/comDemo/node_modules/serialport/compiled/6.0.0/linux/x64/serialport.node
    at bindings (/home/chenhao/Desktop/comDemo/node_modules/bindings/bindings.js:88:9)
    at Object.<anonymous> (/home/chenhao/Desktop/comDemo/node_modules/serialport/lib/bindings.js:3:35)
    at Module._compile (module.js:541:32)
    at Object.Module._extensions..js (module.js:550:10)
    at Module.load (module.js:456:32)
    at tryModuleLoad (module.js:415:12)
    at Function.Module._load (module.js:407:3)
    at Module.require (module.js:466:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (/home/chenhao/Desktop/comDemo/node_modules/serialport/lib/serialport.js:16:25)

### Steps and/or code to reproduce the issue

code is --- var SerialPort = require("serialport")

cmd is--- node app.js

reconbot, 2016-05-20:
The c++ binary is missing. Try running npm install again. And report any
errors.

buzai, 2016-05-21:
> serialport@2.1.2 install /home/chenhao/Desktop/nnnn/node_modules/serialport
> node-pre-gyp install --fallback-to-build
it is this when i remove and reinstall

buzai, 2016-05-21:
make: Entering directory '/home/chenhao/Desktop/nnnn/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
  CXX(target) Release/obj.target/serialport/src/serialport_unix.o
  CXX(target) Release/obj.target/serialport/src/serialport_poller.o
  SOLINK_MODULE(target) Release/obj.target/serialport.node
  COPY Release/serialport.node
make: Leaving directory '/home/chenhao/Desktop/nnnn/node_modules/serialport/build'
/home/chenhao/Desktop/nnnn
└─┬ serialport@2.1.2 
  ├── bindings@1.2.1 
  ├── bluebird@3.4.0 
  ├─┬ debug@2.2.0 
  │ └── ms@0.7.1 
  ├── nan@2.2.1 
  ├─┬ node-pre-gyp@0.6.26 
  │ ├─┬ mkdirp@0.5.1 
  │ │ └── minimist@0.0.8 
  │ ├─┬ nopt@3.0.6 
  │ │ └── abbrev@1.0.7 
  │ ├─┬ npmlog@2.0.3 
  │ │ ├── ansi@0.3.1 
  │ │ ├─┬ are-we-there-yet@1.1.2 
  │ │ │ ├── delegates@1.0.0 
  │ │ │ └─┬ readable-stream@2.1.0 
  │ │ │   ├─┬ inline-process-browser@2.0.1 
  │ │ │   │ ├─┬ falafel@1.2.0 
  │ │ │   │ │ ├── acorn@1.2.2 
  │ │ │   │ │ ├── foreach@2.0.5 
  │ │ │   │ │ ├── isarray@0.0.1 
  │ │ │   │ │ └── object-keys@1.0.9 
  │ │ │   │ └─┬ through2@0.6.5 
  │ │ │   │   └─┬ readable-stream@1.0.34 
  │ │ │   │     └── isarray@0.0.1 
  │ │ │   └─┬ unreachable-branch-transform@0.5.1 
  │ │ │     ├── esmangle-evaluator@1.0.0 
  │ │ │     └─┬ recast@0.11.5 
  │ │ │       ├── ast-types@0.8.16 
  │ │ │       ├── esprima@2.7.2 
  │ │ │       ├── private@0.1.6 
  │ │ │       └── source-map@0.5.4 
  │ │ └─┬ gauge@1.2.7 
  │ │   ├── has-unicode@2.0.0 
  │ │   ├─┬ lodash.pad@4.3.0 
  │ │   │ ├── lodash._baseslice@4.0.0 
  │ │   │ └── lodash.tostring@4.1.2 
  │ │   ├── lodash.padend@4.4.0 
  │ │   └── lodash.padstart@4.4.0 
  │ ├─┬ rc@1.1.6 
  │ │ ├── deep-extend@0.4.1 
  │ │ ├── ini@1.3.4 
  │ │ ├── minimist@1.2.0 
  │ │ └── strip-json-comments@1.0.4 
  │ ├─┬ request@2.72.0 
  │ │ ├── aws-sign2@0.6.0 
  │ │ ├─┬ aws4@1.3.2 
  │ │ │ └─┬ lru-cache@4.0.1 
  │ │ │   ├── pseudomap@1.0.2 
  │ │ │   └── yallist@2.0.0 
  │ │ ├─┬ bl@1.1.2 
  │ │ │ └── readable-stream@2.0.6 
  │ │ ├── caseless@0.11.0 
  │ │ ├─┬ combined-stream@1.0.5 
  │ │ │ └── delayed-stream@1.0.0 
  │ │ ├── extend@3.0.0 
  │ │ ├── forever-agent@0.6.1 
  │ │ ├─┬ form-data@1.0.0-rc4 
  │ │ │ └── async@1.5.2 
  │ │ ├─┬ har-validator@2.0.6 
  │ │ │ ├─┬ chalk@1.1.3 
  │ │ │ │ ├── ansi-styles@2.2.1 
  │ │ │ │ ├── escape-string-regexp@1.0.5 
  │ │ │ │ ├─┬ has-ansi@2.0.0 
  │ │ │ │ │ └── ansi-regex@2.0.0 
  │ │ │ │ ├── strip-ansi@3.0.1 
  │ │ │ │ └── supports-color@2.0.0 
  │ │ │ ├─┬ is-my-json-valid@2.13.1 
  │ │ │ │ ├── generate-function@2.0.0 
  │ │ │ │ ├─┬ generate-object-property@1.2.0 
  │ │ │ │ │ └── is-property@1.0.2 
  │ │ │ │ ├── jsonpointer@2.0.0 
  │ │ │ │ └── xtend@4.0.1 
  │ │ │ └─┬ pinkie-promise@2.0.1 
  │ │ │   └── pinkie@2.0.4 
  │ │ ├─┬ hawk@3.1.3 
  │ │ │ ├── boom@2.10.1 
  │ │ │ ├── cryptiles@2.0.5 
  │ │ │ ├── hoek@2.16.3 
  │ │ │ └── sntp@1.0.9 
  │ │ ├─┬ http-signature@1.1.1 
  │ │ │ ├── assert-plus@0.2.0 
  │ │ │ ├─┬ jsprim@1.2.2 
  │ │ │ │ ├── extsprintf@1.0.2 
  │ │ │ │ ├── json-schema@0.2.2 
  │ │ │ │ └── verror@1.3.6 
  │ │ │ └─┬ sshpk@1.8.1 
  │ │ │   ├── asn1@0.2.3 
  │ │ │   ├── assert-plus@1.0.0 
  │ │ │   ├─┬ dashdash@1.13.1 
  │ │ │   │ └── assert-plus@1.0.0 
  │ │ │   ├─┬ ecc-jsbn@0.0.1 
  │ │ │   │ └── jsbn@0.0.0  (git+https://github.com/rynomad/jsbn.git#bb522b0124f75424f89d49446c40a87111942c7b)
  │ │ │   ├─┬ getpass@0.1.5 
  │ │ │   │ └── assert-plus@1.0.0 
  │ │ │   ├── jodid25519@1.0.2 
  │ │ │   ├── jsbn@0.1.0 
  │ │ │   └── tweetnacl@0.13.3 
  │ │ ├── is-typedarray@1.0.0 
  │ │ ├── isstream@0.1.2 
  │ │ ├── json-stringify-safe@5.0.1 
  │ │ ├─┬ mime-types@2.1.10 
  │ │ │ └── mime-db@1.22.0 
  │ │ ├── node-uuid@1.4.7 
  │ │ ├── oauth-sign@0.8.1 
  │ │ ├── qs@6.1.0 
  │ │ ├── stringstream@0.0.5 
  │ │ ├── tough-cookie@2.2.2 
  │ │ └── tunnel-agent@0.4.2 
  │ ├─┬ rimraf@2.5.2 
  │ │ └─┬ glob@7.0.3 
  │ │   ├── inflight@1.0.4 
  │ │   ├─┬ minimatch@3.0.0 
  │ │   │ └─┬ brace-expansion@1.1.3 
  │ │   │   ├── balanced-match@0.3.0 
  │ │   │   └── concat-map@0.0.1 
  │ │   └── path-is-absolute@1.0.0 
  │ ├── semver@5.1.0 
  │ ├─┬ tar@2.2.1 
  │ │ ├── block-stream@0.0.8 
  │ │ ├─┬ fstream@1.0.8 
  │ │ │ └── graceful-fs@4.1.3 
  │ │ └── inherits@2.0.1 
  │ └─┬ tar-pack@3.1.3 
  │   ├── fstream-ignore@1.0.3 
  │   ├─┬ once@1.3.3 
  │   │ └── wrappy@1.0.1 
  │   ├─┬ readable-stream@2.0.6 
  │   │ ├── core-util-is@1.0.2 
  │   │ ├── isarray@1.0.0 
  │   │ ├── process-nextick-args@1.0.6 
  │   │ ├── string_decoder@0.10.31 
  │   │ └── util-deprecate@1.0.2 
  │   └── uid-number@0.0.6 
  ├─┬ node-pre-gyp-github@1.1.2 
  │ ├─┬ commander@2.9.0 
  │ │ └── graceful-readlink@1.0.1 
  │ ├── github@0.2.4 
  │ └── mime@1.3.4 
  ├─┬ object.assign@4.0.3 
  │ ├─┬ define-properties@1.1.2 
  │ │ └── foreach@2.0.5 
  │ ├── function-bind@1.1.0 
  │ └── object-keys@1.0.9 
  ├─┬ optimist@0.6.1 
  │ ├── minimist@0.0.10 
  │ └── wordwrap@0.0.3 
  └── sf@0.1.7 

npm WARN nnnn No description
npm WARN nnnn No repository field.
npm WARN nnnn No license field.

buzai, 2016-05-21:
ok it can work thank u very much

closed 2016-05-21 by buzai