serialport/node-serialport #817: SetCommState: Unknown error code 87

RogerHardiman, 2016-05-23:
Operating System and Hardware: Windows 10. Virtual COM Port from http://freevirtualserialports.com/
NodeJS Version: 4.4.4
Serialport version: 3.1.2

Python Version (if you're having trouble building serialport):
(Run `node -v && npm -v && python --version` to get the version numbers.)

### Expected behavior
open a serial port and send data.

### Actual behavior
Error message SetCommState: Unknown error code 87
is displayed.

### Steps and/or code to reproduce the issue
1) in the node_modules\serial_port\bin folder I executed this command
    node serialport-terminal.js -p COM1

I get the error message.

2) I also have my own code to open a serial port. I get the same error displayed on screen when I try and open the serial port.

3) If I open a serial port that is already opened by some Windows software, I get the expected result of Error: Opening COM2: Access denied.

I was trying to use Virtual COM ports so that I can do a loopback test between two Node applications.

fivdi, 2016-05-23:
Does [this](http://stackoverflow.com/questions/36357472/accessing-virtual-serial-port-with-node-js-gives-either-error-or-no-open-event) help?

RogerHardiman, 2016-05-24:
I have debugged the Windows C++ code in the package and found the problem with node-serialport and the Virtual Comm Ports.

This [Microsoft MSDN web site](https://msdn.microsoft.com/en-gb/library/windows/desktop/aa363201(v=vs.85).aspx) has an example of using a Serial Port which works with a Virtual Comm Port.

  a) Call CreateFile()
  b) Call GetCommsState() to get current 'DCB' data structure (node-serial port is missing this step)
  c) Change the baud rate in the DCB data structure
  d) Call SetCommState to configure the serial port

node-serialport was not calling GetCommState() to get the current state.
I added this to the C++ source and it all worked.

So here is my change to serialport_win.cpp

```
  DCB dcb = { 0 };
  dcb.DCBlength = sizeof(DCB);
  if (data->hupcl == false) {
    dcb.fDtrControl = DTR_CONTROL_DISABLE;  // disable DTR to avoid reset
  } else {
    dcb.fDtrControl = DTR_CONTROL_ENABLE;
  }
```

To this

```
  DCB dcb = { 0 };
  SecureZeroMemory(&dcb, sizeof(DCB));
  dcb.DCBlength = sizeof(DCB);
  
  if (!GetCommState(file, &dcb)) {
	printf("error getting comm state");
	return;
  }

  if (data->hupcl == false) {
    dcb.fDtrControl = DTR_CONTROL_DISABLE;  // disable DTR to avoid reset
  } else {
    dcb.fDtrControl = DTR_CONTROL_ENABLE;
  }

```

I have not had a chance to test this on any physical serial ports (or USB serial ports) but following the MSDN example this is the proper way to do things.

RogerHardiman, 2016-05-24:
I have created Pull Request #824 with the bug fix

reconbot, 2016-05-24:
This is really helpful, I can take a closer look at that code and see about
adding it but what you say makes sense. Even on linux you're supposed to
get the state, modify it and set it again. If you want to submit a PR in
the meantime people with windows machines can test your branch out. And if
it tests out, I'm happy to merge.

---
Francis Gulotta
wizard@roborooter.com

reconbot, 2016-05-24:
You beat me to the punch! I left some comments there. It's great!

RogerHardiman, 2016-05-25:
Hi
Thanks for looking at the Issue.

Thanks for the comments on the PR. The printf in there was my debugging 
code and I never meant to leave it in so I'm glad you spotted it. I'll 
update the PR.

I'm out on the road at the moment but when I get back I have a PL2303 
based USB serial device and a CH340/HL340 based USB serial port plus a 
short serial cable so I can run open COM1 with Node-serialport and open 
COM2 with node-Serialport and send some data between the two devices to 
test the GetCommState

Roger

reconbot, 2016-05-25:
I released `serialport@4.0.0-beta2` with these changes. Thanks!

closed 2016-05-25 by reconbot

RogerHardiman, 2016-05-25:
Hi
Thanks for applying the PR so quickly and getting a beta release issued.
Roger